analogdevicesinc/libsmu #8: Doesn't build on debian stable

rgetz, 2015-01-18:
Tried debian stable (wheezy):

rgetz@pinky:~/edu/libsmu$ cat /etc/debian_version 
7.8

Other than "sudo apt-get install clang", this is pretty much bare bones, so it could be that I'm missing a library (let me know).

rgetz@pinky:~/edu/libsmu$ make
clang++ -g -std=c++11 -Wall -pedantic -O0 -fPIC -MMD -MP -o cli.o -c cli.cpp 
In file included from cli.cpp:6:
In file included from ./libsmu.hpp:12:
In file included from /usr/include/c++/4.6/mutex:39:
/usr/include/c++/4.6/chrono:666:7: error: static_assert expression is not an integral constant expression
      static_assert(system_clock::duration::min()
      ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make: *** [cli.o] Error 1

so, I tried cmake (per the readme):

rgetz@pinky:~/edu/libsmu$ mkdir build
rgetz@pinky:~/edu/libsmu$ cd build/
rgetz@pinky:~/edu/libsmu/build$ cmake ..
-- The C compiler identification is GNU 4.7.2
-- The CXX compiler identification is GNU 4.7.2
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  CMake 2.8.11 or higher is required.  You are running version 2.8.9

-- Configuring incomplete, errors occurred!
rgetz@pinky:~/edu/libsmu/build$

I'm assuming :
 - debian stable is an supported distribution that we want to be able to build on?
 - it should work both ways?

itdaniher, 2015-01-19:
> debian stable is an supported distribution that we want to be able to build on?

Debian Stable (as we know) runs a few years behind. The CMake issue is one I wasn't expecting, but not a shocker. I know Clang++ on Debian Stable is kind of a joke, LLVM 3.0-only in repos? The test fixtures run Debian Testing to make life easier.

> it should work both ways?

Quite possibly, all of these issues could possibly be fixed moving off cmake and using gcc++ over clang++.

rgetz, 2015-01-19:
If you are not supporting debian stable -- that's OK, it's just something that should be in the docs. (not specifically about Debian, but "Clang++ above version xxx")

itdaniher, 2015-01-21:
Closing this issue as wontfix for now - don't have the bandwidth to try and clear up the backporting issues. A binary build of clang++ from http://llvm.org/apt/ would probably solve most of the issues.

closed 2015-01-21 by itdaniher