analogdevicesinc/libsmu #36: provide compiled .lib file for windows static-linking

itdaniher, 2015-03-23:


itdaniher, 2015-03-23:
https://wiki.videolan.org/GenerateLibFromDll

itdaniher, 2015-03-23:
https://adrianhenke.wordpress.com/2008/12/05/create-lib-file-from-dll/

itdaniher, 2015-03-24:
f1fad63

itdaniher, 2015-03-25:
not quite there yet - the 'lib' command only works from inside msvc, not sure how to do this in appveyor yet....

pcercuei, 2015-03-26:
Libiio uses CMake and VS automatically creates a libiio.lib for me. I pushed a "cmake" branch for libsmu if you want to test this.
The correct way to do it is to use __declspec(dllexport) in the prototype of all the functions that should be exported to the .lib file, and use __declspec(dllimport) in the public header.

See how it's handled in libiio: https://github.com/analogdevicesinc/libiio/blob/master/iio.h#L50

itdaniher, 2015-03-30:
I believe I have a solution working with "dlltool", implementation at https://github.com/analogdevicesinc/libsmu/blob/make.lib/appveyor.yml#L15.

@acozma could you test with https://ci.appveyor.com/project/analogdevicesinc/libsmu/build/1.0.32/artifacts?

Thanks!

itdaniher, 2015-03-30:
@pcercuei Thanks for the information!

itdaniher, 2015-04-13:
```
Error using mx
   Creating library libsmu_mx.lib and object libsmu_mx.exp
libsmu_mx.obj : error LNK2019: unresolved external symbol "public: __cdecl Session::Session(void)" (??0Session@@QEAA@XZ) referenced in function
"int __cdecl initSession(void)" (?initSession@@YAHXZ)
libsmu_mx.obj : error LNK2019: unresolved external symbol "public: __cdecl Session::~Session(void)" (??1Session@@QEAA@XZ) referenced in function
"public: void * __cdecl Session::`scalar deleting destructor'(unsigned int)" (??_GSession@@QEAAPEAXI@Z)
libsmu_mx.obj : error LNK2019: unresolved external symbol "public: int __cdecl Session::update_available_devices(void)"
(?update_available_devices@Session@@QEAAHXZ) referenced in function "int __cdecl initSession(void)" (?initSession@@YAHXZ)
libsmu_mx.obj : error LNK2019: unresolved external symbol "public: class Device * __cdecl Session::add_device(class Device *)"
(?add_device@Session@@QEAAPEAVDevice@@PEAV2@@Z) referenced in function "int __cdecl getDevInfo(class std::vector<struct sl_device_info,class
std::allocator<struct sl_device_info> > &,class std::vector<struct sl_channel_info,class std::allocator<struct sl_channel_info> > &,class
std::vector<struct sl_signal_info,class std::allocator<struct sl_signal_info> > &,class std::vector<char const *,class std::allocator<char const *>
> &)"
(?getDevInfo@@YAHAEAV?$vector@Usl_device_info@@V?$allocator@Usl_device_info@@@std@@@std@@AEAV?$vector@Usl_channel_info@@V?$allocator@Usl_channel_info@@@std@@@2@AEAV?$vector@Usl_signal_info@@V?$allocator@Usl_signal_info@@@std@@@2@AEAV?$vector@PEBDV?$allocator@PEBD@std@@@2@@Z)
libsmu_mx.obj : error LNK2019: unresolved external symbol "public: void __cdecl Session::configure(unsigned __int64)"
(?configure@Session@@QEAAX_K@Z) referenced in function "int __cdecl getInputs(int,int,int,class std::vector<float,class std::allocator<float> >
&,class std::vector<float,class std::allocator<float> > &)" (?getInputs@@YAHHHHAEAV?$vector@MV?$allocator@M@std@@@std@@0@Z)
libsmu_mx.obj : error LNK2019: unresolved external symbol "public: void __cdecl Session::start(unsigned __int64)" (?start@Session@@QEAAX_K@Z)
referenced in function "int __cdecl getInputs(int,int,int,class std::vector<float,class std::allocator<float> > &,class std::vector<float,class
std::allocator<float> > &)" (?getInputs@@YAHHHHAEAV?$vector@MV?$allocator@M@std@@@std@@0@Z)
libsmu_mx.obj : error LNK2019: unresolved external symbol "public: void __cdecl Session::wait_for_completion(void)"
(?wait_for_completion@Session@@QEAAXXZ) referenced in function "int __cdecl getInputs(int,int,int,class std::vector<float,class
std::allocator<float> > &,class std::vector<float,class std::allocator<float> > &)" (?getInputs@@YAHHHHAEAV?$vector@MV?$allocator@M@std@@@std@@0@Z)
libsmu_mx.obj : error LNK2019: unresolved external symbol "public: void __cdecl Session::end(void)" (?end@Session@@QEAAXXZ) referenced in function
"int __cdecl cleanupSession(void)" (?cleanupSession@@YAHXZ)
libsmu_mx.mxw64 : fatal error LNK1120: 8 unresolved externals
```

radhermit, 2016-03-25:
Windows builds are now using VS only along with the newer CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS toggle in >=cmake-3.4 that does all the symbol exporting for us automatically instead of having to mark things manually.

closed 2016-03-25 by radhermit