analogdevicesinc/libsmu #40: Binary distribution of Libsmu and pysmu

damercer, 2015-04-12:
What's the plan for distributing Windows, Linux etc. binaries for libsmu and pysmu?
Since libsmu is obviously compiled for Pixelpulse2 would it make sense to include
these in that bundle?

You might also include more detailed build instructions for Windows. What is there
works for Linux but need more on dependencies for Windows ( i.e MinGW ).

itdaniher, 2015-04-21:
We're going to distribute separate installers for libsmu, libpysmu and Pixelpulse2.

Libpysmu.dll should be available shortly via Appveyor.

itdaniher, 2015-04-21:
Some subtle issues - the build has completed, but it is not importable with Python 2.7.7.

Note that afaict "pyd" and "dll" are interchangable, as per https://docs.python.org/2/faq/windows.html#is-a-pyd-file-the-same-as-a-dll.

itdaniher, 2015-04-21:
After upgrading to Python 2.7.8, attempts to import libpysmu find the relevant DLL in C:\Python27\DLLs, but fail to import it.

```
>>> import libpysmu
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: DLL load failed: %1 is not a valid Win32 application.
```

itdaniher, 2015-04-21:
Apparently this error results from a mismatch in architecture between the host computer, the host python installation, and the binary library...

itdaniher, 2015-04-21:
LibSMU is built x86-only. Local Python install is x86-64. Reinstalling Python...

itdaniher, 2015-04-21:
Implemented & merged.

closed 2015-04-21 by itdaniher