analogdevicesinc/libsmu #44: using shared pointers causes 'smu.exe' to segfault

itdaniher, 2015-04-16:
@kevinmehall - you had asked "why not shared_ptr"

```
(gdb) bt
#0  0x004027e4 in __gnu_cxx::__atomic_add (__mem=0x88324, __val=1) at c:/mingw32/i686-w64-mingw32/include/c++/ext/atomicity.h:53
#1  0x00402881 in __gnu_cxx::__atomic_add_dispatch (__mem=0x88324, __val=1) at c:/mingw32/i686-w64-mingw32/include/c++/ext/atomicity.h:96
#2  0x004245b7 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_add_ref_copy (this=0x88320) at c:/mingw32/i686-w64-mingw32/include/c++/bits/shared_ptr_base.h:133
#3  0x004241e0 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count (this=0x369fc74, __r=...) at c:/mingw32/i686-w64-mingw32/include/c++/bits/shared_ptr_bas
#4  0x00423f99 in std::__shared_ptr<Device, (__gnu_cxx::_Lock_policy)2>::__shared_ptr (this=0x369fc70) at c:/mingw32/i686-w64-mingw32/include/c++/bits/shared_ptr_base.h:9
#5  0x00423a15 in std::shared_ptr<Device>::shared_ptr (this=0x369fc70) at c:/mingw32/i686-w64-mingw32/include/c++/bits/shared_ptr.h:103
#6  0x0040354d in Session::find_existing_device (this=0x793bc8, device=0x27a8788) at session.cpp:181
#7  0x00402de8 in Session::detached (this=0x793bc8, device=0x27a8788) at session.cpp:85
#8  0x00402fb9 in hotplug_callback_usbthread@16 (ctx=0x793ce0, device=0x27a8788, event=LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT, user_data=0x793bc8) at session.cpp:111
#9  0x00409d5d in usbi_hotplug_match_cb (ctx=0x793ce0, dev=<optimized out>, event=<optimized out>, hotplug_cb=0x798138) at hotplug.c:179
#10 0x00409de8 in usbi_hotplug_match (ctx=0x793ce0, dev=0x27a8788, event=LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT) at hotplug.c:192
#11 0x0040a8eb in handle_events (ctx=ctx@entry=0x793ce0, tv=tv@entry=0x369fdf8) at io.c:2072
#12 0x0040b2b8 in libusb_handle_events_timeout_completed@12 (ctx=ctx@entry=0x793ce0, tv=tv@entry=0x369fe38, completed=completed@entry=0x0) at io.c:2186
#13 0x0040b3ae in libusb_handle_events@4 (ctx=0x793ce0) at io.c:2262
#14 0x00402fe4 in Session::<lambda()>::operator()(void) const (__closure=0x798198) at session.cpp:120
#15 0x00404ce3 in std::_Bind_simple<Session::start_usb_thread()::<lambda()>()>::_M_invoke<>(std::_Index_tuple<>) (this=0x798198) at c:/mingw32/i686-w64-mingw32/include/c+
#16 0x00404c5a in std::_Bind_simple<Session::start_usb_thread()::<lambda()>()>::operator()(void) (this=0x798198) at c:/mingw32/i686-w64-mingw32/include/c++/functional:168
#17 0x00404bf8 in std::thread::_Impl<std::_Bind_simple<Session::start_usb_thread()::<lambda()>()> >::_M_run(void) (this=0x79818c) at c:/mingw32/i686-w64-mingw32/include/c
#18 0x6fcb8c4f in libstdc++-6!execute_native_thread_routine () from c:\Users\IDANIHER\Desktop\distrib-release\libstdc++-6.dll
#19 0x64944f11 in pthread_create_wrapper () from c:\Users\IDANIHER\Desktop\distrib-release\libwinpthread-1.dll
#20 0x75131287 in msvcrt!_itow_s () from C:\Windows\syswow64\msvcrt.dll
#21 0x75131328 in msvcrt!_endthreadex () from C:\Windows\syswow64\msvcrt.dll
#22 0x76fd33aa in KERNEL32!BaseThreadInitThunk () from C:\Windows\syswow64\kernel32.dll
#23 0x7772bf32 in ntdll!RtlInitializeExceptionChain () from C:\Windows\system32\ntdll.dll
#24 0x7772bf05 in ntdll!RtlInitializeExceptionChain () from C:\Windows\system32\ntdll.dll
#25 0x00000000 in ?? ()
```

kevinmehall, 2015-04-16:
Memory corruption of a shared_ptr doesn't mean "Don't use shared_ptr and leak all the memory instead", it means something is corrupting memory and would hit a normal pointer too.

itdaniher, 2015-04-16:
Until yesterday with the addition of `destroy_available`, `m_available_devices` wasn't cleaned up except for once with openAllDevices and once with closeAllDevices.

acozma, 2015-12-04:
Fixed in this commit: https://github.com/analogdevicesinc/libsmu/commit/2d24567204e2ecaa6748897b71239f0bd92fe7b7

closed 2015-12-04 by acozma