analogdevicesinc/libsmu #46: continuous waveform generation documentation from pysmu

itdaniher, 2015-04-20:
@radhermit any examples of this? does the pysmu wrapper support it?

itdaniher, 2015-04-20:
@damercer

radhermit, 2015-05-11:
Looks like I missed seeing this a while ago. Yes, some form of continuous support is now available via Device.samples in the python bindings that provides an iterator of sample tuples.

I'll try to figure out some better way to document things going forward either in the ADI wiki or more of an auto-generated approach by pulling docstrings from the bindings.

damercer, 2015-05-11:
I may have miss-communicated what I was asking for. What I was asking about is continuous analog output waveforms not continuous input samples. As a matter of fact I see no reason for continuous input sampling. Unless the input and output are some how inextricably linked.

As to documentation, a users guide on the ADI Wiki including detailed examples would be great.
I doubt pulling some comments from the source code will be sufficient (obviously given the above confusion).