analogdevicesinc/libsmu #47: Windows libpysmu.dll doen't work

damercer, 2015-04-21:
The newly created libpysmu.dll returns the following error when pysmu.py is run:

Traceback (most recent call last):
  File "C:\My_Documents\ALM\pysmu\pysmu.py", line 12, in <module>
    import libpysmu
ImportError: No module named libpysmu
>>> 

libpysmu.dll is located in same dir as pysmu.py

itdaniher, 2015-04-21:
libpysmu.pyd / libpysmu.dll needs to be in c:\Python27, not a local directory.

Also, original build was missing link flags. Make sure you're running Python2.7.8, 32b version, and try https://ci.appveyor.com/project/analogdevicesinc/libsmu/build/1.0.119 when it finishesd.

itdaniher, 2015-04-21:
Fixed at that build. Successfully imported libpysmu.

closed 2015-04-21 by itdaniher