analogdevicesinc/libsmu #50: invalid device serial number

dansteingart, 2015-09-10:
On ubuntu 15.04 and 14.04. as well as OS X 10.10, I get this error upon running pysmu.py:

```
Using libusb hotplug
device insert 0xb40330
Traceback (most recent call last):
  File "pysmu.py", line 143, in <module>
    A.set_mode('v')
  File "pysmu.py", line 84, in set_mode
    libpysmu.set_mode(self.dev, self.chan, 1)
ValueError: invalid device serial number
```

thanks in advance

dansteingart, 2015-09-10:
unplugging and plugging 3 times back in worked.... anything deterministic that can be done here?

dansteingart, 2015-09-10:
after a while it defaults back to this error.  any insight would be much appreciated

dansteingart, 2015-09-11:
if it helps, when it fails I get this error:  

[image]

radhermit, 2016-04-05:
Sorry for responding so slowly, does this still occur for you using the latest code in master? I'm unable to reproduce it.

cristi-iacob, 2019-08-20:
We have modified the hotplug mechanism. It will be available in the next version. It should solve this problem. Thanks!

closed 2019-08-20 by cristi-iacob