analogdevicesinc/libsmu #51: constant voltage mode doesn't work

dansteingart, 2015-09-11:
Regardless of what I set the potential to be, on either channel a or b, the potential is always either ~2.5 V or .3 .  The following code 

```python

import pysmu
from time import sleep,time
x = pysmu.Smu()
A = x.channels['A']

A.set_mode('v')

for i in range(1,10):
    s = 1+i/10.
    print 'set:',s,';get:',
    A.constant(s)
    print A.get_samples(1)
    sleep(1)

```

yields 

```
set: 1.1 ;get: [(0.35614556074142456, 0.0006613832083530724)]
set: 1.2 ;get: [(2.359273672103882, -0.2075403481721878)]
set: 1.3 ;get: [(0.35591667890548706, 0.000722419295925647)]
set: 1.4 ;get: [(2.4430456161499023, -0.2075556069612503)]
set: 1.5 ;get: [(0.35584038496017456, 0.0006384946755133569)]
set: 1.6 ;get: [(2.4439611434936523, -0.2075556069612503)]
set: 1.7 ;get: [(0.35637444257736206, 0.0006537536974065006)]
set: 1.8 ;get: [(0.35599297285079956, 0.0006613832083530724)]
set: 1.9 ;get: [(2.4438085556030273, -0.2075403481721878)]
```

damercer, 2015-10-17:
You need to add a call to set_mode inside your for loop to reconnect the
output source. You see the output(s) go back into an off high impedance
state after a set of samples are taken:

This should work for you:

import pysmu
from time import sleep,time
x = pysmu.Smu()
A = x.channels['A']

A.set_mode('v')

for i in range(1,10):
    s = 1+i/10.
    print 'set:',s,';get:',
    A.set_mode('v')
    A.constant(s)
    print A.get_samples(1)
    sleep(1)

dansteingart, 2015-10-17:
Oh.  Thanks, will give that a go.  Is there way to disable that behavior,
for this application it's important that the potential stay applied without
any interruptions, particular those that require communication of the USB
bus.

damercer, 2015-10-18:
At the moment no, "continuous " mode has not been implemented.
I believe it is on Tim Harder's to-do list. Not sure when it will make
it into the pysmu functions.

damercer, 2015-10-18:
I tried some of the direct control functions of the signal chain switches
and came up with the following brute force way to make constant DC
voltages appear on the analog outputs. I included the magic settings
for channel B as well.

Not sure what will happen if you get more samples with a second 
A.get_samples(1), you may need to resend the control_transfer settings 
each time and the output may glitch.

#
import pysmu
from time import sleep,time
x = pysmu.Smu()
DevID = x.serials[0]   # device ID for 1st M1000 in list
A = x.channels['A']
B = x.channels['B']
A.set_mode('v')
B.set_mode('v')
A.constant(1.5) # set CHA to 1.5V
B.constant(1.75) # set CHB to 1.75V
print A.get_samples(1)
#
x.ctrl_transfer(DevID, 0x40, 0x50, 51, 0, 0, 0, 100) # set A to mode SVMI
x.ctrl_transfer(DevID, 0x40, 0x50, 34, 0, 0, 0, 100) # close A voltage sense loop
x.ctrl_transfer(DevID, 0x40, 0x50, 35, 0, 0, 0, 100) # ChA connect output
#
x.ctrl_transfer(DevID, 0x40, 0x50, 52, 0, 0, 0, 100) # set B to mode SVMI
x.ctrl_transfer(DevID, 0x40, 0x50, 39, 0, 0, 0, 100) # close B voltage sense loop
x.ctrl_transfer(DevID, 0x40, 0x50, 40, 0, 0, 0, 100) # ChB connect output

dansteingart, 2015-10-18:
Very cool, will give that a go as well.  Would love to have a better sense
of how to get down to that level: other than digging into the µC source is
there a guide?

damercer, 2015-10-18:
Some of this, but not all, is documented here:
https://github.com/analogdevicesinc/m1k-fw/blob/master/IO.mkd
But probably will not make a lot of sense without thoroughly understanding
how the hardware works. Here is the schematic:
https://github.com/analogdevicesinc/m1k-cad/blob/master/02-037760-01-d-1.pdf
Some early drafts of Python tutorials at:
https://wiki.analog.com/university/tools/python-tutorial/table-of-contents
More to come as I have time to write and test them.

dansteingart, 2016-06-02:
just wondering if continuous mode been enabled yet?  Thanks!

radhermit, 2016-06-02:
The underlying architecture of libsmu is being redone and this will be supported when that is finished.

dansteingart, 2017-01-02:
Happy New Year: any action on the above? Thanks!

radhermit, 2017-01-06:
Continuous mode is implemented in the rework branch which is where most of the active development has been occurring for now.

closed 2019-07-03 by AlexandraTrifan