analogdevicesinc/libsmu #52: ADALM1000, libpysmu and Python error

Robotek1024, 2015-10-12:
I originally posted this to the AnalogDevices forum and was advised that I should report it here, if this is in error, I apologize.

///start of original message
I am having a problem getting any python programs to compile for the ADALM1000.  To be fair, I am new to both the ADALM1000 and Python, so it is likely something I am doing incorrectly.

I am using Windows 10, Python 2.7.10, and as instructed placed the file libpysmu.pyd into the Python 2.7 dll's directory.  The error that I am getting is that it is failing to load the dll.  Python Shell Output is below.  
Can anyone tell me what I am doing wrong or how to work around this.

/////////////////Python Shell output

Python 2.7.10 (default, May 23 2015, 09:44:00) [MSC v.1500 64 bit (AMD64)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> ================================ RESTART ================================
>>> 

Traceback (most recent call last):
  File "C:\Users\Ken McKinnon\Documents\alm1000_tutorial\toggle_led.py", line 7, in <module>
    import libpysmu
ImportError: DLL load failed: %1 is not a valid Win32 application.
>>> 

 

Thanks.
Ken

itdaniher, 2015-10-12:
Ken,

Thanks for reaching out. I don't have a Windows 10 machine on hand, but I'll try and dig one up. There are a few interactions I find suspect:
 * 32b vs 64b - your install is 64b, the .pyd is 32b. I'm slightly confused by "[MSC v.1500 64 bit (AMD64)] on win32"
 * Python 2.7.8 vs 2.7.10 - the latest version *was* 2.7.8, but this appears to be no longer the case.

Would you try the libpysmu.pyd file (hopefully) produced by https://ci.appveyor.com/project/analogdevicesinc/libsmu/build/1.0.137? I modified the build script to build bindings for 2.7.10, x86-64. Hopefully that solves your issues!

Regards,
Ian

Robotek1024, 2015-10-12:
Thank you for your response, Ian.

Am I to get that from: 
https://ci.appveyor.com/project/analogdevicesinc/libsmu/build/1.0.137/artifacts 
or from the main site?
One thing I am confused about is that there are a number of artifacts 
that are created, are they to be downloaded and placed somewhere as well?

Ken

itdaniher, 2015-10-12:
You want https://ci.appveyor.com/api/buildjobs/eylulmw0h8n6nwx8/artifacts/libpysmu.pyd, replace the libpysmu.pyd in the dlls folder.

Robotek1024, 2015-10-12:
Unfortunately, the same result.

Is it possible that I missed something in the install, or need to 
reinstall Python?

Ken

Python 2.7.10 (default, May 23 2015, 09:44:00) [MSC v.1500 64 bit 
(AMD64)] on win32
Type "copyright", "credits" or "license()" for more information.
 >>> ================================ RESTART 
================================
 >>>

Traceback (most recent call last):
   File "C:\Users\Ken 
McKinnon\Documents\alm1000_tutorial\toggle_led.py", line 7, in <module>
     import libpysmu
ImportError: DLL load failed: %1 is not a valid Win32 application.
 >>>

itdaniher, 2015-10-12:
> Is it possible that I missed something in the install, or need to reinstall Python?
I'm not rightfully sure. Thanks for checking my first-pass solution, I'm going to have to dig into this tomorrow.

Robotek1024, 2015-10-12:
No problem, as I said, thanks for your help.

Ken

itdaniher, 2015-10-12:
My pleasure! :)

radhermit, 2016-01-20:
@Robotek1024: I've fixed the auto-build for win64 support. Please try the latest 64-bit build from https://ci.appveyor.com/api/buildjobs/fy54eknvuqedk28t/artifacts/libpysmu.pyd to see if it works for you. If it doesn't please reopen or file a new issue.

I might try to roll things into a binary wheel file or something that's easier for people to use later on.

closed 2016-01-20 by radhermit