analogdevicesinc/libsmu #53: make file fails in OS X

JP-Git, 2015-10-23:
Hi,

Just wondering if there are any plans to get this working on OS X? Currently it does not look like the Makefile can actually work in OS X. It currently ends with this error:

session.cpp:8:10: fatal error: 'libusb.h' file not found
#include <libusb.h>
         ^
1 error generated.
make: *** [session.o] Error 1

I do have libusb installed, but the makefile doesn't seem to know where to look.

Thanks in advance for your time.

itdaniher, 2015-10-26:
How did you install libusb, via homebrew, or by hand? Our [travis-ci script](https://github.com/analogdevicesinc/Pixelpulse2/blob/master/.travis.yml) for compiling Pixelpulse2 uses the command:

> brew install --build-from-source libusb

Regards!
Ian

radhermit, 2016-01-20:
Try to follow the instructions for OS X now in the readme, if those don't work please file another issue or pull request.

closed 2016-01-20 by radhermit