analogdevicesinc/libsmu #64: no module named libpysmu

bcavileer, 2016-03-26:
I'm trying to use an ADALM1000 with ALICE and I'm having trouble.   

```
import libpysmu
ImportError: No module named libpysmu
```

I'm following the instructions here: https://wiki.analog.com/university/tools/m1k/alice/install#manually_installing_libpysmu_and_alice_python_source

The device works with PixelPulse and the smu command-line tool works as well.

---

I've tried several variations, 

git clone https://github.com/analogdevicesinc/libsmu.git
mkdir libsmu/build && cd libsmu/build
cmake -DBUILD_PYTHON=ON ..
make
sudo make install

git clone https://github.com/analogdevicesinc/libsmu.git
mkdir libsmu/build && cd libsmu/build
cmake ..
make python
sudo make install

---

Description:	Ubuntu 15.10
Release:	15.10
Codename:	wily
Python 2.7.10

➜  ~ python
Python 2.7.10
>>> import pysmu
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "pysmu.py", line 12, in <module>
    import libpysmu
ImportError: No module named libpysmu

---

Here is the result of the build
➜  build git:(master) ✗ cmake -DBUILD_PYTHON=ON ..   
-- The CXX compiler identification is GNU 5.2.1
-- The C compiler identification is GNU 5.2.1
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- checking for module 'libusb-1.0'
--   found libusb-1.0, version 1.0.19
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found version "2.7.10") 
-- Looking for getopt
-- Looking for getopt - found
-- Configuring done
-- Generating done
-- Build files have been written to: /home/benjamin/libsmu/build
➜  build git:(master) ✗ make
Scanning dependencies of target smu
[ 20%] Building CXX object src/CMakeFiles/smu.dir/session.cpp.o
[ 40%] Building CXX object src/CMakeFiles/smu.dir/device_cee.cpp.o
[ 60%] Building CXX object src/CMakeFiles/smu.dir/device_m1000.cpp.o
Linking CXX shared library libsmu.so
[ 60%] Built target smu
Scanning dependencies of target python
[ 80%] Generating timestamp
running build
running build_py
running build_ext
[ 80%] Built target python
Scanning dependencies of target smu_bin
[100%] Building CXX object src/cli/CMakeFiles/smu_bin.dir/smu.cpp.o
Linking CXX executable smu
[100%] Built target smu_bin
➜  build git:(master) ✗ sudo make install
[ 60%] Built target smu
[ 80%] Generating timestamp
running build
running build_py
running build_ext
[ 80%] Built target python
[100%] Built target smu_bin
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/lib/pkgconfig/libsmu.pc
-- Up-to-date: /etc/udev/rules.d/53-adi-m1k-usb.rules
-- Installing: /usr/local/lib/libsmu.so.0.8.9
-- Up-to-date: /usr/local/lib/libsmu.so.0
-- Up-to-date: /usr/local/lib/libsmu.so
-- Up-to-date: /usr/local/include/libsmu.hpp
running install
running bdist_egg
running egg_info
creating pysmu.egg-info
writing pysmu.egg-info/PKG-INFO
writing top-level names to pysmu.egg-info/top_level.txt
writing dependency_links to pysmu.egg-info/dependency_links.txt
writing manifest file 'pysmu.egg-info/SOURCES.txt'
warning: manifest_maker: standard file 'setup.py' not found

reading manifest file 'pysmu.egg-info/SOURCES.txt'
writing manifest file 'pysmu.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_ext
building 'pysmu._pysmu' extension
creating build
creating build/temp.linux-x86_64-2.7
creating build/temp.linux-x86_64-2.7/home
creating build/temp.linux-x86_64-2.7/home/benjamin
creating build/temp.linux-x86_64-2.7/home/benjamin/alice
creating build/temp.linux-x86_64-2.7/home/benjamin/libsmu
creating build/temp.linux-x86_64-2.7/home/benjamin/libsmu/bindings
creating build/temp.linux-x86_64-2.7/home/benjamin/libsmu/bindings/python
creating build/temp.linux-x86_64-2.7/home/benjamin/libsmu/bindings/python/src
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/home/benjamin/libsmu/src -I/usr/include/libusb-1.0 -I/usr/include/python2.7 -c /home/benjamin/libsmu/bindings/python/src/pysmu.cpp -o build/temp.linux-x86_64-2.7/home/benjamin/libsmu/bindings/python/src/pysmu.o -std=c++11
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
/home/benjamin/libsmu/bindings/python/src/pysmu.cpp: In function ‘PyObject* calibration(PyObject*, PyObject*)’:
/home/benjamin/libsmu/bindings/python/src/pysmu.cpp:246:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  for (int i = 0; i < cal.size(); i++) {
                    ^
/home/benjamin/libsmu/bindings/python/src/pysmu.cpp:248:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (int j = 0; j < cal[i].size(); j++) {
                     ^
creating build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/pysmu
c++ -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wl,-Bsymbolic-functions -Wl,-z,relro -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security build/temp.linux-x86_64-2.7/home/benjamin/libsmu/bindings/python/src/pysmu.o -L/home/benjamin/libsmu/src -lsmu -lusb-1.0 -o build/lib.linux-x86_64-2.7/pysmu/_pysmu.so
creating build/bdist.linux-x86_64
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/pysmu
copying build/lib.linux-x86_64-2.7/pysmu/_pysmu.so -> build/bdist.linux-x86_64/egg/pysmu
creating stub loader for pysmu/_pysmu.so
byte-compiling build/bdist.linux-x86_64/egg/pysmu/_pysmu.py to _pysmu.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying pysmu.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pysmu.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pysmu.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pysmu.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
writing build/bdist.linux-x86_64/egg/EGG-INFO/native_libs.txt
zip_safe flag not set; analyzing archive contents...
creating dist
creating 'dist/pysmu-0.88-py2.7-linux-x86_64.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
Processing pysmu-0.88-py2.7-linux-x86_64.egg
Removing /usr/local/lib/python2.7/dist-packages/pysmu-0.88-py2.7-linux-x86_64.egg
Copying pysmu-0.88-py2.7-linux-x86_64.egg to /usr/local/lib/python2.7/dist-packages
pysmu 0.88 is already the active version in easy-install.pth

Installed /usr/local/lib/python2.7/dist-packages/pysmu-0.88-py2.7-linux-x86_64.egg
Processing dependencies for pysmu==0.88
Finished processing dependencies for pysmu==0.88
-- Installing: /usr/local/bin/smu
-- Removed runtime path from "/usr/local/bin/smu"

radhermit, 2016-03-26:
I don't think ALICE has been updated yet to work with the changed layout of the python bindings that I merged today.

I've attached a patch that modifies ALICE that should make it work with the new layout.

[alice-1.0-pysmu.patch.zip](https://github.com/analogdevicesinc/libsmu/files/190259/alice-1.0-pysmu.patch.zip)

bcavileer, 2016-03-26:
Hi Tim thanks for your quick response!

I applied the patch but I'm still having trouble

I'm not at all experienced with Python so I'm not sure about the import

```
python volt-meter-tool.pyw 
Traceback (most recent call last):
  File "volt-meter-tool.pyw", line 9, in <module>
    from pysmu import _pysmu as libpysmu
ImportError: cannot import name _pysmu

```

```
head -n 15 volt-meter-tool.pyw 
#!/usr/bin/python
# ADALM1000 DC Voltmeter Tool Feb 1 2016
#For Python version > = 2.7.8
from Tkinter import *
import atexit
from collections import defaultdict
from operator import add
import time
from pysmu import _pysmu as libpysmu
# define ADALM1000 interface
class Smu(object):
    def __init__(self):
        atexit.register(libpysmu.cleanup)
        libpysmu.setup()
        self.load_channels()
```

radhermit, 2016-03-26:
Hmm, there's probably some bug with how cmake is calling the python setup.py script to run the install process. You can do it manually for now and that should fix things.

From the root directory of the libsmu repo that you've already built things in:

cd bindings/python
sudo python setup.py install

bcavileer, 2016-03-26:
Hmm that didn't work... any ideas?

```
sudo python setup.py install
running install
running bdist_egg
running egg_info
creating pysmu.egg-info
writing pysmu.egg-info/PKG-INFO
writing top-level names to pysmu.egg-info/top_level.txt
writing dependency_links to pysmu.egg-info/dependency_links.txt
writing manifest file 'pysmu.egg-info/SOURCES.txt'
reading manifest file 'pysmu.egg-info/SOURCES.txt'
writing manifest file 'pysmu.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
creating build/bdist.linux-x86_64
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/pysmu
copying build/lib.linux-x86_64-2.7/pysmu/_pysmu.so -> build/bdist.linux-x86_64/egg/pysmu
copying build/lib.linux-x86_64-2.7/pysmu/__init__.py -> build/bdist.linux-x86_64/egg/pysmu
byte-compiling build/bdist.linux-x86_64/egg/pysmu/__init__.py to __init__.pyc
creating stub loader for pysmu/_pysmu.so
byte-compiling build/bdist.linux-x86_64/egg/pysmu/_pysmu.py to _pysmu.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying pysmu.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pysmu.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pysmu.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pysmu.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
writing build/bdist.linux-x86_64/egg/EGG-INFO/native_libs.txt
zip_safe flag not set; analyzing archive contents...
creating dist
creating 'dist/pysmu-0.88-py2.7-linux-x86_64.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
Processing pysmu-0.88-py2.7-linux-x86_64.egg
Removing /usr/local/lib/python2.7/dist-packages/pysmu-0.88-py2.7-linux-x86_64.egg
Copying pysmu-0.88-py2.7-linux-x86_64.egg to /usr/local/lib/python2.7/dist-packages
pysmu 0.88 is already the active version in easy-install.pth

Installed /usr/local/lib/python2.7/dist-packages/pysmu-0.88-py2.7-linux-x86_64.egg
Processing dependencies for pysmu==0.88
Finished processing dependencies for pysmu==0.88
```

```
➜  python git:(master) ✗ python
Python 2.7.10
>>> from pysmu import _pysmu as libpysmu
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "pysmu/__init__.py", line 12, in <module>
    import _pysmu
ImportError: No module named _pysmu
>>> 

```

bcavileer, 2016-03-26:
I got it to work!  I removed both the pysmu.py and pysmu.pyc files from the ALICE directory.

It seems it was shipped with ALICE but the one installed in the system is now the one I should use due to the patch.

Let me know if I'm on the wrong track but the program is running!

closed 2016-03-26 by bcavileer

radhermit, 2016-03-26:
Nope, that's what my patch should have done (removed the pysmu.py file) so you did the right thing. Let me know if you have issues running any of the tools.