analogdevicesinc/libsmu #71: libsmu Windows installer on 64 bit machine

damercer, 2016-05-07:
The libsmu / pysmu installer works properly on two different 32 bit Windows computers but does not put the pysmu stuff in the proper directories on this Dell 64 bit Windows 7 Enterprise laptop. Rather than installing the pysmu files in C:\Python27\Lib\site-packages it puts it in C:\Lib\site-packages. It also adds a C:\Scripts directory with one file in it. This behavior only seems to happen on this computer. Can't confirm if any of the Python bindings actually runs on the 64 bit machine until the installer puts things in the proper place.

radhermit, 2016-05-08:
While this should be fixed, if the main reason you're using 32 bit libsmu/python installs on 64 bit windows is due to numpy not having official binaries, you could try using the various 64 bit builds available at http://www.lfd.uci.edu/~gohlke/pythonlibs/.

damercer, 2016-05-08:
You should mention in the libsmu readme that at least for now the visual studio runtime needs to be loaded for libsmu / pysmu to work.

radhermit, 2016-05-09:
The problem is I'm unable to reproduce this on my 64 bit windows machine. I've wiped all libsmu/python/VS resdists off it, installed python-2.7 (x86 version), installed libsmu-x86 with the python bindings, and everything works fine (see the image below).

[image]

radhermit, 2016-05-10:
I think this should work as noted in #72 so I'm closing this. If you have other issues please reopen or open a new issue.

closed 2016-05-10 by radhermit