analogdevicesinc/libsmu #72: Libsmu Windows installer and Visual Studio Run Time

damercer, 2016-05-08:
While the libsmu-setup installer has a check box to optionaly install the Visual Studio runtime library ( at least I think that is what the check box is for ) it does not actually install the runtime
library, or at least all of it. I had to download vc_redist.x86.exe ( I'm running 32 bit Python ) and manually install it.

radhermit, 2016-05-08:
That's not what the checkbox is for. It's actually for installing support to build/link stuff against libsmu using VS. If you have error messages with missing dlls, please post them as most of the stuff that we can distribute legally should already be bundled.

damercer, 2016-05-08:
I'd have to uninstall the VS runtime to tell you what dll smu.exe said was missing.

radhermit, 2016-05-09:
Can you do that please? I'm unable to reproduce your issues.

damercer, 2016-05-09:
I uninstalled the VS 2015 runtime as well and everything still worked. I don't know what was going on but nothing worked right until I loaded the runtime library. Now everything is fine. My computer must have had something messed up before hand and loading the runtime stuff set it right and now all is as it should be....

radhermit, 2016-05-10:
Closing since things seem to work, please reopen if you have a reproducible issue.

closed 2016-05-10 by radhermit