analogdevicesinc/libsmu #75: Mac version doesn't compile again

David5i6, 2016-05-19:
ld: library not found for -lusb-1.0

radhermit, 2016-05-19:
Are you following the instructions in the readme and installing everything listed there? Obviously you don't have to use brew but that makes things quicker.

Also, what's the output from cmake when you run that part of the build instructions? It should detect pkg-config and then use that to find libusb.

David5i6, 2016-05-20:
I'm using brew, but perhaps it's a version issue on some package here is the output from the command:

```sh
brew install cmake libusb pkg-config python                                                                                                    
Warning: cmake-3.5.2 already installed
Warning: libusb-1.0.20 already installed
Warning: pkg-config-0.29.1 already installed
Warning: python-2.7.11 already installed, it's just not linked
```

Then cloned the git repository:

```sh
git clone https://github.com/analogdevicesinc/libsmu                                                                                         
Cloning into 'libsmu'...
remote: Counting objects: 2309, done.
remote: Total 2309 (delta 0), reused 0 (delta 0), pack-reused 2309
Receiving objects: 100% (2309/2309), 1.05 MiB | 637.00 KiB/s, done.
Resolving deltas: 100% (1422/1422), done.
Checking connectivity... done.
```

And following instructions...
```sh
mkdir libsmu/build && cd libsmu/build
cmake ..                                                                                                                         git:master
-- The CXX compiler identification is AppleClang 7.3.0.7030031
-- The C compiler identification is AppleClang 7.3.0.7030031
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.20
-- Looking for getopt
-- Looking for getopt - found
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/david/Develop/git/libsmu/build
make                                                                                                                             git:master
Scanning dependencies of target smu
[ 16%] Building CXX object src/CMakeFiles/smu.dir/session.cpp.o
[ 33%] Building CXX object src/CMakeFiles/smu.dir/device_cee.cpp.o
[ 50%] Building CXX object src/CMakeFiles/smu.dir/device_m1000.cpp.o
[ 66%] Linking CXX shared library libsmu.dylib
ld: library not found for -lusb-1.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/libsmu.0.8.9.dylib] Error 1
make[1]: *** [src/CMakeFiles/smu.dir/all] Error 2
make: *** [all] Error 2
```

I've tried to install another version of lib-usb, but didn't found the 1.0 version.

radhermit, 2016-05-20:
Try the latest master, I think c34c9a5 should fix your issue.

David5i6, 2016-05-20:
Thanks ! with dthe last change in master de target smu was built correctly but smu_bin fails with the same error.

I've delted my local repository and cloned again.Repeating previous steps the result now is:

```sh
make                                                         git:master
Scanning dependencies of target smu
[ 16%] Building CXX object src/CMakeFiles/smu.dir/session.cpp.o
[ 33%] Building CXX object src/CMakeFiles/smu.dir/device_cee.cpp.o
[ 50%] Building CXX object src/CMakeFiles/smu.dir/device_m1000.cpp.o
[ 66%] Linking CXX shared library libsmu.dylib
[ 66%] Built target smu
Scanning dependencies of target smu_bin
[ 83%] Building CXX object src/cli/CMakeFiles/smu_bin.dir/smu.cpp.o
[100%] Linking CXX executable smu
ld: library not found for -lusb-1.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/cli/smu] Error 1
make[1]: *** [src/cli/CMakeFiles/smu_bin.dir/all] Error 2
make: *** [all] Error 2
```

radhermit, 2016-05-20:
Hah, I figured that might happen but was hoping cmake would just carry the setting along for all targets. Anyway, this should be fixed in cd9c8e6.

David5i6, 2016-05-21:
Thanks again ! Now It compiles without problems.

closed 2016-05-21 by David5i6