analogdevicesinc/libsmu #76: Python binding does not compile on OS X

David5i6, 2016-05-21:
On OS X, compilation gives an error on `#include <cstdint>`, this include must be replaced with `#include <tr1/cstdint>`.

After changing that there is an error on `#include <condition_variable>`.

Here is the output of compilation after replacing `cstdint` with `tr1/cstdint`:

```sh
make                                                        git:master*
[ 57%] Built target smu
[ 71%] Generating timestamp
running build
running build_py
running build_ext
building 'pysmu._pysmu' extension
/usr/bin/clang -fno-strict-aliasing -fno-common -dynamic -arch i386 -arch x86_64 -g -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -I/Users/david/Develop/libsmu/src -I/usr/local/Cellar/libusb/1.0.20/include/libusb-1.0 -I/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c /Users/david/Develop/libsmu/bindings/python/src/pysmu.cpp -o build/temp.macosx-10.6-intel-2.7/Users/david/Develop/libsmu/bindings/python/src/pysmu.o -std=c++11
/Users/david/Develop/libsmu/bindings/python/src/pysmu.cpp:30:10: fatal error: 
      'condition_variable' file not found
#include <condition_variable>
         ^
1 error generated.
error: command '/usr/bin/clang' failed with exit status 1
make[2]: *** [bindings/python/timestamp] Error 1
make[1]: *** [bindings/python/CMakeFiles/python.dir/all] Error 2
make: *** [all] Error 2
```

radhermit, 2016-05-21:
I don't hit those build failures, what llvm/clang versions are you running (clang --version)? I've got "Apple LLVM version 7.0.2 (clang-700.1.81) running on OS X 10.10.5.

David5i6, 2016-05-23:
I'm using the version that comes with xcode 7.3.1:

```sh
Apple LLVM version 7.3.0 (clang-703.0.31)
Target: x86_64-apple-darwin15.5.0
Thread model: posix
```
Running on OSX 10.11.5 (El Capitan)

radhermit, 2016-05-23:
Thanks, I'll update this box to try reproducing it.

radhermit, 2016-05-23:
So I've updated both OS X and Xcode (clang) and can't reproduce this. Current xcode version info is the same as yours:

Apple LLVM version 7.3.0 (clang-703.0.31)
Target: x86_64-apple-darwin15.5.0
Thread model: posix

radhermit, 2016-05-24:
What cmake version do you have installed? Also, can you run make with VERBOSE=1 set to enable verbose build logs and attach or paste the relevant python build output/failures here again?

Other than that, the only remaining thing I can think of right now is different compiler or other Xcode settings. I haven't touched this setup other than to update it so it should be pretty standard.

David5i6, 2016-05-24:
I've found the solution.

Setting `export MACOSX_DEPLOYMENT_TARGET=10.11` it works like a charm.

David5i6, 2016-05-24:
After a deep diggin I've found that even python a pythonw where the `brew` ones, some *pythonic* commands like `pip` where not the `brew` versions.

I forced relinking brew python:

```sh
brew link --overwrite python 
```

And the bug dissapeared.

Thanks for the support and sorry for the inconveniences.

closed 2016-05-24 by David5i6