analogdevicesinc/libsmu #77: Python cannot find libsmu.so.0

dansteingart, 2016-06-02:
When I run the commands per the readme

```
git clone https://github.com/analogdevicesinc/libsmu.git 
mkdir libsmu/build && cd libsmu/build 
cmake -DBUILD_PYTHON=ON .. 
make
sudo make install
```

and then ask python for 

```
import pysmu
```

I get in return

```
Traceback (most recent call last):  
File "/home/dan/pithy/code/shmooze.py", line 1, in 
import pysmu
File "/usr/local/lib/python2.7/dist-packages/pysmu/__init__.py", line 13, in 
import _pysmu
ImportError: libsmu.so.0: cannot open shared object file: No such file or directory
```

I can find the file `libsmu.so.0` file with locate here

```
/usr/local/lib/libsmu.so
/usr/local/lib/libsmu.so.0
```
Where is it supposed to be?

Thanks in advance.

radhermit, 2016-06-02:
Is /usr/local/lib in /etc/ld.so.conf on the system? If it isn't, add it to the right place depending on what you're running, e.g. /etc/ld.so.conf is often auto-generated or sources files inside another directory such as /etc/ld.so.conf.d/.

Then run ldconfig as root to regenerate the /etc/ld.so.cache which should force the runtime linker to look for libraries in /usr/local/lib.

radhermit, 2016-06-02:
I forgot to ask, is this on Linux? The instructions above are Linux specific.

dansteingart, 2016-06-02:
Great, yes Ubuntu will give it a shot

dansteingart, 2016-06-02:
So that worked: I just had to run `ldconfig` and it imports fine.  Thanks!   Perhaps you want to add `sudo ldconfig` to the compilation instructions?

radhermit, 2016-06-02:
Thanks, added some Linux specific docs in 87cf549.

closed 2016-06-02 by radhermit