analogdevicesinc/libsmu #78: add a manpage for the smu cli utility

radhermit, 2016-06-14:
For Linux usage, having some sort of basic man page for smu would be helpful for users familiar with the command line.

peterbuxton, 2020-01-19:
I hope this is appropriate to reply to this old thread. The subject line caught my attention as something I am looking for. I'm trying to use Alice with the ADALM1000 and have built and installed libsmu in Linux. Since I understand that libsmu is a library for communicating with the device over USB, I would like to know how to issue commands to do that, if possible (it sounds possible from the title of this thread). I don't actually have Alice working (because I don't know how, yet), but my ADALM1000 shows connected in the command lsusb, so I'd like to know how I can make it acknowledge a command that I issue. Please tell me if I have this all wrong.

peterbuxton, 2020-01-20:
Please ignore my above post. I reposted it as a new thread.