analogdevicesinc/libsmu #84: rework: questions about read_write.py example.

damercer, 2016-11-29:
In the read_write.py example in the rework branch:

Shouldn't the values printed range from 0 to 5 if the random values written range from 0 to 5?

Here are a few lines of output:
0.019126 0.261063 -0.000350 0.262910
0.019048 0.261063 -0.000272 0.262910
0.019361 0.261063 -0.000505 0.262910
0.019282 0.261063 -0.000428 0.262910
0.019673 0.261063 0.000118 0.262910
0.019673 0.261063 0.000352 0.262910
0.019048 0.261063 0.000352 0.262910
0.019282 0.261063 0.000118 0.262910
0.018579 0.261063 -0.000116 0.262910
0.018892 0.261063 -0.000116 0.262910
0.019204 0.261063 -0.000350 0.262910
0.019204 0.261063 -0.000272 0.262910
0.019361 0.261063 0.000352 0.262910
0.019282 0.261063 -0.000194 0.262910
0.019204 0.261063 -0.000272 0.262910
0.018814 0.261063 -0.000428 0.262910
0.019204 0.261063 -0.000350 0.262910
0.018579 0.261063 -0.000116 0.262910
0.018814 0.261063 -0.000038 0.262910
0.018267 0.261063 0.000040 0.262910

Also ( in Windows at least ) using ctrl-C to stop program loop leaves a Python process running that can only be stopped using the Task Manager. This process needs to be killed before the board can be found again.

radhermit, 2016-11-29:
The first item is a known bug that I think should be solved on the firmware side                                                                                                                                               
(https://github.com/analogdevicesinc/m1k-fw/issues/7).

I'll check out the examples on windows this week and try to fix things.

damercer, 2016-11-30:
I modified the read-write example to output one cycle of a sine wave each write cycle. If you look at the analog outputs with a scope the wave form looks good but only if the number of samples written is exactly 512. Try other values longer than 512 and see what I mean ( program uses numpy ).

[read-write-sinewave.zip](https://github.com/analogdevicesinc/libsmu/files/622378/read-write-sinewave.zip)

Another strange behavior is that after the program is stopped and the orphaned Python process is killed the analog outputs from the board are still producing a sine wave. Maybe this is because of the way the program is being stopped.

closed 2016-12-19 by damercer