analogdevicesinc/libsmu #85: rework: Long delay when getting samples in HI_Z mode

damercer, 2016-11-30:
Try running the attached python 2.7 program while connecting the channel inputs to GND then 2.5 V then 5 V  etc. and notice how long it takes for the displayed values to change.

[volt-meter-tool-1.2.zip](https://github.com/analogdevicesinc/libsmu/files/622344/volt-meter-tool-1.2.zip)

The longer the program runs the longer the delay gets???

What am I doing wrong?

radhermit, 2016-11-30:
Do you have multiple devices plugged in? Currently you're running in non-continuous mode which means the devices will try to resync every time your loop iterates. Right now syncing can take a significant, but seemingly non-deterministic amount of time when it's initially done (and I'd like to decrease that). I have no idea right now why that time would increase as you keep iterating, but that's probably what's happening.

If I remove the code causing the multi-device syncs, everything works as expected.

damercer, 2016-12-01:
I use only one board at a time. Maybe you should default to not doing the syncing for multiple boards when one one is attached. The software should know if only one was found?

I know one of the key motivations for reworking libsmu was to support multiple boards but .... It needs to function the same as before.

radhermit, 2016-12-01:
Syncing doesn't occur with only one board, I was mentioning syncing because I couldn't reproduce your case with one board. But I'll try again on windows if that was the issue.

damercer, 2016-12-02:
One more observation. It looks like the measurements are sitting in a long buffer because as you move the input connection you see ( after the delay ) that the values go from say 0 V for GND to the value for open ( 300 mV ) for the few seconds it takes to move the wire then to say 2.5 V. So all the data is there just sitting in a very long buffer?

And BTW it is probably best for you to try to reproduce any anomalies I find in Windows first as that is what I'm using to test. I see this problem on two different Windows systems as well.

closed 2016-12-19 by damercer