analogdevicesinc/libsmu #89: rework: New multi-run-flush example hangs

damercer, 2016-12-06:
After installing latest rework version ( X86 installer for Windows ) the new multi-run-flush example program does not run. It hangs after printing the following:

Python 2.7.11 (v2.7.11:6d1b6a68f775, Dec  5 2015, 20:32:19) [MSC v.1500 32 bit (Intel)] on win32
Type "copyright", "credits" or "license()" for more information.
 
 RESTART: C:\Documents and Settings\DELL\My Documents\RPI\ADiscovery\ALM\rework\multi-run-flush.py 
dev: 1: chan A voltage: 0

Similar behavior i.e. hang when trying any of the other examples and some variations I've created. It would seem that something has been broken ( in Windows at least ) for the last few builds of rework.

radhermit, 2016-12-08:
Thanks, I noticed that too. I think it looks like a threading issue. Do you recall which version you used before where things didn't hang? Otherwise I can probably make a decent educated guess.

damercer, 2016-12-08:
I think it might be between 763 and 764. I think the last version that I had that "worked" was down loaded 12/2.

radhermit, 2016-12-13:
Should be fixed now with a workaround for libusb issues on windows, please test. Only script that hangs on windows specifically now for me is the non-continuous read-write script which I'm looking into.

closed 2016-12-13 by radhermit