analogdevicesinc/libsmu #90: rework: Multiple Boards

damercer, 2016-12-15:
I have a program that functions properly with just one M1K attached but crashes after the first time in the loop with two M1Ks attached.

Python 2.7.11 (v2.7.11:6d1b6a68f775, Dec  5 2015, 20:32:19) [MSC v.1500 32 bit (Intel)] on win32
Type "copyright", "credits" or "license()" for more information.
> > > 
 RESTART: C:\Documents and Settings\DELL\My Documents\RPI\ADiscovery\ALM\rework\volt-meter-tool-1.2.pyw 
2031305436363037323530323130303
2.09
E
100000
loop Number  0

Traceback (most recent call last):
  File "C:\Documents and Settings\DELL\My Documents\RPI\ADiscovery\ALM\rework\volt-meter-tool-1.2.pyw", line 242, in <module>
    Analog_in()
  File "C:\Documents and Settings\DELL\My Documents\RPI\ADiscovery\ALM\rework\volt-meter-tool-1.2.pyw", line 44, in Analog_in
    session.run(512)
  File "pysmu\libsmu.pyx", line 201, in pysmu.libsmu.Session.run (C:\projects\libsmu\bindings\python\pysmu\libsmu.cpp:4279)
SessionError: failed running session stream: Input/output error

Doug

damercer, 2016-12-15:
I now have it working. It seems that you need to at least set the mode for all channels on all boards before running the session.

radhermit, 2016-12-15:
If you don't set them, all channel modes for all devices default to HI-Z on session run. You shouldn't have to explicitly set this. If you attach your script I could perhaps see what the issue is.

damercer, 2016-12-16:
After closer examination whether the program runs or crashes does not depend on how many times the mode is set or not but is indeterminate. Try running the attached program with two boards plugging in.
Use Windows. I've commented out the extra places where the Mode is set. You can try adding them back in cause sometimes it helps to get the program to run.

[dc-meter-source-tool-1.2.zip](https://github.com/analogdevicesinc/libsmu/files/656338/dc-meter-source-tool-1.2.zip)

closed 2016-12-19 by damercer