analogdevicesinc/libsmu #94: rework: program does not return in discontinuous mode

damercer, 2017-01-06:
When not in Hi-Z mode i.e using .arbitrary, repeat flag set, the program never returns from session.run() in discontinuous mode when requesting more than 2000 samples ( i.e. works for 2000 but not for 4000 ). Buffer size set to 100000. Works fine when both channels are set to Hi-Z mode.

Following functions are used:

session = Session(ignore_dataflow=True, queue_size=100000)
session.add_all()
session.configure()
devx = session.devices[0]
CHA = devx.channels['A']    # Open CHA
CHA.mode = Mode.HI_Z # Put CHA in Hi Z mode
CHB = devx.channels['B']    # Open CHB
CHB.mode = Mode.HI_Z # Put CHB in Hi Z mode
( loop goes around these functions )
session.flush()
CHA.mode = Mode.SVMI 
CHA.arbitrary(AWGBwaveform, AWGBRepeatFlag.get())
CHB.mode = Mode.SVMI 
CHB.arbitrary(AWGBwaveform, AWGBRepeatFlag.get())
session.run(SHOWsamples) 
ADsignal1 = devx.get_samples(SHOWsamples)

SHOWsamples is either 2000 or 4000.... session.run(4000) never returns. This use to work a few builds ago?

Is there something wrong with the order of doing this? If so why does it work with 2000 and not 4000?

Doug

radhermit, 2017-01-19:
Sorry for the delayed response, a few notes on your code:

- explicit add_all() and configure() calls aren't necessary as Session() initialization does them if the related parameters aren't changed from the defaults
- explicit flush() and run() calls aren't needed anymore assuming you use get_samples()
- I'd advise using session.get_samples() which handles multi-device cases transparently, otherwise device.get_samples() will still work for single device sessions, but not for multi-device sessions.

With those changes, what you're doing should work. What's probably causing the issue is you're explicitly calling run() which also gets called internally by get_samples() since some relatively recent revision.

damercer, 2017-01-19:
I removed the add_all(), configure(), flush() and run() calls ( commented them out ) and the program still hangs when the AWG output are used ( i.e. not in HI-Z mode ). So you missed the part about not being in Hi-Z mode. 

I'm still using device.get_samples() because session.get_samples() seems to return a different shaped array and I didn't feel like sorting that out at the moment.

Here is one of the simpler programs. If you run with the AWG sources turned off the program works. Turn on the sources and the program hangs,

[dc-meter-source-tool-1.2.zip](https://github.com/analogdevicesinc/libsmu/files/716996/dc-meter-source-tool-1.2.zip)

cristi-iacob, 2019-08-20:
It doesn't seem to hang anymore. It works properly on my computer. Thanks!

closed 2019-08-20 by cristi-iacob