analogdevicesinc/libsmu #95: rework: read-write.py example fails if num_samples set > 2816

damercer, 2017-02-13:
The read-write.py example in rework fails if num_samples set greater than 2816 ( 11 * 256 ).
Remove for loop that prints output values and replace with print(len(samples)) to speed up printing.
After failure board needed to be unplugged from USB port ( powered down ) before it can be found again.

This is same issue as previous one....
You need to run your tests with larger number of samples.

Thanks, Doug

claudiupop, 2017-10-12:
I tried the read-write.py example, and it works for me even with values greater than 2816. Do you still have the problem? If you do, can you give me more details about the problem?

damercer, 2017-10-19:
Try running the attached version of ALICE 1.2 source code. You need to have libsmu rework-skipmode installed with pysmu. I have changed it to work with the newly added skip mode for continuous running. In the AWG controls window selecting Sync AWG runs in discontinuous mode ( not selecting is continuous mode ). As the time/div is increased ( requesting more samples ) the program will eventually crash.

[alice-Scope-1.2.zip](https://github.com/analogdevicesinc/libsmu/files/1399788/alice-Scope-1.2.zip)

Doug

claudiupop, 2017-10-23:
I tried your program with that settings, and it works for me, even at 200 ms / div. The only problem that I encountered is that after 950 ms mark the signals are not printed anymore. I tried it on a Linux virtual machine. Could you give me more details about your system?

damercer, 2017-10-24:
Assuming that the AWG Sync option was checked it always locks-up for me not returning as if it was waiting for the buffer to fill. Maybe it is a Windows related issue.

The max record length is set to 95,000 samples so that is all the data that will be displayed.

I'm running Windows 7 Pro. 32 bit Python on a dual core 64 bit Intel processor.

I could try to write a striped down program that shows the problem more clearly.

Doug

claudiupop, 2017-10-24:
The AWG Sync option was checked. I will test it on Windows to see if I can replicate the problem.

claudiupop, 2017-10-25:
I tried it on Windows 7 with 32 bit Python and it works.  I used this to install libsmu and pysmu ( https://ci.appveyor.com/api/buildjobs/fv3g8exrd8t3egdc/artifacts/libsmu-0.9.0-ga8dab6a-setup-x86.exe ). Could you try to use this installer to install pysmu to see if you still have the problem ?

damercer, 2017-10-25:
This is very strange. It works on this ADI issued Windows 7 pro laptop 
but not on the other Windows 7 computer I do all my development on????

Any thoughts on how to track down what is different?

Doug

damercer, 2017-10-25:
I tried this on a 1.3 GHz ( quad core ) Windows 8.1 tablet computer and it is also not working in discontinuous mode . Same exact result it seems to stop as if waiting for the buffer to fill or somthing.
It also failed after awhile ( changing time base settings ) in continuous mode as well.
Do you have access to a computer with a slower processor speed to test on?

BTW the original version using master version (0.88 0.89?) of libsmu works flawlessly on the Windows 8.1 computer.

Thanks

Doug

claudiupop, 2017-10-26:
I tried it again on a Linux Virtual machine but this time I limited the CPU time to half.
The program worked fine, up until 50 ms / div and then the GUI froze. The only way to close the program was to either kill it or unplug the board. When I unplugged the board I got this error https://pastebin.com/Pym23zuW . 
It looks like it was stuck in get_samples() method, like you described. 
When I raised the CPU time and tried again the program was working fine.

It looks like it is a performance issue.

damercer, 2017-10-26:
I agree, it think there is a timing issue, maybe due to multiple threads 
needing to start and end at the right times. I'm thinking that adding 
the continuous mode circular buffer ( that is not in master ) added a 
lot of complexity and timing issues between input and output channel 
data is at the root of the problem. The reason I say this is because it 
seems to work when not sending data to the AWG outputs either one at a 
time or neither, only when sending to both.

Doug

claudiupop, 2017-10-30:
I have made a change on branch libsmu-fix-getsample (dd3b20da6ab5145ec36af0dab67a979c6458f24a). Could you try it, to see if it fixes the problem?

damercer, 2017-10-30:
I installed this fix and nothing has changed. The problem happens at exactly the same place in the same way on my development Windows 7 computer. Have not bothered to check on other Windows computers yet. I doubt anything will be different on them as well.

Thanks

Doug

claudiupop, 2017-11-01:
It looks like the libsmu-installer doesn't always install the newer libsmu.dll if you already have a version installed. 
Could you try to uninstall the previous version of libsmu/pysmu and then try again the version from libsmu-fix-getsample branch ( https://ci.appveyor.com/project/analogdevicesinc/libsmu/build/1.0.1037 ).

If you still have the problem, could you install this version (it has debug messages enabled) https://ci.appveyor.com/project/analogdevicesinc/libsmu/build/1.0.1040  and run your application from command line ( `python alice-desktop-1.2.pyw > log.txt 2>1&`) and post here the log.txt file.

Thanks

damercer, 2017-11-01:
I always do that as a general practice.

I will try the debug version.

Thanks

Doug

damercer, 2017-11-01:
I installed the Debug Release version from appveyor build 1.0.140 and when I run the program it does not find pysmu? see error message from Python shell:

>>> from pysmu import *

Traceback (most recent call last):
  File "<pyshell#0>", line 1, in <module>
    from pysmu import *
  File "C:\Python27\lib\site-packages\pysmu\__init__.py", line 6, in <module>
    from .libsmu import *
ImportError: DLL load failed: The specified module could not be found.
>>> 

I looked in the Python 27 site-packages pymu dir where libsmu.pyd resides and it is there so pysmu was installed?

Thanks

Doug

damercer, 2017-11-01:
OK I tried again and installed the Release version from 1.0.40 and it finds pysmu but my path does not have python included in it so just typing python in the command line does not work.

Python programs are launched by double clicking.

Will keep experimenting....

damercer, 2017-11-01:
OK I got the program to run ( using full path to python.exe ) but the log.txt file that was created was empty after the program hang.

Doug

claudiupop, 2017-11-01:
Try to run it without `> log.txt 2>1&` part to see if it prints any messages.

claudiupop, 2017-11-01:
My mistake, it it `> log.txt 2>&1` instead of `> log.txt 2>1&`. Now it should write the messages in the file.

damercer, 2017-11-02:
It had put the messages in a file named 1. I renamed it log.txt and attached it here:

[log.txt](https://github.com/analogdevicesinc/libsmu/files/1437767/log.txt)

Thanks

Doug

claudiupop, 2017-11-03:
There is something odd in the log, but I can't tell where the problem is. Could you try again with this version https://ci.appveyor.com/project/analogdevicesinc/libsmu/build/1.0.1041 ? 

Thanks

damercer, 2017-11-03:
Here are the results running two different ways. The first is with the time/div set such that it requests 4000 from the start ( never requesting a smaller number of samples)

[log.txt](https://github.com/analogdevicesinc/libsmu/files/1441197/log.txt)

The second way is to set the time/div such that it request only 2000 samples from the start and then after it runs properly for a second set time/div to request 4000 samples.

[log_2.txt](https://github.com/analogdevicesinc/libsmu/files/1441203/log_2.txt)

Thanks

Doug

claudiupop, 2017-11-06:
I looked over the logs. 
In the first one, for some reason no samples were written, so there were no samples to fill the buffer and so it thrown an error. 
The second one looks exactly like the one generated on my computer. From the log it looks like the program is working well.

In both cases, does the program crash/freeze. Or it only doesn't show the requested number of samples on screen and the interface is responsive ?

I think that the problem is somewhere in the writing process.

Could you test it again in both ways with this version? ( https://ci.appveyor.com/project/analogdevicesinc/libsmu/build/1.0.1042 )

Thanks

damercer, 2017-11-06:
OK, here is the result of doing the same two tests with 1.0.1042. In the first case when the first request is for 4000 samples the program crashes that is the Python windows disappear.
[log-3.txt](https://github.com/analogdevicesinc/libsmu/files/1446388/log-3.txt)
For the second case when initially requesting only 2000 samples then switching to 4000 samples the program freezes ( not responding ) and I have to click on the X window icon in the main Python window to close the program. The initial 2000 samples is displayed but not the 4000.
[log.txt](https://github.com/analogdevicesinc/libsmu/files/1446401/log.txt)
The log file is very large because I waited a few seconds before killing off the program.

Thanks

Doug

damercer, 2017-11-06:
One further observation on the second test case. It looks like it gets one set of 4000 samples after switching from 2000 samples but freezes on the next ( second request?).

Doug

damercer, 2017-11-28:
Just checking. Any progress on getting to the bottom of this issue with Write to both channels in discont. mode?

claudiupop, 2017-11-29:
There is no progress on this issue. I can't reproduce the problem. I tried it on two other laptops (windows 7 and 10) and a raspberry pi, and it was working fine.

damercer, 2017-11-29:
Well, what can we do to get this fixed so I can produce better versions of ALICE for the schools who are using this in classes? 

Doug

rgetz, 2017-11-29:
Doug:

I understand your frustration - but we can't fix what we can't reproduce.

@claudiupop What operating system are you testing on? What version of Python are you testing on? We can be a little more descriptive - so Doug can point us in the right direction so we can reproduce what he is seeing. If it is a performance issue - that's a bug. We can't assume that everyone is going to have the same speed CPU as you develop on.

@damercer : Can you detail your environment a bit, so we can get something closer?

-Robin

damercer, 2017-11-29:
I think I've already done some of that. I copied the debug log report here and it narrowed the problem down to when you are also writing to both channels while reading from both channels. Writing to just A or just B works fine but not both. 

I did a little more testing this morning and if I add some delay in the loop ( 20 mS up the time for the number of samples requested the program returns good data up about 7500 to 7700 samples, any more and it hangs. This is an increase over the about 3000 samples it hangs on with no added delay. Adding 10 mS of delay extended the number of samples a little but no amount of delay gets me past 7700 samples.

I also tested different USB ports on the computer. I normally use the ones on the front panel but I tried the ones on the back and also with an external USB hub and there was no change.

I'm running plain Windows 7 and 8 setups without any of the software development tools you all might have installed so I might be missing something you have.

rgetz, 2017-11-29:
Doug:

I appreciate the patience - what @claudiupop said is trying the same thing - doesn't crash for us. That is the difficulty.

How fast (MHz) is your PC where you can replicate this? What chipset it is using? You said Win 7 and Win 8 - you are make it fail on more than one machine?

Sorry if I missed some of the description above,

damercer, 2017-11-30:
The Windows 7 is a high end Dell desktop like the ones ADI supplied the circuit designers ( but they run red hat linux of course ) dual core 64 bit but I have 32 bit Python 2.7.12 ( .13 will not install ) . I install the 32 bit libsmu to use with it. The Windows 8 is a 1.2 GHz quad 64 bit core  tablet running 32 bit OS so it to has  32 bit Python 2.7.12 and 32 bit libsmu. I do this to be consistent across the two machines and figure it is the least common denominator so should work for all likely users out there. The third computer I've tested is this rather old ADI Dell laptop. I think it is a 64 bit dual core. It ( like the ones @claudiupop is using ) seems to work.

The desktop is probably at the highest speed end of the spectrum while the tablet is at the other extreme. The fact that adding delay ( increasing time between sample requests ) extended the time before the program hangs would indicate to me that some low level routine needs more time to complete. I'd have to go back and do more detailed checking on the slower tablet computer to see how many samples it can write before it hangs.

Doug

damercer, 2017-12-01:
OK, I reconfirmed that I'm seeing the same problem on the Windows 8 tablet computer. It hangs for 7500 samples but not 4500 samples. Didn't take the time to narrow down the range but it looks a lot like what I see with the desktop computer. This would again point to a timing issue writing buffers to both channels at the same time.

By the way this is using Rev F hardware running 2.15 firmware.

Doug

damercer, 2017-12-01:
Here is some additional info that might help. When I set the number of samples to values slightly more than 7500 but less than 10000 the program does not hang immediately on the first request. It might take 20 or 30 seconds before it hangs. The closer to 10000 samples ( 100 mS ) the sooner it hangs. When it hangs the output stops. For greater than 10000 samples where it hangs immediately the output continues but it seems to be just looping on the last packet that was sent ( 256 samples maybe ) which is probably what's in the buffer in the M1K itself. Output continues until power is removed from the board.

Based on this one might guess that a USB handshake was dropped or missed.

Just trying to give you as much guidance to track down issue as I can.

Doug

claudiupop, 2017-12-04:
Now I am using windows 7 64bit with 32bit python 2.7.14. 
I will test it again using 2.7.12 version and the new information to see if I can find the issue.

damercer, 2017-12-04:
Great, If you still cant' get it to reliably hang on one of your computers maybe I can help remotely. In a way you sort of have access to my two computers that show the problem. I understand that it is not ideal but I can try running any testing, debug you might do otherwise and report back what happens.

Thanks

Doug

claudiupop, 2017-12-05:
Could you tell me please, where exactly did you put the delay that made the application work a little longer ?

damercer, 2017-12-05:
Here is a short section from the analog_time_in(): subroutine where I added lines to test how many samples makes the program hang and add a delay based on number of samples.

if AWGSync.get() > 0: # awg syn flag set so run in discontinuous mode
            if discontloop > 0:
                session.flush()
            else:
                discontloop = 1
            # print "just before awg enable"
            BAWGEnab()
            # print "just before get samples"
            if SHOWsamples > 4000:
                SHOWsamples = 7500 # Change this number to test when program hangs
            ADsignal1 = devx.get_samples(SHOWsamples) # get samples for both channel A and B
            # ADsignal1 = session.get_samples(SHOWsamples) # get samples for both channel A and B
            # waite to finish then return to open termination
            devx.ctrl_transfer( 0x40, 0x51, 32, 0, 0, 0, 100) # set CHA 2.5 V switch to open
            devx.ctrl_transfer( 0x40, 0x51, 33, 0, 0, 0, 100) # set CHA GND switch to open
            devx.ctrl_transfer( 0x40, 0x51, 37, 0, 0, 0, 100) # set CHB 2.5 V switch to open
            devx.ctrl_transfer( 0x40, 0x51, 38, 0, 0, 0, 100) # set CHB GND switch to open
            if AWGAIOMode.get() > 0: # Split Input / Output mode
                devx.ctrl_transfer( 0x40, 0x51, 34, 0, 0, 0, 100) # open voltage sense loop
            if AWGBIOMode.get() > 0: # Split Input / Output mode
                devx.ctrl_transfer( 0x40, 0x51, 39, 0, 0, 0, 100) # open voltage sense loop
            time.sleep(1000.0/SHOWsamples) # added delay based on number of samples

I have been bringing this version of ALICE (1.2) up to the same level of changes and features in the current released version (1.1).

Here is the 1.2 version source code. 
[alice-desktop-1.2.zip](https://github.com/analogdevicesinc/libsmu/files/1531330/alice-desktop-1.2.zip)

Doug

claudiupop, 2017-12-07:
Do you still have a problem with "read-write.py example in rework fails if num_samples set greater than 2816 ( 11 * 256 )." 
Because if I run the read-write.py example , let it run for a second or two, then I close it, after repeating this a few times it will freeze. (in the example it requests 10001 samples). It doesn't freeze every time. 

Could you do some tests with these versions (https://ci.appveyor.com/project/analogdevicesinc/libsmu/build/1.0.1084) 
(https://ci.appveyor.com/project/analogdevicesinc/libsmu/build/1.0.1085)
 and post the logs ? 

Thanks

damercer, 2017-12-07:
Hi:
I installed version 1.0.1085 and with no extra delay added the program seems to now work all the way up to 100,000 samples ( 1 Sec ) in discontinuous mode. I'll do some more testing to make sure but I think this latest change in 1.0.1085 may be the answer.

Doug

damercer, 2017-12-07:
I tested version 1.0.1085 on the Windows 8 tablet computer that was also showing the problem and the program seems to now work all the way up to 100,000 samples ( 1 Sec ) in discontinuous mode. I'll continue do some more testing to make sure every thing is working but so far this latest change in 1.0.1085 may be the answer.

Thanks

Doug

damercer, 2017-12-07:
Hi again:
It looks like version 1.0.1085 still has some debug messages turned on. I made a Windows executable with py2exe and when I run from that a bunch of messages scroll by in the Windows console screen. You'll want to turn that stuff back off in the main version of rework. Just a reminder.

Doug

rgetz, 2017-12-07:
Just so I know - what gittag is version 1.0.1085?

AlexandraTrifan, 2017-12-07:
This is the commit that triggered build 1.0.1085: 6f33b5a83124e42d4336add4924681f3212e11f6

rgetz, 2017-12-07:
Are there any side effects of this? (reducing transfers to 2 - this should decrease the number of USB packets?)

claudiupop, 2017-12-11:
Reducing transfers to 2 increases the number of packets per transfer and also the number of samples per transfer.

claudiupop, 2017-12-11:
I have merged the fix into rework branch. I also merged the rework-LED branch into rework which adds to libsmu a method to control the LEDs  and changed the pysmu method set_led ( now it have a different signature) to use the libsmu implementation.

closed 2019-07-03 by AlexandraTrifan