analogdevicesinc/libsmu #100: Rework: Two remaining major issues

damercer, 2017-03-23:
Hi:
There are two major issues remaining that are preventing me from switching over to the rework version of libsmu ( i.e. making my Python program work properly ).

The first was posted as issue number 94 and again as number 95. The session.get_samples() function ( discontinuous mode ) crashes if you request more than 11*256 samples.

The second is a request to add an option to continuous mode session.read() to optional start new read from where you left off ( current way it works ) in the circular buffer or optionally start new read from current write position in buffer effectively skipping over samples. This new mode is needed when the program can't process the data stream in real time ( which is the case for most Python programs that need to do any amount of signal processing ).

There may be other issues that have yet to be uncovered but these two are holding up the show at the moment.

Thanks

Doug

claudiupop, 2017-10-18:
I have made a commit (on branch rework-skipmode https://github.com/analogdevicesinc/libsmu/commit/a8dab6aaaffdce383b1fd7d6bd8cb8dcfb201aa7)   regarding your request. Is that the expected behaviour?

damercer, 2017-10-18:
> I have made a commit (on branch rework-skipmode a8dab6a

It sounds right but it will need to be tested. When will the additions 
appear in rework itself? Or do I test from the other branch?

It may take some time before I can get the now very old version of the 
program back in sync with the most current version. It's been months 
since I stopped working with libsmu rework.

Thanks

Doug

claudiupop, 2017-10-19:
Could you test it from the other branch? And if it works well I will merge it into rework branch.

damercer, 2017-10-19:
The new feature added to ship ahead in continuous mode seems to work properly. However the first item in the original issue: "The first was posted as issue number 94 and again as number 95. The session.get_samples() function ( discontinuous mode ) **crashes if you request more than 11*256 samples.**"
is still broken.

Based on the new skip mode when running continuous, I will proceed with bringing ALICE 1.2 for libsmu-rework up to the same revisions as the current ALICE 1.1 for libsmu-master.

Hopefully you can also fix discontinuous mode.

Thanks

Doug

kellyjp, 2018-01-04:
Now that rework has been merged into master, ALICE 1.1 no longer functions with libsmu-master (at least on Mac OS X).  Could either:
* A version of ALICE that works with the current libsmu-master be made available for download somewhere (ALICE 1.2 ???) OR
* The documentation at https://wiki.analog.com/university/tools/m1k/alice/desk-top-users-guide be updated to specify which tag/branch for libsmu should be used with ALICE 1.1 (the version made available for download on the wiki page) as the current instructions point to the master branch which now appears incompatible with that version.

damercer, 2018-01-04:
That is in the works.

An obvious question is why did you re-compile libsmu if it was working 
already?

damercer, 2018-01-04:
I'm still seeing issues with the reworked libsmu ( see issue #109) so a "reworked ALICE 1.2 still hangs randomly sometimes (on my development system anyway) and I'm reluctant to release it to the wild and get a bunch of users angry.

Not sure why rework was moved to master without my knowledge ( approval ) as I'm a major user of libsmu.

Doug

damercer, 2018-01-04:
Build 1.0.123 on AppVeyor (master b6bc6cba) seems to work for ALICE 1.1. Not sure how to get to the source code for compiling on Mac.

Doug

rgetz, 2018-01-04:
I think https://github.com/{username}/{projectname}/archive/{sha}.zip should work.

I tried : 
https://github.com/analogdevicesinc/libsmu/archive/b6bc6cb.zip

and it downloaded something for me (I never checked the contents).

-Robin

kellyjp, 2018-01-05:
Thank you all for sorting this out.

I can confirm commit b6bc6cb compiles on macOS 10.13.2 and ALICE 1.1 now loads successfully on macOS using a libsmu built from that commit.  After installing the prerequisites with homebrew the following steps were required:

```
git clone https://github.com/analogdevicesinc/libsmu.git
cd libsmu
git checkout b6bc6cb
mkdir build; cd build
cmake -DBUILD_PYTHON=ON -DCMAKE_INSTALL_PREFIX=/usr/local ..
make
sudo make install
python2 <path to alice>/alice-desktop-1.1.pyw
```

James

closed 2019-07-03 by AlexandraTrifan