analogdevicesinc/libsmu #103: cannot compile on OS X

YY1205, 2017-08-30:
ld: warning: ignoring file /usr/local/Cellar/libusb/1.0.21/lib/libusb-1.0.dylib, file was built for x86_64 which is not the architecture being linked (i386): /usr/local/Cellar/libusb/1.0.21/lib/libusb-1.0.dylib
Undefined symbols for architecture i386:

claudiupop, 2017-08-30:
try to use this command to install libusb 
`brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/cb21e9ac30e47e3ea161c4247a6397967f83c83f/Formula/libusb.rb --universal` and it should work.

YY1205, 2017-09-08:
I used the command but still got the same error

ld: symbol(s) not found for architecture i386
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/libsmu.0.8.9.dylib] Error 1
make[1]: *** [src/CMakeFiles/smu.dir/all] Error 2
make: *** [all] Error 2

claudiupop, 2017-09-13:
with the latest version of libsmu, it should work if you follow the steps from the readme file.

commodo, 2019-01-07:
Closing since it's outdated.
There's also duplicate issue https://github.com/analogdevicesinc/libsmu/issues/105

closed 2019-01-07 by commodo