analogdevicesinc/libsmu #105: Cannot compile on OS X

cradhakrish, 2017-09-04:
ld: symbol(s) not found for architecture i386
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/libsmu.0.8.9.dylib] Error 1
make[1]: *** [src/CMakeFiles/smu.dir/all] Error 2
make: *** [all] Error 2

claudiupop, 2017-09-05:
install libusb with this command and it should work 
`brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/cb21e9ac30e47e3ea161c4247a6397967f83c83f/Formula/libusb.rb --universal`

cradhakrish, 2017-09-05:
Thanks! that did the first step. Could you please help with this:

/Users/uofiece/libsmu/bindings/python/src/pysmu.cpp:27:10: fatal error: 
      'cstdint' file not found
#include <cstdint>
         ^
1 error generated.
error: command '/usr/bin/clang' failed with exit status 1
make[2]: *** [bindings/python/timestamp] Error 1
make[1]: *** [bindings/python/CMakeFiles/python.dir/all] Error 2
make: *** [all] Error 2

claudiupop, 2017-09-06:
please take a look at https://github.com/analogdevicesinc/libsmu/issues/76 and see if that can help.

sfeng15, 2017-09-06:
Hi, I have encountered and issue when typing ''sudo make install''. The error message is attached. Thanks.
[image]

cradhakrish, 2017-09-07:
I tried this but got the same error. Will the version of Python matter? I am doing this on 2.7.13. When I used another machine with 2.7.12 the process went through fine.

claudiupop, 2017-09-07:
sfeng15: when you do cmake try something like `cmake -DCMAKE_INSTALL_PREFIX=/usr/local/ ..` and then `sudo make install`

claudiupop, 2017-09-07:
cradhakrish: I can't tell you if the version of python matter. Did you installed python via homebrew ?

cradhakrish, 2017-09-09:
Yes I installed python using homebrew and it installs version 2.7.13. Is there way to install 2.7.12 using homebrew

cradhakrish, 2017-09-11:
I noticed that cmake finds python at the location below:
-- Detecting C compile features - done
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.21
-- Found PythonLibs: /usr/lib/libpython2.7.dylib (found version "2.7.10") 
-- Looking for getopt

is this correct - I tried to link to the version installed by homebrew but I still get the same path.

claudiupop, 2017-09-15:
It is ok to find the same path. I tried to install it and i don't get that error. Could you tell me what clang version do you have? ( `clang --version` )  I have "Apple LLVM version 7.0.2 (clang-700.1.81)"  running on OS X 10.10.5. What steps did you make to get that error ?

YY1205, 2017-09-18:
I got the same error. This is my version:

clang --version
Apple LLVM version 8.1.0 (clang-802.0.42)
Target: x86_64-apple-darwin16.7.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin

cradhakrish, 2017-09-18:
I was able to build libsmu. I get the following warnings:

warning: no files found matching '*.[ch]' under directory 'src'
warning: no previously-included files matching 'CMakeLists.txt' found anywhere in distribution
warning: no previously-included files matching '*.pyc' found anywhere in distribution
warning: no previously-included files matching '*.pyo' found anywhere in distribution
warning: no previously-included files matching '__pycache__' found anywhere in distribution
writing manifest file 'pysmu.egg-info/SOURCES.txt'

Is that okay? 

Also, I had to set 

-DCMAKE_INSTALL_PREFIX=/usr/local/

dNechita, 2017-09-21:
Hi,

We will look into as soon as Claudiu returns from vacation, which is next week.

claudiupop, 2017-10-04:
@cradhakrish Those warning shouldn't be a problem.  What steps have you done to get past that error ?

commodo, 2019-01-07:
Closing since outdated.
If it's still an issue, feel free to re-open.

closed 2019-01-07 by commodo