analogdevicesinc/libsmu #108: Using py2exe with rework libsmu?

damercer, 2017-11-29:
Building a Windows executable using py2exe does not work with the rework version of Libsmu. It works fine for master version of libsmu and iiolib. 

For master version just doing this works:

from distutils.core import setup
import py2exe
import numpy

setup(console=['alice-desktop-1.1.pyw'])

But for rework I had to add line to import pysmu ( does not find Enum otherwise ):

from distutils.core import setup
import py2exe
from pysmu import *
import numpy

setup(console=['alice-desktop-1.2.pyw'])

py2exe runs and makes the dist directory with the alice-desktop-1.2.exe but it does not run and does not print any errors in the console?

Thanks for looking into this.

Doug

damercer, 2017-11-29:
If you run in admin command windows you actually get the following error message:

C:\Users\Doug\My Documents\RPI\ADiscovery\ALM\ALM Software\M1K\dist>alice-deskto
p-1.2.exe
Traceback (most recent call last):
  File "alice-desktop-1.2.pyw", line 23, in <module>
  File "pysmu\__init__.pyc", line 6, in <module>
  File "pysmu\libsmu.pyc", line 12, in <module>
  File "pysmu\libsmu.pyc", line 10, in __load
  File "pysmu\libsmu.pyx", line 18, in init pysmu.libsmu
ImportError: No module named _vendor.enum

C:\Users\Doug\My Documents\RPI\ADiscovery\ALM\ALM Software\M1K\dist>

Which seems to indicate that it did not copy Emun package into distribution. Probably because Enum is not installed in the right place.

Please advise.

Doug

claudiupop, 2017-12-04:
It looks like py2exe doesn't find all the dependent modules by it self. You have to manually tell it what modules to bundle. With this setup.py it worked for me: 

from distutils.core import setup
import py2exe
from pysmu import *
import numpy

setup(console=['alice-desktop-1.2.pyw'],
      options={ "py2exe":{
        "includes":["pysmu._vendor.enum","pysmu.utils"]
        }}
)

damercer, 2017-12-04:
Thanks. That works.

Doug

closed 2017-12-04 by damercer