analogdevicesinc/libsmu #109: Gracefully pausing, stoping, restarting continious sessions in Rework?

damercer, 2017-12-13:
What is the recommended procedure for starting, pausing, stopping restarting continuous sessions in rework?

I'm having problems where the program hangs ( i.e. does not return to Python level ) sometimes after issuing a session.end().

I start the session by doing something like this:
    if not session.continuous:
        session.flush()
        session.start(0)
        print "starting session"
        time.sleep(0.01)

To stop, pause session or exit the Python program all together I do this:

    if session.continuous:
        print "ending session"
        session.end() # end continuous session mode

Most of the time but not always the program hangs and does not return to Python from the session.end().

Any advice on how to do this would be very helpful.

Here is a short example of what I'm seeing:

[strip-chart-tool-1.2.zip](https://github.com/analogdevicesinc/libsmu/files/1556209/strip-chart-tool-1.2.zip)

If you run the program and hit the Run button to start a session and take data then hit the Stop button to pause the program all is fine because I have not done the session,end() yet. Then hit the Exit button where I do the session.end() and the program hangs.

Thanks

Doug

claudiupop, 2017-12-18:
Calling end() method to pause / stop a continuous session is the right way. 

I couldn't make the program to hang when I followed those steps. 
Instead if I repeatedly hit the Run button a few times it will eventually crash.

damercer, 2017-12-18:
That program I think only hangs consistently when you hit the Exit button ( far right on screen ). Here is another simple program I'm trying to rewrite for rework / continuous mode. It is supposed to control the DC outputs of the two channels and display the DC voltages and currents.  If you turn on CHA and CHB and hit run it works once but if you then stop ( pause ) and hit run again it hangs ( at the session.end()? ). There are various lines in the start / stop and analog_in functions commented out in my attempts to get this to work. What is strange is that I think I'm doing the same things that I to in the BIG ALICE desktop program which I think is working ( but can hang at times and seems sensitive to a short delay of 10 to 20 mSec I had to add in one spot when ending / restarting ).

[dc-meter-source-tool-cont-1.2.zip](https://github.com/analogdevicesinc/libsmu/files/1568208/dc-meter-source-tool-cont-1.2.zip)
 
Thanks

Doug

claudiupop, 2017-12-19:
I confirm that this program hangs after first start / stop cycle. But that's because in BStop() function you call session.flush() method. 

When you start a session with start() the only allowed methods to call while the session is active, are session.cancel() and session.end(). 

I commented the session.flush() line and uncommented the session.end() line  in BStop function and it works like it should.

damercer, 2017-12-19:
Ok, but I thought, and sort of confirmed by experiment, that in order to get the new AWG output configuration. changes to amplitude, frequency wave shape etc, I needed the session.flush()???
Could you explain in more detail what the functions do and when to use them please?
The comments in the source code is not sufficient.

Thanks

Doug

damercer, 2017-12-19:
I'm still seeing the program hang in BStop() with the following:

def BStop():
    global RUNstatus, session, CHA, CHB
    
    RUNstatus.set(0)
    # print "Stoping continuous mode"
    # session.cancel() # cancel continuous session mode while paused
    if session.continuous:
        CHA.mode = Mode.HI_Z # Put CHA in Hi Z mode
        CHB.mode = Mode.HI_Z # Put CHB in Hi Z mode
        CHA.constant(0.0)
        CHB.constant(0.0)
        #print "Is Continuous? ", session.continuous
        #session.flush()
        #session.end()
        time.sleep(0.02)
        print "Stoping continuous mode"
        #print "Is Continuous? ", session.continuous
    else:
        session.cancel()
    
It is somewhat random but if you do Run / Stop a few times it eventually hangs.
I also tried it with the session.end() included and it works a for a few more Run / Stop cycles but still eventually hangs? Could this be another timing issue?

Doug

claudiupop, 2017-12-21:
When you use a continuous session, you have to start the session with start(0). 
While the session is running you can read/write samples from devices. While a continuous session is running you can only call session.cancel() and session.end(). Calling other methods on a continuous session will generate a dead-lock. 
When you are done with the session(or when you want to pause it), you have to call the session.end() method to cancel all the running transfers and to close all the threads.
You can use the flush() method if you want to empty internal read and write buffers of the connected devices.

When you change the signal you need to write the new signal to the device. If you don't see the new signal it could be because there are queued samples in the internal buffers, after sometime you should see the new signal.

claudiupop, 2017-12-21:
With that code it is expected that at some time to hang because you are not ending the running session and some threads remain active. 

If it still hangs with the session.end() (on my computer if i call the end method in the BStop() it works well) in the BStop() function that could be a bug.

damercer, 2017-12-21:
I put the session.end() back into the stop function and it still hangs if I've configured the AWG output to make an output voltage.

So that leads me to a few questions about your above explanation on what is allowed when. Are you allowed to change the AWG channels in the middle of a continuous session? I first start the session with session.start(0) then do things like CHA.mode = Mode.HI_Z or CHA.mode.SVMI etc. then set the output value with CHA.constant(some_voltage).

Should I be only using these functions between session.end() and session.start(0)?

By the way it would be good if you could write down explanations, like the one above, in a Doc file and put it either on the top of the libsmu github section or in a documentation / user guide directory. Don't expect people to read through these issues to learn how to program the M1K.

It might not have been make clear but the indented User audience for writing programs for M1K is High School students and teachers who, you have to assume, have little if any formal computer science or programing training.

Thanks

Doug

damercer, 2018-01-04:
This issue is still not resolved.
Now running build version 1.0.1117, master b05a69fb.
Program still very often hangs at session.end() in the following function when updating AWG channels.

```
def UpdateAwgCont():
    global session, CHA, CHB, RUNstatus
    # if running and in continuous streaming mode temp stop, flush buffer and restart to change AWG settings
    if RUNstatus.get() == 1:
        if session.continuous:
            print "About to chang awg and Stopping continuous mode"
            session.end()
            time.sleep(0.02) # wait awhile here for some reason
            print session.continuous
            if not session.continuous:
                EnabAwg() # set-up new AWG settings
                print "Just changed awg and Starting continuous mode"
                time.sleep(0.02)
                session.start(0)
                time.sleep(0.02)
#
```

To reproduce hit run and turn on Channel A. The using mouse wheel ( running Windows ) change / increment the channel A voltage setting. It might work once or twice and then it will eventually hang.

Best to run from Python editor so you can see the print messages just before / after the session.end().

Doug
[dc-meter-source-tool-cont-1.2.zip](https://github.com/analogdevicesinc/libsmu/files/1604425/dc-meter-source-tool-cont-1.2.zip)

claudiupop, 2018-01-09:
I will make a file with all the explanations. 

You can change the mode of a channel( CHA.mode = Mode.HiZ ) only when the session is not running, but you can set the output value ( CHA.constant(voltage) ) while the session is active.

I tried this program and it works, but if I remove those sleeps it will eventually freeze.
What I noticed is that if I slowly increment the value (every 2-4 seconds) it works (without sleeps) but if I increment the value quickly it will freeze.

I am looking into this to see why it behaves like this.

damercer, 2018-01-09:
I think this may be a timing issue ( sort of like the other issue with 
the number of samples read? ) because I needed to add the delays to make 
it work better. The ones in the program were chosen to make it work at 
least some of the time on my system. In the bigger ALICE 1.2 program I 
also had to add a delay and it only worked better for a small range of 
delay values. Sometimes longer delays actually made it worse.

It might also be related to when the session.end() comes with respect to 
some timing in the continuous streaming threads? Like the process does 
not respond right away and something misses timing.

Hope this is easy to track down.

Thanks

Doug

damercer, 2018-01-09:
Here is some additional info that might help. In the program I attached earlier the session is configured by:
session = Session(ignore_dataflow=True, queue_size=10000)
If I change the queue size to 2000 it hangs on the first attempt to end the session. If I increase the queue size to 100000 it works much longer before hanging.

Doug

claudiupop, 2018-01-12:
I made a fix, could you try to see if it works? It works on my laptop, without any delays in your program.  https://ci.appveyor.com/project/analogdevicesinc/libsmu/build/1.0.1119 

One thing that will help, is to change your program a little. Don't end the session when you just write a new voltage/current, only when you change the mode of a channel. 

Thanks

damercer, 2018-01-12:
That version seems to work much better. I removed any and all delays and the program didn't hang after many start / stop / changes to AWG etc.

I'm still in the process of testing the main ALICE 1.2 for reworked libsmu. I'll post anything else I find.

Thanks

Doug

cristi-iacob, 2019-08-20:
This bug was solved in one of the previous versions. It works properly now.

closed 2019-08-20 by cristi-iacob

jmball, 2021-03-27:
I think I'm still seeing this issue (or something related) with python v3.7.10 and libsmu v1.0.3 (installed from conda-forge) on Windows 10.

I'm finding that calling `session.end()` after doing a write-read cycle in continuous mode always hangs indefinitely. This is a minimal example that shows the problem:

`import pysmu

session = pysmu.Session()
dev = session.devices[0]

# set up channels
chan_a = dev.channels['A']
chan_b = dev.channels['B']
chan_a.mode = pysmu.Mode.SVMI
chan_b.mode = pysmu.Mode.HI_Z

num_samples = 1

# start continuous mode
print("Starting continuous mode")
session.start(0)

# write-read data
chan_a.write([1] * num_samples, -1)
data = chan_a.read(num_samples, -1)
print(data)

# end continuous mode
print("Ending continuous mode...")
session.end()
print("Continuous mode ended")`

The program hangs after `print("Ending continuous mode...")`. Is this a bug or am I doing something wrong?

damercer, 2021-03-29:
This a very old closed issue. If you want to get the attention of the programmers who support Libsmu then open a new issue and point to this one as a reference.

damercer, 2021-03-29:
By the way, I just tested your script with the current released version of libsmu (that I use) in both Python 2.7 and 3.7 and it works as expected. That is it executes the last print statement. I changed the number of samples to 10 but it worked the same for 1.

Starting continuous mode
[(1.0145008563995361, -0.016219930723309517), (1.0145008563995361, -0.016188615933060646), (1.0145790576934814, -0.01603986695408821), (1.0145008563995361, -0.016071181744337082), (1.0144225358963013, -0.016219930723309517), (1.0144225358963013, -0.016219930723309517), (1.0145790576934814, -0.016259076073765755), (1.0145008563995361, -0.016290390864014626), (1.0145008563995361, -0.016219930723309517), (1.0145008563995361, -0.016266904771327972)]
Ending continuous mode...
Continuous mode ended
>>>

jmball, 2021-03-29:
Ok, I'll open a new issue. Thanks for giving the script a try.