analogdevicesinc/libsmu #110: Power needs to be cycled after Write Calibration

damercer, 2018-01-13:
Hi:
I don't know if this is related to the new reworked libsmu and/or new firmware.

I'm running firmware 2.16 on Rev F hardware.

After writing new calibration values to the board in Python ( devx.write_calibration(filename) ) the outputs of the AWG are all messed up and you need to cycle power to the board to get it working again. After cycling the power the board seems to be properly calibrated.

This worked before in old master libsmu with new firmware.

Is there a way to issue a power-on-reset to the board from software? If there isn't I think that might be a useful thing to have.

Thanks

Doug

damercer, 2018-01-13:
I think it might be a problem with 2.16 firmware.

I have a Rev E board with 2.09 loaded on it and it works with either version of libsmu.

Doug

rgetz, 2018-01-13:
Yeah, it sounds like a firmware issue.

dbogdan, 2018-01-13:
We are aware of this issue but we couldn't find the cause. I've opened an issue regarding this on the m1k-fw repository (https://github.com/analogdevicesinc/m1k-fw/issues/15) so we can close this one.

closed 2018-01-13 by dbogdan