analogdevicesinc/libsmu #113: `smu -p` doesn't display hotplug attach events on Ubuntu 17.10

itdaniher, 2018-02-01:
expected result: `smu -p` shows all plug/unplug events, as per my understanding of smu.cpp source code.

observed result: only the first detach event is printed, all attach events, and all but the first detach event are ignored:

```
ubuntu:~/Pixelpulse2/libsmu $ sudo smu -p
waiting for hotplug events...
removed device: ADALM1000: serial 2031315434393235313430333430313: fw 023314a*: hw 0
```

is all I get starting with the device attached, removing it, and reinserting it.

No hotplug events are logged to stdout if the device is not attached when `smu -p` is started.

radhermit, 2018-02-06:
What version of libusb does Ubuntu 17.10 come with? If it comes with 1.0.21, see https://github.com/analogdevicesinc/libsmu/issues/97.

Something obviously changed on the libusb side between versions breaking how hotplugging is implemented.

itdaniher, 2018-02-07:
"excellent"

closed 2018-02-07 by itdaniher