analogdevicesinc/libsmu #114: Incompatability with ALICE

rgw3d, 2018-02-05:
With a fresh install of libsmu on Ubuntu 16.04, I am unable to get [ALICE ](https://wiki.analog.com/university/tools/m1k/alice/desk-top-users-guide) to work. PixelPulse2 works perfectly. Firmware of my ADALM1000 is up to 2.06.

Specifically, [this is](https://wiki.analog.com/_media/university/tools/m1k/alice/alice-1.1.zip) the download of Alice1.1. In the `alice-desktop-1.1.pyw` it seems to be calling either old python bindings. Executing `./alice-desktop-1.1.pyw` just doesn't work. 

Everything is being run with python2.7

I am unsure if this is the right place to post this issue, but figured I would start here.

claudiupop, 2018-02-05:
ALICE 1.1 doesn't work with the latest version of libsmu. To make ALICE work you have to use an older version of libsmu. You can download the source code from https://github.com/analogdevicesinc/libsmu/archive/b6bc6cb.zip or if you cloned the repository use 
`git checkout b6bc6cb`

rgw3d, 2018-02-20:
That solved it! Thank you!

closed 2018-02-20 by rgw3d