analogdevicesinc/libsmu #115: Controling voltage feadback switch in Rev F hardware changed?

damercer, 2018-02-13:
Hi:

I'm not sure how to explain this but it seems that the setting of the switch that connects the voltage input opamp to either the original single I/O pin on the connector or new input only pin on Rev F hardware has changed in libsmu 1.0.

This happens only in discontinuous mode where the AWG is restarted each time a new set of samples are requested. 

In the original version of libsmu (0.89) I could do the following before ( and after ) devx.get_samples(SHOWsamples) to open the switch and use the new input only pin:

devx.ctrl_transfer( 0x40, 0x51, 34, 0, 0, 0, 100) # open voltage sense loop CH A
devx.ctrl_transfer( 0x40, 0x51, 39, 0, 0, 0, 100) # open voltage sense loop CH B

In libsmu 1.0 this switch is being closed automatically as part of preparing channel for new sweep (get_samples). No matter where I put the ctrl_transfer the switch is never left open.

So can we **please** put this back to the way it was before to allow the option for use of the new input only pin in the Rev F hardware? Or add new mode settings for split input output option in Rev F?

Thanks

Doug

AlexandraTrifan, 2018-06-12:
Sorry about this. I will look into it and get back to you.

-Alexandra

damercer, 2018-06-20:
Hi:
I just tested the changes made in libsmu-1.0.1-g6c2ec2a-setup-x86 and they seem to fix the issue I was seeing where the split I/O switch was being closed again after each sweep in discontinuous mode.
I can (will) make a new release of ALICE when this test version makes it into Master.

Thanks
Doug

damercer, 2018-06-21:
Great, be sure to document any new modes so I can make the proper changes to my code.

Thanks
Doug