analogdevicesinc/libsmu #119: New Split I/O modes

damercer, 2018-09-13:
I realize this not fully tested but just to let you know.
The Mode.SVMI and Mode.HI_Z still work as before.
However the Mode.SVMI_SPLIT does not do the right thing. While the new input pin is properly disconnected from the output pin, the output pin no longer sources the waveform. Seems it might be still in Hi_Z configuration but the DAC is still running because there is a small feed-through of the waveform which is not there in Mode.HI_Z or Mode.HI_Z_SPLIT mode.
This is the same for both channels.
Doug

AlexandraTrifan, 2018-09-14:
I am looking into this. Not sure yet why the output no longer works.

AlexandraTrifan, 2018-09-24:
Hi,
I managed to track down one issue regarding this. Could you try this version: https://ci.appveyor.com/project/analogdevicesinc/libsmu/build/1.0.1157  ?

Thanks
-Alexandra

damercer, 2018-09-24:
Hi:
By my testing all six modes seem to be working in this version (1.0.1157).
Thanks
Doug

AlexandraTrifan, 2018-09-24:
I'll clean this and add it to master.
Thank you
-Alexandra

closed 2019-06-11 by AlexandraTrifan