analogdevicesinc/libsmu #121: Complie from source using Windows

damercer, 2018-12-17:
How do I (i.e. what do I need to have installed ) compile / build libsmu from source on Windows 7?

Thanks

Doug

AlexandraTrifan, 2018-12-17:
Hi, 

Do you need it to be compiled with MSVC or MinGW? Is this important? 
For the MinGW version:

Download the corresponding version of MSYS2 from here: http://www.msys2.org/ . 
Follow the instructions on the page linked above to install it, and update it.

You need to install a couple of packages using the MSYS2 shell, in order to build libsmu:

For **32 bit**
`pacman --noconfirm -Sy mingw-w64-i686-gcc mingw-w64-i686-cmake mingw-w64-i686-pkg-config mingw-w64-i686-python2-pip mingw-w64-i686-openblas mingw-w64-i686-lapack`

`pacman --noconfirm -Sy mingw-w64-i686-python3 mingw-w64-i686-python3-pip mingw-w64-i686-binutils`

`pacman -U --noconfirm http://repo.msys2.org/mingw/i686/mingw-w64-i686-curl-7.61.1-2-any.pkg.tar.xz`

`pacman -U --noconfirm http://repo.msys2.org/mingw/i686/mingw-w64-i686-boost-1.62.0-1-any.pkg.tar.xz `

For **64 bit**
`pacman --noconfirm -Sy mingw-w64-x86_64-gcc mingw-w64-x86_64-cmake mingw-w64-x86_64-pkg-config mingw-w64-x86_64-python2-pip mingw-w64-x86_64-openblas mingw-w64-x86_64-lapack`

`pacman --noconfirm -Sy mingw-w64-x86_64-python3 mingw-w64-x86_64-python3-pip mingw-w64-x86_64-binutils`

`pacman -U --noconfirm http://repo.msys2.org/mingw/x86_64/mingw-w64-x86_64-curl-7.61.1-2-any.pkg.tar.xz`

`pacman -U --noconfirm http://repo.msys2.org/mingw/x86_64/mingw-w64-x86_64-boost-1.62.0-1-any.pkg.tar.xz` 

Get libusb hotplug for **32 bit** from here: http://swdownloads.analog.com/cse/build/libusb-1.0-hp.7z and extract it into C:\libusb-mingw

Compile libusb hotplug for **64 bit**:
`git clone --branch=hotplug https://github.com/analogdevicesinc/libusb.git "C:\libusb-mingw64"`

`cd C:/libusb-mingw64 && ./autogen.sh && ./configure --prefix=/mingw64 --build=x86_64-w64-mingw32 --host=x86_64-w64-mingw32 && make -j4`

Build libsmu MinGW **32 bit**
>mkdir -p c:\libsmu\mingw-32
cd C:/projects/libsmu/mingw-32
cmake -G 'Unix Makefiles' -DCMAKE_INSTALL_PREFIX=/mingw32 \
						-DCMAKE_BUILD_TYPE=Release \
						-DCMAKE_C_COMPILER:FILEPATH=C:/msys64/mingw32/bin/i686-w64-mingw32-gcc.exe \
						-DCMAKE_CXX_COMPILER:FILEPATH=C:/msys64/mingw32/bin/i686-w64-mingw32-g++.exe \
						-DLIBUSB_LIBRARIES=C:/libusb-mingw/MinGW32/static/libusb-1.0.a \
						-DLIBUSB_INCLUDE_DIRS=C:/libusb-mingw/include/libusb-1.0 \
						-DBUILD_STATIC_LIB=ON \
						-DBUILD_EXAMPLES=ON \
						-DBUILD_TESTS=ON \
						-DBOOST_INCLUDE_DIR=C:/msys32/mingw32/include \
						-DBUILD_PYTHON=OFF ..
cmake --build . --config Release

Build libsmu MinGW **64 bit**
>mkdir -p c:\libsmu\mingw-64
cd C:/projects/libsmu/mingw-64
cmake -G 'Unix Makefiles' -DCMAKE_INSTALL_PREFIX=/mingw64 \
						-DCMAKE_BUILD_TYPE=Release \
						-DCMAKE_C_COMPILER:FILEPATH=C:/msys64/mingw64/bin/x86_64-w64-mingw32-gcc.exe \
						-DCMAKE_CXX_COMPILER:FILEPATH=C:/msys64/mingw64/bin/x86_64-w64-mingw32-g++.exe \
						-DLIBUSB_LIBRARIES=C:/libusb-mingw64/libusb/.libs/libusb-1.0.a \
						-DLIBUSB_INCLUDE_DIRS=C:/libusb-mingw64/libusb \
						-DBoost_USE_STATIC_LIBS=ON \
						-DBUILD_STATIC_LIB=ON \
						-DBUILD_EXAMPLES=ON \
						-DBUILD_TESTS=ON \
						-DBOOST_ROOT=C:/msys32/mingw64/include \
						-DBUILD_PYTHON=OFF .. 				
cmake --build . --config Release

Could you try these and let me know if you encounter any problems?

Thank you.

-Alexandra

damercer, 2018-12-17:
Thanks, I had already installed mingw64-gcc4.7.1 exe but I think I need to install more along with MSYS2.
I'll let you know if I encounter any difficulties along the way.

Doug

damercer, 2018-12-31:
I think I must be missing some steps.
I only want to make a 32 bit version of libsmu so I skipped the steps listed under "for 64 bit"
in a freshly opened MSYS2 MinGW 32-bit terminal screen:
I entered the four steps under the "for 32 bit", A copy of the contents of the terminal screen is here:
[msys-MINGW32-steps1-4.zip](https://github.com/analogdevicesinc/libsmu/files/2718799/msys-MINGW32-steps1-4.zip)
I have the  libusb hotplug for 32 bit from here: http://swdownloads.analog.com/cse/build/libusb-1.0-hp.7z extracted into C:\libusb-mingw.
In the pwd of the terminal I have the downloaded and extracted libsmu-master.
I next stated entering the stuff listed under Build libsmu MinGW 32 bit. And this is what I get.

 Doug@Doug-PC MINGW32 ~
$ pwd
/home/Doug

Doug@Doug-PC MINGW32 ~
$ ls
c:libsmumingw-32  cmake.txt  libsmu-master  libsmu-master.zip

Doug@Doug-PC MINGW32 ~
$ mkdir -p c:\libsmu\mingw-32

Doug@Doug-PC MINGW32 ~
$ cd C:/projects/libsmu/mingw-32
bash: cd: C:/projects/libsmu/mingw-32: No such file or directory

Doug@Doug-PC MINGW32 ~
$

So I must be missing a step somewhere?

Thanks

Doug

damercer, 2019-01-04:
I've made some small additional progress here.
I installed MinGW to C:/MinGW (all packages)
I then opened MSYS2 32 bit again and typed in the following:

Doug@Doug-PC MINGW32 ~
$ mkdir -p c:\libsmu\mingw-32

Doug@Doug-PC MINGW32 ~
$ cd C:/projects/libsmu/mingw-32

Doug@Doug-PC MINGW32 /c/projects/libsmu/mingw-32
$ cmake -G 'Unix Makefiles' -DCMAKE_INSTALL_PREFIX=/mingw32 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER:FILEPATH=C:/msys64/mingw32/bin/i686-w64-mingw32-gcc.exe -DCMAKE_CXX_COMPILER:FILEPATH=C:/msys64/mingw32/bin/i686-w64-mingw32-g++.exe -DLIBUSB_LIBRARIES=C:/libusb-mingw/MinGW32/static/libusb-1.0.a -DLIBUSB_INCLUDE_DIRS=C:/libusb-mingw/include/libusb-1.0 -DBUILD_STATIC_LIB=ON -DBUILD_EXAMPLES=ON -DBUILD_TESTS=ON -DBOOST_INCLUDE_DIR=C:/msys32/mingw32/include -DBUILD_PYTHON=OFF ..
CMake Error: Error: generator : Unix Makefiles
Does not match the generator used previously: NMake Makefiles
Either remove the CMakeCache.txt file and CMakeFiles directory or choose a different binary directory.

It now got past the step to make the project dir but the cmake step failed as you can see.

So What do I Do next?

Thanks
Doug

AlexandraTrifan, 2019-01-07:
Hi, 

It looks like the build directory contains some cache files ( CMakeCache.txt file and CMakeFiles ) which were generated on an older build using NMake Makefiles. I think you can **cd c:\libsmu\mingw-32** and remove everything in this folder, then run cmake. The new cache files will be generated using the correct cmake generator.

-Alexandra

damercer, 2019-01-07:
Sorry but still not working. 
Here is the contents of the terminal window as I was trying what you suggested after deleted CMakeCache.txt file and CMakeFiles directory from C:/projects/libsmu/mingw-32. Added a copy of CMakeLists.txt from libsmu-master archive zip file from gitub to  C:/projects/libsmu after first error.

Doug@Doug-PC MINGW32 ~
$ cd C:/projects/libsmu/mingw-32

Doug@Doug-PC MINGW32 /c/projects/libsmu/mingw-32
$ cmake -G 'Unix Makefiles' -DCMAKE_INSTALL_PREFIX=/mingw32 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER:FILEPATH=C:/msys64/mingw32/bin/i686-w64-mingw32-gcc.exe -DCMAKE_CXX_COMPILER:FILEPATH=C:/msys64/mingw32/bin/i686-w64-mingw32-g++.exe -DLIBUSB_LIBRARIES=C:/libusb-mingw/MinGW32/static/libusb-1.0.a -DLIBUSB_INCLUDE_DIRS=C:/libusb-mingw/include/libusb-1.0 -DBUILD_STATIC_LIB=ON -DBUILD_EXAMPLES=ON -DBUILD_TESTS=ON -DBOOST_INCLUDE_DIR=C:/msys32/mingw32/include -DBUILD_PYTHON=OFF ..
CMake Error: The source directory "C:/projects/libsmu" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

Doug@Doug-PC MINGW32 /c/projects/libsmu/mingw-32
$ cmake -G 'Unix Makefiles' -DCMAKE_INSTALL_PREFIX=/mingw32 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER:FILEPATH=C:/msys64/mingw32/bin/i686-w64-mingw32-gcc.exe -DCMAKE_CXX_COMPILER:FILEPATH=C:/msys64/mingw32/bin/i686-w64-mingw32-g++.exe -DLIBUSB_LIBRARIES=C:/libusb-mingw/MinGW32/static/libusb-1.0.a -DLIBUSB_INCLUDE_DIRS=C:/libusb-mingw/include/libusb-1.0 -DBUILD_STATIC_LIB=ON -DBUILD_EXAMPLES=ON -DBUILD_TESTS=ON -DBOOST_INCLUDE_DIR=C:/msys32/mingw32/include -DBUILD_PYTHON=OFF ..
CMake Error: The source directory "C:/projects/libsmu" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

Doug@Doug-PC MINGW32 /c/projects/libsmu/mingw-32
$ ls
appveyor.yml  CMakeLists.txt  dist  examples  LICENSE    src
bindings      contrib         doc   include   README.md  tests

Doug@Doug-PC MINGW32 /c/projects/libsmu/mingw-32
$ cmake -G 'Unix Makefiles' -DCMAKE_INSTALL_PREFIX=/mingw32 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER:FILEPATH=C:/msys64/mingw32/bin/i686-w64-mingw32-gcc.exe -DCMAKE_CXX_COMPILER:FILEPATH=C:/msys64/mingw32/bin/i686-w64-mingw32-g++.exe -DLIBUSB_LIBRARIES=C:/libusb-mingw/MinGW32/static/libusb-1.0.a -DLIBUSB_INCLUDE_DIRS=C:/libusb-mingw/include/libusb-1.0 -DBUILD_STATIC_LIB=ON -DBUILD_EXAMPLES=ON -DBUILD_TESTS=ON -DBOOST_INCLUDE_DIR=C:/msys32/mingw32/include -DBUILD_PYTHON=OFF ..
CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
-- Configuring incomplete, errors occurred!
See also "C:/projects/libsmu/mingw-32/CMakeFiles/CMakeOutput.log".

Doug@Doug-PC MINGW32 /c/projects/libsmu/mingw-32
$

I must not be following the right steps here.

Please could you list EXACTLY what I need to do from a blank start?

Thanks

Doug

AlexandraTrifan, 2019-01-08:
Hi,

I attached a script to setup and build libsmu. This script will clone the github repo and create the initial build structure for libsmu. 
You can run it like this from the MSYS2 MINGW32 command line: **./libsmu-build.sh < path to the location where you want to keep libsmu >**
The parameter is optional. If you leave it blank, it will automatically get your currently working directory and clone libsmu there.

You do not need to run this everytime, because it will clean and clone the repo again. Just modify the sources and after building this once, you can just **cd < path to the location where you want to keep libsmu >/libsmu/mingw-32** and run **cmake --build . --config Release**.

[libsmu-build.zip](https://github.com/analogdevicesinc/libsmu/files/2736857/libsmu-build.zip)

-Alexandra

damercer, 2019-01-08:
Great, thanks, but I seem to be missing the **git command**!

Doug@Doug-PC MINGW32 ~
$ ls
c:libsmumingw-32  libsmu-build.sh   libsmu-master.zip
cmake.txt         libsmu-build.zip  libsmu-oroginal

Doug@Doug-PC MINGW32 ~
$  ./libsmu-build.sh libsmu-master
Saving libsmu to  libsmu-master
**./libsmu-build.sh: line 6: git: command not found**
./libsmu-build.sh: line 7: cd: libsmu: No such file or directory
Building libsmu in  /home/Doug/mingw-32
CMake Error: The source directory "C:/msys64/home/Doug" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
Error: could not load cache

Doug@Doug-PC MINGW32 ~
$

How do I add git to MSYS2?

Is there anything else you can think of I might be missing? Please try to anticipate everything because of the time difference I only really get to do one experiment per day and this will take forever if I don't have everything in place fro the start.

Thanks Again:

Doug

AlexandraTrifan, 2019-01-08:
Sorry about this. 
You can install new packages with pacman. So you can run: **pacman -S git mingw-w64-i686-ninja**
Please make sure you remove the libsmu folder (if you have one) before running the build script again.

-Alexandra

damercer, 2019-01-08:
Up-date:
I added git by typing pacman -S git.
I then tried again and got the following:

Doug@Doug-PC MINGW32 ~
$  ./libsmu-build.sh libsmu-master
Saving libsmu to  libsmu-master
Cloning into 'libsmu-master/libsmu'...
remote: Enumerating objects: 11733, done.
remote: Total 11733 (delta 0), reused 0 (delta 0), pack-reused 11733
Receiving objects: 100% (11733/11733), 6.95 MiB | 4.04 MiB/s, done.
Resolving deltas: 100% (7332/7332), done.
Checking out files: 100% (95/95), done.
./libsmu-build.sh: line 7: cd: libsmu: No such file or directory
Building libsmu in  /home/Doug/mingw-32
CMake Error: The source directory "C:/msys64/home/Doug" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
Error: could not load cache

Doug@Doug-PC MINGW32 ~

Still no luck getting to a working libsmu. Git made libsmu-master/libsmu and everything seems to be below that dir. Script seems to still have trouble finding what it needs?

Doug

damercer, 2019-01-08:
Looking at the dir that it made and where it was looking I tried this:

Doug@Doug-PC MINGW32 ~
$  ./libsmu-build.sh
Saving libsmu to  /home/Doug
Cloning into '/home/Doug/libsmu'...
remote: Enumerating objects: 11733, done.
remote: Total 11733 (delta 0), reused 0 (delta 0), pack-reused 11733
Receiving objects: 100% (11733/11733), 6.95 MiB | 4.16 MiB/s, done.
Resolving deltas: 100% (7332/7332), done.
Building libsmu in  /home/Doug/libsmu/mingw-32
CMake Error at CMakeLists.txt:7 (project):
  Running

   'C:/msys64/mingw64/bin/ninja.exe' '--version'

  failed with:

   The system cannot find the file specified

-- Configuring incomplete, errors occurred!
See also "C:/msys64/home/Doug/libsmu/mingw-32/CMakeFiles/CMakeOutput.log".
The system cannot find the file specified
CMake Error: Generator: execution of make failed. Make command was: "C:/msys64/mingw64/bin/ninja.exe"

Doug@Doug-PC MINGW32 ~

I seem to be missing another dependency?

Doug

damercer, 2019-01-08:
Did it again with:

Doug@Doug-PC MINGW32 ~
$ **pacman -S git mingw-w64-i686-ninja**
warning: git-2.20.1-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (2) git-2.20.1-1  mingw-w64-i686-ninja-1.8.2-3

Total Download Size:    0.35 MiB
Total Installed Size:  31.25 MiB
Net Upgrade Size:       1.46 MiB

:: Proceed with installation? [Y/n] y
:: Retrieving packages...
 mingw-w64-i686-ninj...   360.7 KiB  1478K/s 00:00 [#####################] 100%
(2/2) checking keys in keyring                     [#####################] 100%
(2/2) checking package integrity                   [#####################] 100%
(2/2) loading package files                        [#####################] 100%
(2/2) checking for file conflicts                  [#####################] 100%
(2/2) checking available disk space                [#####################] 100%
:: Processing package changes...
(1/2) reinstalling git                             [#####################] 100%
(2/2) installing mingw-w64-i686-ninja              [#####################] 100%

Doug@Doug-PC MINGW32 ~
$  ./libsmu-build.sh
Saving libsmu to  /home/Doug
fatal: destination path '/home/Doug/libsmu' already exists and is not an empty directory.
Building libsmu in  /home/Doug/libsmu/mingw-32
CMake Error at CMakeLists.txt:7 (project):
  Running

   'C:/msys64/mingw64/bin/ninja.exe' '--version'

  failed with:

   The system cannot find the file specified

-- Configuring incomplete, errors occurred!
See also "C:/msys64/home/Doug/libsmu/mingw-32/CMakeFiles/CMakeOutput.log".
The system cannot find the file specified
CMake Error: Generator: execution of make failed. Make command was: "C:/msys64/mingw64/bin/ninja.exe"

Doug@Doug-PC MINGW32 ~

Still not getting to the end?

Doug

damercer, 2019-01-08:
It added ninja here:
Doug@Doug-PC MINGW32 ~
$ which ninja
/mingw32/bin/ninja

Doug@Doug-PC MINGW32 ~
in mingw32/bin/ninja  not 'C:/msys64/mingw64/bin/ninja.exe' '--version'

AlexandraTrifan, 2019-01-08:
Did you try to remove the "/home/Doug/libsmu" folder before running the script again? It looks like it ran the first time without finding the good ninja and used the cached files the second time you tried to run it ("fatal: destination path '/home/Doug/libsmu' already exists and is not an empty directory.").

damercer, 2019-01-08:
Yes and it gave the same result.

AlexandraTrifan, 2019-01-08:
I found the problem. I am really sorry about this. 
The build script I sent you was adapted from the 64 bit build. There is a path in there in the cmake command: " -DCMAKE_MAKE_PROGRAM:FILEPATH=C:/msys64/mingw64/bin/ninja.exe" which needs to be modified to  -DCMAKE_MAKE_PROGRAM:FILEPATH=C:/msys64/mingw32/bin/ninja.exe

damercer, 2019-01-08:
I changed the line using EMACS and tried again ( with libsmu dir deleted before hand)

Doug@Doug-PC MINGW32 ~
$  ./libsmu-build.sh
Saving libsmu to  /home/Doug
Cloning into '/home/Doug/libsmu'...
remote: Enumerating objects: 11733, done.
remote: Total 11733 (delta 0), reused 0 (delta 0), pack-reused 11733
Receiving objects: 100% (11733/11733), 6.95 MiB | 4.18 MiB/s, done.
Resolving deltas: 100% (7332/7332), done.
Building libsmu in  /home/Doug/libsmu/mingw-32
CMake Error at CMakeLists.txt:7 (project):
  Running

   'C:/msys64/mingw64/bin/ninja.exe' '--version'

  failed with:

   The system cannot find the file specified

-- Configuring incomplete, errors occurred!
See also "C:/msys64/home/Doug/libsmu/mingw-32/CMakeFiles/CMakeOutput.log".
The system cannot find the file specified
CMake Error: Generator: execution of make failed. Make command was: "C:/msys64/mingw64/bin/ninja.exe"

Doug@Doug-PC MINGW32 ~
$  ./libsmu-build.sh
Saving libsmu to  /home/Doug
Cloning into '/home/Doug/libsmu'...
remote: Enumerating objects: 11733, done.
remote: Total 11733 (delta 0), reused 0 (delta 0), pack-reused 11733
Receiving objects: 100% (11733/11733), 6.95 MiB | 4.20 MiB/s, done.
Resolving deltas: 100% (7332/7332), done.
Building libsmu in  /home/Doug/libsmu/mingw-32
-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- Check for working CXX compiler: C:/msys64/mingw32/bin/g++.exe
-- Check for working CXX compiler: C:/msys64/mingw32/bin/g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: C:/msys64/mingw32/bin/gcc.exe
-- Check for working C compiler: C:/msys64/mingw32/bin/gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Boost version: 1.69.0
-- Found OpenMP_C: -fopenmp (found version "4.5")
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Looking for getopt
-- Looking for getopt - found
-- Downloading/updating googletest
-- Configuring done
-- Generating done
-- Build files have been written to: C:/msys64/home/Doug/libsmu/mingw-32/googletest-download
[1/9] Creating directories for 'googletest-download'
[2/9] Performing download step (git clone) for 'googletest-download'
Cloning into 'googletest-src'...
Checking out files: 100% (337/337), done.
Already on 'master'
Your branch is up to date with 'origin/master'.
[3/9] No patch step for 'googletest-download'
[4/9] Performing update step for 'googletest-download'
Current branch master is up to date.
[5/9] No configure step for 'googletest-download'
[6/9] No build step for 'googletest-download'
[7/9] No install step for 'googletest-download'
[... 831 lines cut ...]
           ^~~~~~~~~~~~~~~~~~~~
../include/libsmu/libsmu.hpp:313:45: error: expected ')' before '*' token
   Device* find_existing_device(libusb_device* usb_dev);
                                             ^
In file included from ../src/signal.cpp:12:0:
../include/libsmu/libsmu.hpp:451:22: error: 'libusb_device' has not been declared
   Device(Session* s, libusb_device* usb_dev, libusb_device_handle* usb_handle,
                      ^~~~~~~~~~~~~
../include/libsmu/libsmu.hpp:451:46: error: 'libusb_device_handle' has not been declared
   Device(Session* s, libusb_device* usb_dev, libusb_device_handle* usb_handle,
                                              ^~~~~~~~~~~~~~~~~~~~
../include/libsmu/libsmu.hpp:496:3: error: 'libusb_device' does not name a type
   libusb_device* const m_usb_dev = NULL;
   ^~~~~~~~~~~~~
../include/libsmu/libsmu.hpp:498:3: error: 'libusb_device_handle' does not name a type; did you mean 'sl_device_info'?
   libusb_device_handle* m_usb = NULL;
   ^~~~~~~~~~~~~~~~~~~~
   sl_device_info
ninja: build stopped: subcommand failed.

Doug@Doug-PC MINGW32 ~
$

Did a lot more but still did not complete?
Doug

damercer, 2019-01-08:
Some of the errors seem to be related to libusb. I have C:\libusb-mingw installed to the computer by the way from here: http://swdownloads.analog.com/cse/build/libusb-1.0-hp.7z .

AlexandraTrifan, 2019-01-08:
Inside C:\libusb-mingw do you have a single folder named libusb-1.0-hp or multiple folders like: include, static, MinGW32 ? If you have a single folder, you need to copy everything inside it below C:\libusb-mingw.

In the build script, the explicit path to libusb is given, and it's something like: C:\libusb-mingw\MinGW32\static\libusb-1.0.a . So, if you have a folder libusb-1.0-hp, this path will fail.

damercer, 2019-01-08:
This is what is in C:\libusb-mingw

Doug@Doug-PC MINGW32 ~
$ cd C:\libusb-mingw

Doug@Doug-PC MINGW32 /c/libusb-mingw
$ ls
include  libusb.h  libusb-1.0  libusb-1.0.a  libusb-1.0-hp  MinGW32  static

Doug@Doug-PC MINGW32 /c/libusb-mingw

Should I delete something here?

Doug

AlexandraTrifan, 2019-01-08:
No. This is the correct structure for the folder. Let me check what other issue we might have because of libusb.

AlexandraTrifan, 2019-01-08:
Inside the mingw-32 build folder, do you have a CMakeCache.txt file and a build.ninja file? Can you archive those and send them here?

damercer, 2019-01-08:
[mingw-32.zip](https://github.com/analogdevicesinc/libsmu/files/2737277/mingw-32.zip)
Here are all the non-folder files under libsmu/mingw-32

AlexandraTrifan, 2019-01-08:
Couldn't find any differences between the log files you provided and the ones I have locally. Based on the errors, it looks to me like the libusb.h file inside C:\libusb-mingw\include\libusb-1.0 has some problems or it's not properly included in usb.hpp.  I downloaded the libusb archive again, to be 100% sure we both have the same version and it still works for me. Could you also send the C:\libusb-mingw\include\libusb-1.0\libusb.h file, just to take a look if anything doesn't look ok?

damercer, 2019-01-08:
[include.zip](https://github.com/analogdevicesinc/libsmu/files/2737788/include.zip)
There doesn't seem to be a C:\libusb-mingw\include\libusb-1.0 just C:\libusb-mingw\include\libusb
The above zip is everything below include

damercer, 2019-01-08:
Hi down loaded a new copy of libusb-1.0-hp.7z from here: http://swdownloads.analog.com/cse/build/libusb-1.0-hp.7z . And when I used WinRAR to extract the files I get these messages:

!   C:\Users\Doug\Documents\GNU MCU Build\libusb-1.0-hp.7z: Unknown method in include\libusb-1.0\libusb.h
!   C:\Users\Doug\Documents\GNU MCU Build\libusb-1.0-hp.7z: Unknown method in libusb-1.0-hp\include\libusb-1.0\libusb.h
!   C:\Users\Doug\Documents\GNU MCU Build\libusb-1.0-hp.7z: Unknown method in libusb-1.0\libusb.h
!   C:\Users\Doug\Documents\GNU MCU Build\libusb-1.0-hp.7z: Unknown method in libusb.h
!   C:\Users\Doug\Documents\GNU MCU Build\libusb-1.0-hp.7z: Unknown method in libusb-1.0-hp\MinGW32\static\libusb-1.0.a
!   C:\Users\Doug\Documents\GNU MCU Build\libusb-1.0-hp.7z: Unknown method in libusb-1.0.a
!   C:\Users\Doug\Documents\GNU MCU Build\libusb-1.0-hp.7z: Unknown method in MinGW32\static\libusb-1.0.a
!   C:\Users\Doug\Documents\GNU MCU Build\libusb-1.0-hp.7z: Unknown method in static\libusb-1.0.a
!   C:\Users\Doug\Documents\GNU MCU Build\libusb-1.0-hp.7z: Error - operation failed

Probably the cause of the missing stuff in libusb?

Doug

AlexandraTrifan, 2019-01-09:
Hi, 

The cmake command tries to point libsmu to libusb using the following path for the include files: C:\libusb-mingw\include\libusb-1.0. If the folder does not exist, or does not contain a libusb.h, then libsmu will not find some needed symbols and will fail during build. 

We extracted the libusb archive using the latest WinRAR and got the following folder structure:

```
C:/libusb-mingw
+---include
|   \---libusb-1.0
+---libusb-1.0
+---libusb-1.0-hp
|   +---include
|   |   \---libusb-1.0
|   \---MinGW32
|       \---static
+---MinGW32
|   \---static
\---static
```
Could you try to extract it using the 7-Zip software? 
Thank you. 

-Alexandra

damercer, 2019-01-09:
I don't think I have the 7-Zip software installed. I'll have to find it on the web somewhere.
Any direct link suggestions?
Doug

AlexandraTrifan, 2019-01-09:
Sure, you should find it [here](https://www.7-zip.org/download.html) .

-Alexandra

damercer, 2019-01-09:
That looks like it must have been the problem. I started again with libsmu folder deleted and it ran to completion with a few warnings that I assume are normal:

BTW. You might want to see if a .zip file of the libusb-1.0 can be provide along side that one.

Doug@Doug-PC MINGW32 ~
$ ./libsmu-build.sh
Saving libsmu to  /home/Doug
Cloning into '/home/Doug/libsmu'...
remote: Enumerating objects: 11733, done.
remote: Total 11733 (delta 0), reused 0 (delta 0), pack-reused 11733
Receiving objects: 100% (11733/11733), 6.95 MiB | 3.84 MiB/s, done.
Resolving deltas: 100% (7332/7332), done.
Checking out files: 100% (95/95), done.
Building libsmu in  /home/Doug/libsmu/mingw-32
-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- Check for working CXX compiler: C:/msys64/mingw32/bin/g++.exe
-- Check for working CXX compiler: C:/msys64/mingw32/bin/g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: C:/msys64/mingw32/bin/gcc.exe
-- Check for working C compiler: C:/msys64/mingw32/bin/gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Boost version: 1.69.0
-- Found OpenMP_C: -fopenmp (found version "4.5")
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Looking for getopt
-- Looking for getopt - found
-- Downloading/updating googletest
-- Configuring done
-- Generating done
-- Build files have been written to: C:/msys64/home/Doug/libsmu/mingw-32/googletest-download
[1/9] Creating directories for 'googletest-download'
[2/9] Performing download step (git clone) for 'googletest-download'
Cloning into 'googletest-src'...
Checking out files: 100% (334/334), done.
Already on 'master'
Your branch is up to date with 'origin/master'.
[3/9] No patch step for 'googletest-download'
[4/9] Performing update step for 'googletest-download'
Current branch master is up to date.
[5/9] No configure step for 'googletest-download'
[6/9] No build step for 'googletest-download'
[7/9] No install step for 'googletest-download'
[8/9] No test step for 'googletest-download'
[9/9] Completed 'googletest-download'
-- Found PythonInterp: C:/msys64/mingw32/bin/python.exe (found version "2.7.15")
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    BOOST_INCLUDE_DIR

-- Build files have been written to: C:/msys64/home/Doug/libsmu/mingw-32
[1/46] Building CXX object src/CMakeFiles/smu-static.dir/device.cpp.obj
[2/46] Building CXX object src/CMakeFiles/smu-static.dir/session.cpp.obj
[3/46] Building CXX object src/CMakeFiles/smu-static.dir/usb.cpp.obj
[4/46] Building CXX object src/CMakeFiles/smu.dir/device.cpp.obj
[5/46] Building CXX object src/CMakeFiles/smu-static.dir/device_m1000.cpp.obj
In file included from ../src/device_m1000.hpp:22:0,
                 from ../src/device_m1000.cpp:7:
../src/device_m1000.cpp: In member function 'virtual ssize_t smu::M1000_Device::read(std::vector<std::array<float, 4> >&, size_t, int, bool)':
../src/debug.hpp:14:68: warning: format '%lu' expects argument of type 'long unsigned int', but argument 5 has type 'size_t {aka unsigned int}' [-Wformat=]
[... 84 lines cut ...]
[30/46] Linking CXX executable examples\smu_stream_multi_out.exe
[31/46] Building CXX object tests/CMakeFiles/test-multi-read.dir/test-multi-read.cpp.obj
[32/46] Building CXX object tests/CMakeFiles/test-session.dir/test-session.cpp.obj
[33/46] Building CXX object tests/CMakeFiles/test-read.dir/test-read.cpp.obj
[34/46] Building CXX object googletest-build/googlemock/CMakeFiles/gmock_main.dir/src/gmock_main.cc.obj
[35/46] Building CXX object googletest-build/googlemock/gtest/CMakeFiles/gtest_main.dir/src/gtest_main.cc.obj
[36/46] Building CXX object googletest-build/googlemock/CMakeFiles/gmock.dir/src/gmock-all.cc.obj
[37/46] Building CXX object googletest-build/googlemock/gtest/CMakeFiles/gtest.dir/src/gtest-all.cc.obj
[38/46] Linking CXX shared library bin\libgtest.dll
[39/46] Linking CXX shared library bin\libgtest_main.dll
[40/46] Linking CXX shared library bin\libgmock.dll
[41/46] Linking CXX executable tests\test-device.exe
[42/46] Linking CXX executable tests\test-read-write.exe
[43/46] Linking CXX executable tests\test-multi-read.exe
[44/46] Linking CXX executable tests\test-session.exe
[45/46] Linking CXX shared library bin\libgmock_main.dll
[46/46] Linking CXX executable tests\test-read.exe

Doug@Doug-PC MINGW32 ~
$

I'll proceed from here to install and test the resultant libsmu. I'll keep you informed if there are any problems.

Thanks
Doug

AlexandraTrifan, 2019-01-09:
I'll put together a guide for Windows builds. Please let us know whether you encounter or not any issues. 

Thank you.
-Alexandra

damercer, 2019-01-09:
I'm sorry but where did it put the results? All I found was a couple of .iss files in libsmu/mingw-32/dist ( besides the test .exe )? I looked inside the libsmu-x86.iss file and what that contained didn't make much sense for what the dir structure on this system is.

Doug

damercer, 2019-01-09:
There is a libsmu.dll in the src folder. Much larger than the one in the distribution I presently have installed. Probably due to DeBug stuff? I need new pysmu stuff as well to do my eventual testing?

What do I need to put where to make this all work?

Thanks

Doug

AlexandraTrifan, 2019-01-09:
You can find the libsmu.dll in libsmu/mingw-32/src. 
In order to build the python bindings you need to install and run a few more commands. 
`python2 -c \"import struct; print(struct.calcsize('P') * 8)\"`

` C:/msys64/mingw32/bin/python2.exe -m pip install --only-binary :all: --disable-pip-version-check --user --upgrade pip`

`C:/msys64/mingw32/bin/python2.exe -m  pip install --only-binary :all: wheel`

`C:/msys64/mingw32/bin/python2.exe -m  pip install cython`

To build it, navigate to C:\msys64\home\Doug\libsmu\bindings\python and run the following commands:
`python2 setup.py build_ext --compiler=mingw32 -L C:/home/Doug/libsmu/mingw-32/src -I C:/libusb-mingw/include/libusb-1.0`

`python2 setup.py build`

To rebuild after some updates, first you need to run the command: **cmake --build . --config Release** inside C:\msys64\home\Doug\libsmu\mingw-32 and for the python bindings just rerun the last two commands.

Does this provide everything you need?

damercer, 2019-01-09:
We will find out when I give this a try later this afternoon.
I'll keep you informed how things go.
Doug

damercer, 2019-01-09:
Oops didn't get past first thing on your list:

Doug@Doug-PC MINGW32 ~
$ python2 -c \"import struct; print(struct.calcsize('P') * 8)\"
bash: syntax error near unexpected token `struct.calcsize'

Doug@Doug-PC MINGW32 ~
$

Also, please be very specific with what I'm supposed to do with libsmu.dll. Do I just overwrite / replace the libsmu.dll in the C:\Program Files (x86)\Analog Devices\libsmu folder?

Thanks
Doug

damercer, 2019-01-09:
I went a head and ran the rest of the things listed anyway and this is was the result:

Doug@Doug-PC MINGW32 ~
$ python2 -c \"import struct; print(struct.calcsize('P') * 8)\"
bash: syntax error near unexpected token `struct.calcsize'

Doug@Doug-PC MINGW32 ~
$ C:/msys64/mingw32/bin/python2.exe -m pip install --only-binary :all: --disable-pip-version-check --user --upgrade pip
Requirement already up-to-date: pip in c:/msys64/mingw32/lib/python2.7/site-packages (18.1)

Doug@Doug-PC MINGW32 ~
$ C:/msys64/mingw32/bin/python2.exe -m pip install --only-binary :all: wheel
Collecting wheel
  Downloading https://files.pythonhosted.org/packages/ff/47/1dfa4795e24fd6f93d5d58602dd716c3f101cfd5a77cd9acbe519b44a0a9/wheel-0.32.3-py2.py3-none-any.whl
Installing collected packages: wheel
Successfully installed wheel-0.32.3

Doug@Doug-PC MINGW32 ~
$ C:/msys64/mingw32/bin/python2.exe -m pip install cython
Collecting cython
  Downloading https://files.pythonhosted.org/packages/c1/f2/d1207fd0dfe5cb4dbb06a035eb127653821510d896ce952b5c66ca3dafa4/Cython-0.29.2.tar.gz (2.0MB)
    100% |▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒| 2.0MB 2.0MB/s
Building wheels for collected packages: cython
  Running setup.py bdist_wheel for cython ... done
  Stored in directory: C:/Users/Doug/AppData/Local/pip/Cache/wheels/c0/d5/fa/c1942954b4e08e12fe6c10bab6ed5e8f391cdf577b81af4495
Successfully built cython
Installing collected packages: cython
Successfully installed cython-0.29.2

Doug@Doug-PC MINGW32 ~
$ pwd
/home/Doug

Doug@Doug-PC MINGW32 ~
$ cd libsmu

Doug@Doug-PC MINGW32 ~/libsmu
$ cd bindings

Doug@Doug-PC MINGW32 ~/libsmu/bindings
$ cd python

Doug@Doug-PC MINGW32 ~/libsmu/bindings/python
$ python2 setup.py build_ext --compiler=mingw32 -L C:/home/Doug/libsmu/mingw-32/src -I C:/libusb-mingw/include/libusb-1.0
running build_ext
Compiling C:/msys64/home/Doug/libsmu/bindings/python/pysmu/libsmu.pyx because it changed.
[1/1] Cythonizing C:/msys64/home/Doug/libsmu/bindings/python/pysmu/libsmu.pyx
building 'pysmu.libsmu' extension
creating build
creating build/temp.mingw-2.7
creating build/temp.mingw-2.7/msys64
creating build/temp.mingw-2.7/msys64/home
creating build/temp.mingw-2.7/msys64/home/Doug
creating build/temp.mingw-2.7/msys64/home/Doug/libsmu
creating build/temp.mingw-2.7/msys64/home/Doug/libsmu/bindings
creating build/temp.mingw-2.7/msys64/home/Doug/libsmu/bindings/python
creating build/temp.mingw-2.7/msys64/home/Doug/libsmu/bindings/python/pysmu
C:\msys64\mingw32\bin/i686-w64-mingw32-gcc.exe -fno-strict-aliasing -march=i686 -mtune=generic -O2 -pipe -fwrapv -D__USE_MINGW_ANSI_STDIO=1 -DNDEBUG -DNDEBUG -IC:/msys64/home/Doug/libsmu/include -IC:/libusb-mingw/include/libusb-1.0 -IC:/msys64/mingw32/include/python2.7 -c C:/msys64/home/Doug/libsmu/bindings/python/pysmu/libsmu.cpp -o build/temp.mingw-2.7/msys64/home/Doug/libsmu/bindings/python/pysmu/libsmu.o -std=c++11
writing build/temp.mingw-2.7/msys64/home/Doug/libsmu/bindings/python/pysmu/libsmu.def
creating build/lib.mingw-2.7
creating build/lib.mingw-2.7/pysmu
C:\msys64\mingw32\bin/i686-w64-mingw32-c++.exe -shared -Wl,--enable-auto-image-base -pipe -s -s build/temp.mingw-2.7/msys64/home/Doug/libsmu/bindings/python/pysmu/libsmu.o build/temp.mingw-2.7/msys64/home/Doug/libsmu/bindings/python/pysmu/libsmu.def -LC:/home/Doug/libsmu/mingw-32/src -LC:/msys64/mingw32/lib/python2.7/config -LC:/msys64/mingw32/lib -llibsmu -lpython2.7 -o build/lib.mingw-2.7/pysmu/libsmu.pyd
C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/7.4.0/../../../../i686-w64-mingw32/bin/ld.exe: cannot find -llibsmu
collect2.exe: error: ld returned 1 exit status
C:/msys64/mingw32/lib/python2.7/site-packages/Cython/Compiler/Main.py:367: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: C:/msys64/home/Doug/libsmu/bindings/python/pysmu/libsmu.pyx
  tree = Parsing.p_module(s, pxd, full_module_name)
error: command 'C:\\msys64\\mingw32\\bin/i686-w64-mingw32-c++.exe' failed with exit status 1

Doug@Doug-PC MINGW32 ~/libsmu/bindings/python
$ python2 setup.py build
running build
running build_py
copying pysmu/exceptions.py -> build/lib.mingw-2.7/pysmu
copying pysmu/utils.py -> build/lib.mingw-2.7/pysmu
copying pysmu/__init__.py -> build/lib.mingw-2.7/pysmu
creating build/lib.mingw-2.7/pysmu/_vendor
copying pysmu/_vendor/enum.py -> build/lib.mingw-2.7/pysmu/_vendor
copying pysmu/_vendor/__init__.py -> build/lib.mingw-2.7/pysmu/_vendor
running build_ext
building 'pysmu.libsmu' extension
C:\msys64\mingw32\bin/i686-w64-mingw32-gcc.exe -fno-strict-aliasing -march=i686 -mtune=generic -O2 -pipe -fwrapv -D__USE_MINGW_ANSI_STDIO=1 -DNDEBUG -DNDEBUG -IC:/msys64/home/Doug/libsmu/include -IC:/msys64/mingw32/include/python2.7 -c C:/msys64/home/Doug/libsmu/bindings/python/pysmu/libsmu.cpp -o build/temp.mingw-2.7/msys64/home/Doug/libsmu/bindings/python/pysmu/libsmu.o -std=c++11
In file included from C:/msys64/home/Doug/libsmu/bindings/python/pysmu/libsmu.cpp:657:0:
C:/msys64/home/Doug/libsmu/include/libsmu/libsmu.hpp:23:10: fatal error: libusb.h: No such file or directory
 #include <libusb.h>
          ^~~~~~~~~~
compilation terminated.
error: command 'C:\\msys64\\mingw32\\bin/i686-w64-mingw32-gcc.exe' failed with exit status 1

Doug@Doug-PC MINGW32 ~/libsmu/bindings/python
$

I looked and C:/msys64/home/Doug/libsmu/include/libsmu/libsmu.hpp indeed exists? Not sure what it was looking for?

Thanks

Doug

AlexandraTrifan, 2019-01-10:
[libsmu-scripts.zip](https://github.com/analogdevicesinc/libsmu/files/2745838/libsmu-scripts.zip)

I attached a couple of scripts. "install-deps-32bit.sh" and "install-py-32bit.sh" will install the dependecies needed to build everything. 
You already did this. You still need to run "libsmu-rebuild-all-32.sh" which will rebuild libsmu (if there were any changes), and will also rebuild the python bindings. 
The "libsmu-build.sh" was already used and should only be used the first time, when we clone the repository and setup the intial build.

So, you only need to run libsmu-rebuild-all-32.sh because you already did the setup for all the other things.

You can copy cli/smu.exe, libsmu.dll, libsmu.dll.a and libsmu.a from the libsmu/mingw-32/src folder inside the previous installation of libsmu (which should be C:\Program Files x86)\Analog Devices\libsmu). 
However, in order to have libsmu.dll working, you need to copy some more dlls (on which libsmu depends on). The libsmu installed in your system using an installer is built using the Visual Studio compiler. The one we built now, using MinGW, requires other dlls. 
To get those dlls, we can use the archives provided by the Appveyor builds. So for the libsmu 32bit build, we can download the following archive: https://ci.appveyor.com/api/buildjobs/1h46kefivfkvvbq3/artifacts/libsmu-1.0.1-gaf3a53d-MinGW-win32.zip . We need to extract this and copy everything inside it (except libsmu.dll, libsmu.dll.a, libsmu.a and the .exe files) in our installation folder (C:\Program Files (x86)\Analog Devices\libsmu).
Right now we have the new dlls and its dependencies. We can test if everything works fine by opening a Command Prompt in the previously mentioned location and trying smu.exe.

The built Python bindings (libsmu.pyd, pysmu) are located in libsmu/bindings/python/, having the following structure:
+---bin
|       pysmu
|       
+---build
|   +---bdist.mingw
|   +---lib.mingw-2.7
|   |   \---pysmu
|   |       |   exceptions.py
|   |       |   libsmu.pyd
|   |       |   utils.py
|   |       |   __init__.py
|   |       |   
|   |       \---_vendor
|   |               enum.py
|   |               __init__.py
|   |               
|   +---scripts-2.7
|   |       pysmu
|                                       
+---dist
|       pysmu-1.0.0-cp27-cp27m-mingw.whl
|       pysmu-1.0.0.mingw-py2.7.msi
|       
+---examples
|       get_samples.py
|       hotplug.py
|       leds.py
|       multi-cyclic-run.py
|       plot-voltage.py
|       read-continuous.py
|       read-write-continuous.py
|       read-write.py
|       read.py
|       
+---pysmu
|   |   array.pxd
|   |   cpp_libsmu.pxd
|   |   exceptions.py
|   |   libsmu.cpp
|   |   libsmu.pyx
|   |   utils.py
|   |   __init__.py
|   |   
|   \---_vendor
|           enum.py
|           vendored.txt
|           __init__.py
|           
+---pysmu.egg-info
|       dependency_links.txt
|       PKG-INFO
|       SOURCES.txt
|       top_level.txt
|       
\---tests
        misc.py
        test_channel.py
        test_device.py
        test_read.py
        test_read_write.py
        test_session.py
        test_signal.py

damercer, 2019-01-10:
First off tanks for all the effort so far on all this.
I decided to start over from the beginning just to test things out. I deleted the libsmu directory I had been working from (for a clean start). Then ran the ./libsmu-build.sh script to build from github original. That seemed to run to completion.
Next as a test I navigated to ~/libsmu/mingw-32/src/cli where smu.exe is located and with a m1k plugged in I typed the following which should start streaming samples:

Doug@Doug-PC MINGW32 ~/libsmu/mingw-32/src/cli
$ ./smu.exe -s
C:/msys64/home/Doug/libsmu/mingw-32/src/cli/smu.exe: error while loading shared libraries: ?: cannot open shared object file: No such file or directory

I guess I can't run smu.exe from here as a test?
I downloaded a copy of https://ci.appveyor.com/api/buildjobs/1h46kefivfkvvbq3/artifacts/libsmu-1.0.1-gaf3a53d-MinGW-win32.zip  and moved the contents to C:\Program Files(x86)\Analog Devices\libsmu as you suggested. I then copied the new cli/smu.exe, libsmu.dll, libsmu.dll.a and libsmu.a files in there as well.
Running new smu.exe in msys2 command line still did not work.

 I opened a Windows CMD prompt screen and tried smu -s and it worked ( I assume using new version in Analog Devices\libsmu) and then navigated to /libsmu/mingw-32/src/cli a tried smu -s there and it worked I assume it was using the local smu.exe? Typing sum --version will return same version number as the released version so I need to change that in the source code to make sure things have actually changed.

Also you seemed to indicated the running libsmu-build.sh also builds the python bindings. But it does not seem to move anything new to Pyhton27/...../site-packages/pysmu?

Do I need to move new stuff to Pyhton27/...../site-packages/pysmu manually?

Again Thanks
Doug

damercer, 2019-01-10:
Also, I just tried to run my Python code (ALICE 1.2) with this new stuff in C:\Program Files(x86)\Analog Devices\libsmu but old stuff in Pyhton27/...../site-packages/pysmu and Python can no longer import pysmu.

Doug

AlexandraTrifan, 2019-01-10:
Hi, 

The libsmu-rebuild-all-32.sh builds the Python stuff locally, so those need to be moved manually to the official Python folder.

-Alexandra

damercer, 2019-01-10:
Using Windows Explorer I deleted the contents of site-packages/pysmu and copied over the contents of the new libsmu/bindings/python/pysmu.
Still get error in Python shell: First attempt was before moving new files, second after moving files.
>>> from pysmu import *

Traceback (most recent call last):
  File "<pyshell#0>", line 1, in <module>
    from pysmu import *
  File "C:\Python27\lib\site-packages\pysmu\__init__.py", line 6, in <module>
    from .libsmu import *
ImportError: DLL load failed: The specified procedure could not be found.
>>> from pysmu import *

Traceback (most recent call last):
  File "<pyshell#1>", line 1, in <module>
    from pysmu import *
  File "C:\Python27\lib\site-packages\pysmu\__init__.py", line 6, in <module>
    from .libsmu import *
ImportError: No module named libsmu
>>> 

Still need another file or a link is broken?

Doug

AlexandraTrifan, 2019-01-11:
Hi, 

If you modify the CMakeLists.txt, you need to rerun the cmake command, in order to apply those changes. 
Regarding the python issue, let me investigate further. I'll come back with some updates. 

Thank you.
-Alexandra

damercer, 2019-01-11:
Is there a option switch to on which makes it stop on the first failing error? I'm not use to c++ syntax yet and likely will often make minor typos. Right now it there is one it goes whizzing by and I won't catch it. I'll have to scroll back through the lines of messages reading each one to see if I made a mistake. ( the cause of the above confusion ).

The Python error would indicate that there is something in the libsmu.pyx file that it is not finding because the file is there.

Thanks

Doug

AlexandraTrifan, 2019-01-11:
Hi, 

I attached the updated scripts. What I did:
- Added a **"set -e"** flag to make the scripts exit whenever a build error occurs. This way, you can catch all the minor typos that might appear and the script will not continue. 
- The "libsmu-rebuild-all-32.sh" script will :   **Run cmake** to make sure it uses the updated CMakeLists.txt, **build the source code** if anything changed and **install libsmu**. Libsmu will be installed in **/c/msys64/mingw32/bin/libsmu.dll** and **/c/msys64/mingw32/lib/libsmu.dll.a**. ( You can check this to make sure it's fine, or check the output in the terminal while it's installing. In the output you will see what and where got installed.
- The "libsmu-rebuild-all-32.sh" script will also : **build** the python bindings and **install** them in the appropriate location. (You should be able to find them in : /c/msys64/mingw32/pysmu ,  c:/msys64/mingw32/lib/python2.7/site-packages/pysmu-1.0.0-py2.7-mingw.egg, etc. ). The output from the command line should print the location of the installed files while they are being installed.

How did I test things: 
- Removed the libsmu folder and **ran "libsmu-build.sh"** to setup the repository and build the libsmu.dll
- **Ran "libsmu-rebuild-all-32.sh"** in order to build and install the python bindings. 
- In the same terminal (MSYS2 MinGW32) I ran **python.exe** (this will use the MinGW python, installed in /c/msys64/mingw32/python.exe)
- Tried "from pysmu import Session" and "s = Session()", "s.devices" and didn't encounter any other issues.

This way you don't have to copy any .dlls in your initial Python folder or Program Files/libsmu install folder. Those will all be installed in the MSYS2 MinGW32 subsystem and you should be able to use the python provided by it.

If changes occur in the library source code, or the CMakeLists.txt, **"libsmu-rebuild-all-32.sh"** should rebuild all (including the bindings) and install them in the required location.

Please let me know how this works for you. I am really sorry again for this inconvenience.

[libsmu-scripts.zip](https://github.com/analogdevicesinc/libsmu/files/2750119/libsmu-scripts.zip)

-Alexandra

damercer, 2019-01-11:
Here is another wrinkle. I use numpy and the python in  /c/msys64/mingw32/bin/ does not have that installed but the main one in in C:/Python27 does.
Another not so minor inconvenience with not moving the python bindings into the main version of Python..

damercer, 2019-01-16:
Right now I have two versions of Python 2.7.15 the one in the Windows registry with a working numpy and no new pysmu and the one under msys64 with a new pysmu and no numpy. I NEED both in the same version of python, preferably the one in the Windows registry (it has all the extensions I use).
Is there a way to have the build script put the results of the build (pysmu)  under C:/Python27 (the one in the Windows registry)? And I guess libsmu in C:/Program Files/Analog Devices/libsmu?

Progress is sort of at a stand still until this gets resolved.

Thanks
Doug

AlexandraTrifan, 2019-01-17:
Hi, 

You can install numpy under msys64 using the following command:  **pacman -Sy mingw-w64-i686-python2-numpy**

In order to put things under C:/Python27, all the libraries and executables should be copied manually. I am investigating this method and I'll come back with some guidelines when it's done.

Thank you.

-Alexandra

damercer, 2019-01-18:
Hi:
I'm having trouble making changes to the Python Bindings. Sorry to keep bothering you with seemingly simple issues.
A couple of basic questions.
First, which file to I need to edit to change the version number from 1.0.0?
Second, I make a couple of simple changes to libsmu.pyx ( only text file in bindings/python/pysmu that seemed right) to test if my changes actually change what results when used. For example simply changing the line that sets the leds from:
self._device.set_led(leds)
to:
self._device.set_led(7)
Which should just ignore the value passed in and turn on all 3 leds.
When I rebuild and then test the change in Python nothing has changed. Calling set_leds still changes based on value that is sent and not just turning on all three leds?
Am I not changing the right file? Or what do I need to do to effect the change made to libsmu.pyx?

Thanks
Doug

AlexandraTrifan, 2019-01-23:
Hi, 

In order to change the version of the pysmu bindings you can modify the libsmu/bindings/python/pysmu/_ _ init.py _ _ file. If you run **python setup.py install** inside libsmu/bindings/python, are the changes you made applied?

-Alexandra

damercer, 2019-01-23:
I changed the line in __init__.py and the version number changed after I ran setup install as can be seen from what was typed in the terminal screen:

$ python2 setup.py install
top level bindings directory= C:/msys64/home/Doug/libsmu/bindings/python
top level repo directory= C:/msys64/home/Doug/libsmu
running install
running bdist_egg
running egg_info
writing pysmu.egg-info/PKG-INFO
writing top-level names to pysmu.egg-info/top_level.txt
writing dependency_links to pysmu.egg-info/dependency_links.txt
reading manifest file 'pysmu.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no previously-included files matching 'CMakeLists.txt' found anywhere in distribution
warning: no previously-included files matching '*.pyc' found anywhere in distribution
warning: no previously-included files matching '*.pyo' found anywhere in distribution
warning: no previously-included files matching '__pycache__' found anywhere in distribution
writing manifest file 'pysmu.egg-info/SOURCES.txt'
installing library code to build/bdist.mingw/egg
running install_lib
running build_py
copying pysmu/__init__.py -> build/lib.mingw-2.7/pysmu
running build_ext
creating build/bdist.mingw/egg
creating build/bdist.mingw/egg/pysmu
copying build/lib.mingw-2.7/pysmu/exceptions.py -> build/bdist.mingw/egg/pysmu
copying build/lib.mingw-2.7/pysmu/libsmu.pyd -> build/bdist.mingw/egg/pysmu
copying build/lib.mingw-2.7/pysmu/utils.py -> build/bdist.mingw/egg/pysmu
creating build/bdist.mingw/egg/pysmu/_vendor
copying build/lib.mingw-2.7/pysmu/_vendor/enum.py -> build/bdist.mingw/egg/pysmu/_vendor
copying build/lib.mingw-2.7/pysmu/_vendor/__init__.py -> build/bdist.mingw/egg/pysmu/_vendor
copying build/lib.mingw-2.7/pysmu/__init__.py -> build/bdist.mingw/egg/pysmu
byte-compiling build/bdist.mingw/egg/pysmu/exceptions.py to exceptions.pyc
byte-compiling build/bdist.mingw/egg/pysmu/utils.py to utils.pyc
byte-compiling build/bdist.mingw/egg/pysmu/_vendor/enum.py to enum.pyc
byte-compiling build/bdist.mingw/egg/pysmu/_vendor/__init__.py to __init__.pyc
byte-compiling build/bdist.mingw/egg/pysmu/__init__.py to __init__.pyc
creating stub loader for pysmu/libsmu.pyd
byte-compiling build/bdist.mingw/egg/pysmu/libsmu.py to libsmu.pyc
creating build/bdist.mingw/egg/EGG-INFO
installing scripts to build/bdist.mingw/egg/EGG-INFO/scripts
running install_scripts
running build_scripts
creating build/bdist.mingw/egg/EGG-INFO/scripts
copying build/scripts-2.7/pysmu -> build/bdist.mingw/egg/EGG-INFO/scripts
copying pysmu.egg-info/PKG-INFO -> build/bdist.mingw/egg/EGG-INFO
copying pysmu.egg-info/PKG-INFO~ -> build/bdist.mingw/egg/EGG-INFO
copying pysmu.egg-info/SOURCES.txt -> build/bdist.mingw/egg/EGG-INFO
copying pysmu.egg-info/dependency_links.txt -> build/bdist.mingw/egg/EGG-INFO
copying pysmu.egg-info/top_level.txt -> build/bdist.mingw/egg/EGG-INFO
writing build/bdist.mingw/egg/EGG-INFO/native_libs.txt
zip_safe flag not set; analyzing archive contents...
creating 'dist/pysmu-1.0.1-py2.7-mingw.egg' and adding 'build/bdist.mingw/egg' to it
removing 'build/bdist.mingw/egg' (and everything under it)
Processing pysmu-1.0.1-py2.7-mingw.egg
Copying pysmu-1.0.1-py2.7-mingw.egg to c:/msys64/mingw32/lib/python2.7/site-packages
Removing pysmu 1.0.0 from easy-install.pth file
Adding pysmu 1.0.1 to easy-install.pth file
Installing pysmu script to C:/msys64/mingw32/bin

Installed c:/msys64/mingw32/lib/python2.7/site-packages/pysmu-1.0.1-py2.7-mingw.egg
Processing dependencies for pysmu==1.0.1
Finished processing dependencies for pysmu==1.0.1

I don't think there were any errors but I might be missing something?

However, I have also changed the following in libsmu.pyx:

def set_led(self, leds):
        """Set device LEDs.

        Args:
            leds: an integer number, the bits of the number represents the states of the leds (1-on 0-off) in order (RGB or DS3,DS2,DS1 on rev F) 
        Raises: IOError on USB failures.
        """
        self._device.set_led(7)
Which should turn on all three leds.
And I added this extra property at the end of:

cdef class _DeviceSignal(Signal):
    """Wrapper for device specific signal properties."""
-snip-
property resolution:
        """Get the signal's resolution value.

        >>> from pysmu import Session, Mode
        >>> dev = session.devices[0]
        >>> chan_a = dev.channels['A']
        >>> chan_a.mode = Mode.SVMI
        >>> chan_a.signal.resolution
        0.2
        """
        def __get__(self):
            return self._signal.info().resolution
When I go to my python test folder and run the program which does these test lines after setting up the session:

    devx.set_led(0b001)
    time.sleep(1)
    devx.set_led(0b010)
    time.sleep(1)
    devx.set_led(0b100)
    time.sleep(1)
    print "Channel A label = " + str(CHA.signal.label)
    print "Channel A max = " + str(CHA.signal.max)
    print "Channel A min = " + str(CHA.signal.min)
    print "Channel A resloution = " + str(CHA.signal.resolution)

The LEDs turn on one at a time rather than all three at once as they should (with this change  self._device.set_led(7) ) and it then fails when it gets to CHA.signal.resolution:

Channel A label = Voltage
Channel A max = 5.0
Channel A min = 0.0
Traceback (most recent call last):
  File "alice-desktop-1.3.pyw", line 17257, in <module>
    ConnectDevice()
  File "alice-desktop-1.3.pyw", line 15678, in ConnectDevice
    SelectBoard()
  File "alice-desktop-1.3.pyw", line 15725, in SelectBoard
    print "Channel A resloution = " + str(CHA.signal.resolution)
AttributeError: 'pysmu.libsmu._DeviceSignal' object has no attribute 'resolution'

So I think this demonstrates that the altered libsmu.pyx is not being used?

Thanks

Doug