analogdevicesinc/libsmu #122: Building libsmu with Raspberry pi not woring anymore?

damercer, 2018-12-22:
Building libsmu on a Raspberry Pi use to work. Have not tried it in a while.
On a freshly updated Pi 3 Jessie.
I followed the steps in the Read.me and at the cmake step it returned an error.

pi@raspberrypi:~/libsmu $ cmake build
CMake Error at /usr/share/cmake-3.0/Modules/FindBoost.cmake:1198 (message):
  Unable to find the requested Boost libraries.

  Unable to find the Boost header files.  Please set BOOST_ROOT to the root
  directory containing Boost or BOOST_INCLUDEDIR to the directory containing
  Boost's headers.
Call Stack (most recent call first):
  src/CMakeLists.txt:33 (find_package)

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Boost_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /home/pi/libsmu/src

-- Configuring incomplete, errors occurred!
See also "/home/pi/libsmu/build/CMakeFiles/CMakeOutput.log".

Here is the Cmake output log file as well:

[CMakeOutput.log](https://github.com/analogdevicesinc/libsmu/files/2705601/CMakeOutput.log)

Something about boost missing? Seems something has changed to break this on a Pi.

Thanks

Doug

damercer, 2018-12-22:
I was able to install Boost 1.53 with the following:
sudo apt-get install libboost1.53-all
There might be newer versions but this worked to get me to the next step.

Need to include something like this to the Read.me list of needed steps.

But the next problem is with make command error(s):

pi@raspberrypi:~/libsmu/build $ make
Scanning dependencies of target smu
[ 14%] Building CXX object src/CMakeFiles/smu.dir/device.cpp.o
[ 28%] Building CXX object src/CMakeFiles/smu.dir/session.cpp.o
[ 42%] Building CXX object src/CMakeFiles/smu.dir/usb.cpp.o
[ 57%] Building CXX object src/CMakeFiles/smu.dir/device_m1000.cpp.o
In file included from /home/pi/libsmu/src/device_m1000.hpp:22:0,
                 from /home/pi/libsmu/src/device_m1000.cpp:7:
/home/pi/libsmu/src/device_m1000.cpp: In member function â€˜virtual ssize_t smu::M1000_Device::read(std::vector<std::array<float, 4u> >&, size_t, int, bool)â€™:
/home/pi/libsmu/src/debug.hpp:14:68: warning: format â€˜%luâ€™ expects argument of type â€˜long unsigned intâ€™, but argument 5 has type â€˜size_t {aka unsigned int}â€™ [-Wformat=]
 #define DEBUG(...) do { if (DEBUG_TEST) fprintf(stderr, __VA_ARGS__); } while(0);
                                                                    ^
/home/pi/libsmu/src/device_m1000.cpp:562:4: note: in expansion of macro â€˜DEBUGâ€™
    DEBUG("%s: waiting %i ms for incoming samples: requested: %lu, available: %u\n",
    ^
/home/pi/libsmu/src/device_m1000.cpp: In member function â€˜virtual void smu::M1000_Device::flush(int, bool)â€™:
/home/pi/libsmu/src/device_m1000.cpp:650:29: error: â€˜class boost::lockfree::spsc_queue<float>â€™ has no member named â€˜consume_allâ€™
   m_out_samples_q[channel]->consume_all(flush_write_queue);
                             ^
/home/pi/libsmu/src/device_m1000.cpp:656:18: error: â€˜class boost::lockfree::spsc_queue<std::array<float, 4u> >â€™ has no member named â€˜consume_allâ€™
   m_in_samples_q.consume_all(flush_read_queue);
                  ^
src/CMakeFiles/smu.dir/build.make:123: recipe for target 'src/CMakeFiles/smu.dir/device_m1000.cpp.o' failed
make[2]: *** [src/CMakeFiles/smu.dir/device_m1000.cpp.o] Error 1
CMakeFiles/Makefile2:75: recipe for target 'src/CMakeFiles/smu.dir/all' failed
make[1]: *** [src/CMakeFiles/smu.dir/all] Error 2
Makefile:137: recipe for target 'all' failed
make: *** [all] Error 2
pi@raspberrypi:~/libsmu/build $ sudo make install
[ 14%] Building CXX object src/CMakeFiles/smu.dir/device_m1000.cpp.o
In file included from /home/pi/libsmu/src/device_m1000.hpp:22:0,
                 from /home/pi/libsmu/src/device_m1000.cpp:7:
/home/pi/libsmu/src/device_m1000.cpp: In member function â€˜virtual ssize_t smu::M1000_Device::read(std::vector<std::array<float, 4u> >&, size_t, int, bool)â€™:
/home/pi/libsmu/src/debug.hpp:14:68: warning: format â€˜%luâ€™ expects argument of type â€˜long unsigned intâ€™, but argument 5 has type â€˜size_t {aka unsigned int}â€™ [-Wformat=]
 #define DEBUG(...) do { if (DEBUG_TEST) fprintf(stderr, __VA_ARGS__); } while(0);
                                                                    ^
/home/pi/libsmu/src/device_m1000.cpp:562:4: note: in expansion of macro â€˜DEBUGâ€™
    DEBUG("%s: waiting %i ms for incoming samples: requested: %lu, available: %u\n",
    ^
/home/pi/libsmu/src/device_m1000.cpp: In member function â€˜virtual void smu::M1000_Device::flush(int, bool)â€™:
/home/pi/libsmu/src/device_m1000.cpp:650:29: error: â€˜class boost::lockfree::spsc_queue<float>â€™ has no member named â€˜consume_allâ€™
   m_out_samples_q[channel]->consume_all(flush_write_queue);
                             ^
/home/pi/libsmu/src/device_m1000.cpp:656:18: error: â€˜class boost::lockfree::spsc_queue<std::array<float, 4u> >â€™ has no member named â€˜consume_allâ€™
   m_in_samples_q.consume_all(flush_read_queue);
                  ^
src/CMakeFiles/smu.dir/build.make:123: recipe for target 'src/CMakeFiles/smu.dir/device_m1000.cpp.o' failed
make[2]: *** [src/CMakeFiles/smu.dir/device_m1000.cpp.o] Error 1
CMakeFiles/Makefile2:75: recipe for target 'src/CMakeFiles/smu.dir/all' failed
make[1]: *** [src/CMakeFiles/smu.dir/all] Error 2
Makefile:137: recipe for target 'all' failed
make: *** [all] Error 2

Not sure how to proceed from here.

Doug

msuciu, 2018-12-24:
Hi Doug,
From boost, you will need to install libboots-dev. 
There is a repository on gitlab https://gitlab.analog.com/Platformation/m1k-production-test on which we build libsmu on a Raspberry Pi for production testing. 
What we install is in setup_env.sh file.
I hope this will be helpful.
Mihai

damercer, 2018-12-27:
Clicking on the above link (gitlab.analog.... ) says server not found??? Where is the correct location?
Thanks
Doug.

damercer, 2018-12-28:
I was able to install libboots-dev and build libsmu. It seems to have made the smu executable, which seems to run, but the make install step does not seem to put the python bindings in the right place ( or anywhere I can see ) so import pysmu does not work.

Any help on making the steps outlined in the readme work correctly would be great.
Thanks
Doug

commodo, 2019-01-07:
We may have to improve the readme a bit.

In any case [and in the meantime] one way to guess/understand how something gets built on Github is to take a look at the `.travis.yml` or `appveyor.yml` files. Travis is usually used for Linux/OS X builds, AppVeyor for Windows builds [at least on ADI repos].
[ On a broader spectrum, some repos are using CircleCI, which has a different configuration format than Travis-CI & AppVeyor, but this is more of a side-note ].

If you take a look here: https://github.com/analogdevicesinc/libsmu/blob/master/.travis.yml
The steps and the requirements should be described [in various configuration elements].
These steps are usually meant to build on Ubuntu 14.04 [for Travis CI], but they do hold some good insight into how to extend to other platforms.

Now [obviously], this is not meant to replace a good readme, but [typically] source-code outpaces documentation [for various reasons] and this is one way that helps with building stuff.

commodo, 2019-01-07:
Will close this as there's a duplicate now: https://github.com/analogdevicesinc/libsmu/issues/123

closed 2019-01-07 by commodo