analogdevicesinc/libsmu #127: pysmu: unable to connect with ADALM1000 on Linux

balbirthomas, 2019-07-30:
Pysmu/libsmu fails to connect with a ADALM1000 device even though Pixelpulse2 works perfectly fine. It gives the following error message

``` python
user@pc: python
Python 2.7.16 (default, Apr  6 2019, 01:42:57) 
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from pysmu import *
>>> session = Session()
>>> session.add_all()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "pysmu/libsmu.pyx", line 200, in pysmu.libsmu.Session.add_all (/home/user/src/utilities/libsmu/bindings/python/pysmu/libsmu.cpp:4449)
pysmu.exceptions.SessionError: failed scanning and/or adding all supported devices: Device or resource busy
```

The same issue is seen with Alice 1.3 as shown below

``` python
user@pc: alice
No Init File Read
Traceback (most recent call last):
  File "/home/user/bin/alice", line 18339, in <module>
    ConnectDevice()
  File "/home/user/bin/alice", line 16660, in ConnectDevice
    session.add_all()
  File "pysmu/libsmu.pyx", line 200, in pysmu.libsmu.Session.add_all (/home/user/src/utilities/libsmu/bindings/python/pysmu/libsmu.cpp:4449)
pysmu.exceptions.SessionError: failed scanning and/or adding all supported devices: Device or resource busy
```

My software stack is
**OS** : Debian GNU/Linux v10 (Buster)
**libsmu** : Built from git (sha 9bf56333f008f56bc10a5e85927cb617ccc30b0f )
**alice** : Version-1.3 from Alice's git repository

Should you need any further information I would be happy to help.

Regards

cristi-iacob, 2019-08-19:
Hi!

It is a normal behaviour. When creating a Session-type object, it automatically scans the connected devices. Therefore, when you're trying to call the add_all() method you'll get a "Device or resource busy" error, because the device is already used by the created object.

However, you've probably found that piece of code in the examples. They are out of date for the moment. We will change them as soon as possible.

damercer, 2019-10-07:
The current version of ALICE 1.3 works fine under Windows with current master version of libsmu so this is a Linux only issue. Not being a Linux user I can't help debug the problem.

Doug

cristi-iacob, 2019-10-08:
https://github.com/analogdevicesinc/libsmu/blob/master/src/session.cpp#L61 
This means that if your operating system doesn't support the hotplug mechanism, libsmu will find all the available devices and add them in the devices list, when creating the session. That's why the error appears only on some platforms. However, the functionality is not affected by this error, because the devices are still found and can be used by libsmu.

Anyway, the next libsmu release will not support hotplug mechanism anymore, because of the compatibility problems and because libusb changed it's working mechanisms, which made it (almost) unusable for libsmu.

@damercer Can you please take a look on the new release or maybe even test it on Alice? Thank you!
(link to the pull request: https://github.com/analogdevicesinc/libsmu/pull/129)

damercer, 2019-10-08:
I can only test under Windows. That version from 1.0.1216 ( which is/was the same as the version I installed a few weeks ago when it first appeared) works as expected. I generally track changes as they appear and keep current with the latest version of master.

Doug

balbirthomas, 2019-10-10:
I can confirm that Alice version 1.3 from its git repository with latest libsmu from the git repository does indeed work with two minor changes to Alice. The changes required to Alice is commenting out the `session.add_all()` lines in Alice. I am closing this ticket in view of the above. I have noted the same on https://github.com/analogdevicesinc/alice/issues/2 .

closed 2019-10-10 by balbirthomas