analogdevicesinc/libsmu #130: Latest master build on appveyor missing my changes?

damercer, 2019-08-19:
Hi:

I just installed the latest build from master (appveyor 1.0.1204) and it seems to be missing my changes made back in Feb. See pull request #13. You said they would be added to master back then. Why are they still not there?

I now get the following error which clearly indicates that my new functions are missing. 

Traceback (most recent call last):
  File "C:\Users\Doug\Documents\RPI\ADiscovery\ALM\rework\alice-desktop-1.3.pyw", line 18544, in <module>
    ConnectDevice()
  File "C:\Users\Doug\Documents\RPI\ADiscovery\ALM\rework\alice-desktop-1.3.pyw", line 16871, in ConnectDevice
    SelectBoard()
  File "C:\Users\Doug\Documents\RPI\ADiscovery\ALM\rework\alice-desktop-1.3.pyw", line 16943, in SelectBoard
    devx.set_adc_mux(0)
AttributeError: 'pysmu.libsmu.SessionDevice' object has no attribute 'set_adc_mux'

Please do what you promised to do 6 months ago ASAP.

Thanks

Doug

AlexandraTrifan, 2019-08-19:
Merged the #124 PR

damercer, 2019-08-19:
Didn't seem to complete properly in appveyor? The last one (I've been using) that worked and included the changes was 1.0.1179 from 6 months ago.

Thanks

Doug

damercer, 2019-08-21:
Installed new build from 1.0.1212 and everything seems to be there now.
Thanks
Doug

closed 2019-08-21 by damercer