analogdevicesinc/libsmu #143: How to use libsmu?

peterbuxton, 2020-01-20:
I'm trying to use Alice with the ADALM1000 and have built and installed libsmu in Linux. Since I understand that libsmu is a library for communicating with the device over USB, I would like to know how to issue commands to do that, if possible (it sounds possible from the title of thread #78). I don't actually have Alice working (because I don't know how, yet), but my ADALM1000 shows connected in the command lsusb, so I'd like to know how I can make it acknowledge a command that I issue. Please tell me if I have this all wrong.

damercer, 2020-01-20:
There should have been a command line tool built with libsmu.
Try typing smu on the command line with the board connected. You should see something like this list of options:

smu: utility for managing M1K devices

 -h, --help                   print this help message and exit
 --version                    show libsmu version
 -l, --list-devices           list supported devices currently attached to the system
 -p, --hotplug-devices        simple session device hotplug testing
 -s, --stream-samples         stream samples to stdout from a single attached device
 -d, --display-calibration    display calibration data from all attached devices
 -r, --reset-calibration      reset calibration data to the defaults on all attached devices
 -w, --write-calibration <cal file> write calibration data to a single attached device
 -f, --flash <firmware image> flash firmware image to a single attached device

The -s option will simply stream samples to the command line text screen.

To learn how to use the various functions in Libsmu I suggest you use Python over C or C#. It is much more interactive. There are a number of very simple examples here under the GitHub Repo in the Python bindings but there are also a few example introductory labs available that have small programing exercises here on the ADI Wiki:
https://wiki.analog.com/university/labs/intro_ee
Look at links next to  class #19  Concept: Digital Electronics Interfacing Hardware and Software

peterbuxton, 2020-01-20:
Thank you so much, Doug. I do see the help menu so you got me started! I didn't think to use the command "smu". I will get some practice with this and follow up.
-Peter

closed 2020-01-23 by peterbuxton