analogdevicesinc/libsmu #146: Reading state of PIO0-3 via ctrl_transfer() fails

StefanMack, 2020-06-07:
Hi Alexandra,
thank you for the great pysmu package. The ADC and DAC functions of the libsmu python examples work fine. The ADALM1000 is an ideal tool for university teaching electrical engineering labs. Especially nowadays during Corona we can give the ADALM1000 to the students for a take home lab!

Setting the PIO0-3 (e.g. `device.ctrl_transfer(0x40, 0x51, PIO_0, 0, 0, 0, 100)`) also works fine, by I do not manage to read the PIO0-3 state.

The following code fails on **Python 3.69** with error:

**print(list(state))
TypeError: ord() expected string of length 1, but int found**

```
import time
from pysmu import Session

# assign digital pins
PIO_0 = 28
PIO_1 = 29
PIO_2 = 47
PIO_3 = 3 

try:
    session = Session(ignore_dataflow=True, queue_size=10000)
    if session.devices:
        dev = session.devices[0]
        while True:
            # get state of PIO0
            state = dev.ctrl_transfer(0xc0, 0x91, PIO_3, 0, 0, 1, 100)
            print(list(state))
            time.sleep(1)
    else:
        print('no devices attached')
    
except KeyboardInterrupt:
    print()
    print('Strg + C erkannt...')
    
finally:
    pass
```
I guess it has something to to with the following statement in the `libsmu.pyx` function definition of `ctrl_transfer()`:

```
if bm_request_type & 0x80 == 0x80:
    return map(ord, data)
```
To me it seems that the `map()` function causes trouble for Python 3.
Is there any work around? Or how and where can I change the implementation of the `ctrl_transfer()` function?

AlexandraTrifan, 2020-06-09:
Hi, 

Let us take a look and we will come back with a fix as soon as possible. 
We are working on a new release for some bug fixes, we will take a look at this and add this too.

-Alexandra

AlexandraTrifan, 2020-09-16:
Hi,

This causes trouble only for Python3. One idea would be to change the **return map(ord, data)** to **return bytearray(data)**.
Would this work for you?

-Alexandra

StefanMack, 2020-09-28:
Hi Alexandra,

I installed libsmu and the python bindings on Ubuntu 18.04. Where can I find the file` libsmu.pyx` to test the changes suggested by you?

-Stefan

AlexandraTrifan, 2021-02-05:
Hi,

We are sorry for the delay. We added the fix for libsmu.pyx on master. We will prepare a new release soon. We're just working on fixing the nightly builds on Appveyor and migrating the linux builds.
Here is the link to the commit that fixes the bindings: https://github.com/analogdevicesinc/libsmu/commit/aaad9af471647922baa306b4a424c6c713ccc41b 

-Alexandra