analogdevicesinc/libsmu #148: Mac OS and Linux package simple install instruction guide

damercer, 2020-07-04:
Hi:
We are going to have a lot of schools bringing their Labs Remote in Aug/Sept this year and many Students use Mac or Linux computers vs Windows. While we do now have Mac and Linux install packages (so they should not need to compile from source) we don't have simple "How to install or use" instructions (step by step hand-holding) for novice users.

Can someone with access to a Mac and / or Linux machine write such a quick guide?

I know everyone is busy with other prep work for supporting Fall Lab launches but I think this is a necessary part of supporting schools in the Fall term.

Thanks

Doug

saraheno, 2020-08-05:
I am very much hoping to use this with my students this fall, so I hope somebody can work on this.

rgetz, 2020-08-05:
yes - we purchased a few more Macs for people - so we can have a look at this soon. Sorry for the delay.

prebys, 2020-08-20:
The pre-compiled pysmu library for High Sierra works for all MacOS also works for Mojave and Catalina, but the package installer puts things somewhere that at least the HomeBrew python can't find them.   I found it was easier to just copy the libraries locally and set pointers to them with some shell scripts to set things up and run them.  I bundled this together with the latest Alice Release and there are step-by-step instructions here:
https://canvas.ucdavis.edu/courses/475222/pages/setting-up-the-alice-software-suite-on-macos-and-ubuntu
It's been tested on all three OSs and (at the moment) by one other person.

Note that at the moment, only the Alice Desktop is working (with some bugs that are discussed on the page).  The DC-meter hangs with the spinning ball of death.  I'm about to post that to the Alice discussion board.  I haven't tried any others.

I had it working with Ubuntu 20.04 until yesterday, when a seemingly innocuous system upgrade caused things to break with a segmentation fault as soon as it tried to talk to the module (I'm about to post that to this board as soon as I gather details).  

Hopefully this helps.

damercer, 2020-08-20:
Linux only fixes to the desktop and Voltmeter, Meter-Source Ohmmeter stand alone tools (including Pyhton3 support for the latter 3) have been committed to the Version1.3 branch. In my testing on Ubuntu 18.04 and 20.04 they now all work as well as they do in Windows 10.

The alice-desktop-1.3.pyw that was committed just this afternoon has a first attempt at OSX specific code to generate a “minimum” main menu ( for scope only right now). If this works it can possibly be expanded to address the “exit” complaint for OSX.

Doug

[External]

https://canvas.ucdavis.edu/courses/475222/pages/setting-up-the-alice-software-suite-on-macos-and-ubuntu<https://urldefense.com/v3/__https:/canvas.ucdavis.edu/courses/475222/pages/setting-up-the-alice-software-suite-on-macos-and-ubuntu__;!!A3Ni8CS0y2Y!vLb5Ugj-L_exiOGoSZdmfniQ2fOainMzZItG13DoXWCbxrXEHNbKsiHtmWtSoTgDYg$>

AlexandraTrifan, 2020-09-16:
Hi,

The build guide and all the instructions have been updated and can be found in the README file or on the official Documentation main page. ( https://github.com/analogdevicesinc/libsmu/blob/master/README.md ).
This includes steps on how to install the packages for all the systems and instructions on how to manually build the library and the Python bindings.

The DC-meter hang that was caused by libsmu was also fixed with this new release. The new release can be found here: https://github.com/analogdevicesinc/libsmu/releases/tag/v1.0.3

Please let us know if there are any other issues.

Thank you,
-Alexandra

saraheno, 2020-09-18:
Hi Alexandra,
We were having today with one of our student's macs today.  I cc him so he
can explain in detail with screen grabs the problem.  However, we were
going to try and download your new fixes and got confused.  We went to

https://github.com/analogdevicesinc/libsmu/blob/master/README.md

and saw text saying "Download the specific libsmu .pkg package for your
MacOS distribution from the Releases section. Currently we are supporting
MacOS 10.13, 10.14 and 10.15. The package name should start with libsmu and
contain the MacOS version."

but we don't know where to find the releases section.  Any help would be
appreciated

Professor Sarah Eno

saraheno, 2020-09-18:
I went to the ALICE directory and opened the ALICE desktop. When it opened,
I immediately got a pop up (err1). I clicked "ok". It then asked me to
flash firmware (err2), so I clicked "yes". It pulled open this window
(err3), but I didn't know what to do, so I clicked "cancel". ALICE froze
for a moment. I then got this pop up (err4). I clicked "ok". I got the same
pop up again (err5). I clicked "ok". I got a pop up saying ALICE needed to
restart (err6). I clicked "ok" and it ALICE closed.
I have previously been able to use the dc-meter without issue to power and
measure circuits.

-Arthur Carlton-Jones

AlexandraTrifan, 2020-09-21:
Hi,

The latest release packages can be found here: https://github.com/analogdevicesinc/libsmu/releases/tag/v1.0.3 .

saraheno, 2020-09-21:
Thanks Alexandra.  The other problem seems to be the need to install new
firmware on the adalm1000.  We found that installing pixelpulse would
prompt and then install it.  but is there a more straightforward way to put
the new firmware on?

AlexandraTrifan, 2020-09-21:
Hi,

The Pixelpulse automatic way is the simplest way to install the new firmware. Pixepulse would look for the latest firmware and flash it on the board.

A different way would be to use the SMU Command Line Interface. 
For MacOS, in order to have the SMU CLI installed you should use the second option to install the library (the one that involves the **.tar.gz** release file, not the .pkg). 
As stated in the README file, the .pkg installs the base library, but the .tar.gz also installs the command line interface. Once that is installed, you can test it by typing "smu" in a terminal (this will display the help message).

After that, download the latest firmware release file ( https://github.com/analogdevicesinc/m1k-fw/releases/tag/v2.17 ) and run :
`smu -f m1000.bin`

You should see the following in your terminal:
_"smu: successfully updated firmware"
"Please unplug and replug the device to finish the process"_

After replugging, run:
`smu -l`
This will list your connected device, the serial number and the firmware version you just installed.

-Alexandra

damercer, 2020-09-21:
No, not really. The smu command line utility can flash new firmware files on to a board. If you are not going to use Pixelpulse2 at all then that might be a simpler path, assuming the libsmu Mac install package includes the smu cli.

In a terminal screen type smu. It should print a list of the possible command options. The -f option followed buy the desired .bin file name should do the trick. Make sure the .bin file is in the current directory. You will need to cycle the power to the board to reboot the new firmware.

Doug

[External]

damercer, 2020-09-21:
By the way, just as an aside. You can stop ALICE from complaining about the firmware not being up to date by changing these two lines in the alice_init.ini file:

global AllowFlashFirmware; AllowFlashFirmware = 0
global IgnoreFirmwareCheck; IgnoreFirmwareCheck = 1

Setting the AllowFlashFrimware to 0 removes that Option
Setting the IgnoreFirmwareCheck to 1 disables the prompt to update on startup.

ALICE will run fine with the factory installed firmware (rev F boards) it will just not provide the 200 KSPS modes.

Doug

saraheno, 2020-09-21:
Hi Alexandra, Doug,
We are now having another problem with our MacOS boards.  A student was
running the dc-meter.sh last week successfully on Cataline MacOS.  He was
playing with the ADALM1000 between the labs, and now the program hangs.  It
does open up, but he cannot change any of the settings, etc.  We noticed
there were fixes for hanging with macos in the latest release and so wanted
to install it and see if this fixed his issue.  We went to
https://github.com/analogdevicesinc/libsmu/blob/master/README.md  to follow
the MacOS instructions.  However, one of the listed commands does not
work.  brew install python3 python3-setuptools python3-pip

 I attach a screen grab with the error
Any help would be appreciated.

[image: Screen Shot 2020-09-21 at 10.29.57 AM.png]

AlexandraTrifan, 2020-09-21:
Hi,

The hangs related to libsmu are fixed in the latest release.
We can't see the attached screenshot.

Please try to run only:
`brew install python3`

I checked now and it seems there were some updates to the Homebrew package manager/installer in the last couple of days. The Python3 package should include "pip" and "setuptools"

-Alexandra

saraheno, 2020-09-21:
Hi Alexandra,
I copy the student Siyuan here so he can try.  You can see the image at:
https://github.com/analogdevicesinc/libsmu/issues/167

saraheno, 2020-09-21:
just do
> brew install python3

It includes the other two packages.  They’re separate on linux.

-Eric

--------------------------------------------------------------------
Eric Prebys
Professor, Department of Physics and Astronomy
Director, Crocker Nuclear Laboratory
University of California, Davis
http://physics.ucdavis.edu/people/faculty/eric-prebys
---------------------------------------------------------------------

On Sep 21, 2020, at 7:43 AM, Sarah C. Eno <eno@umd.edu<mailto:eno@umd.edu>> wrote:

Hi Alexandra, Doug,
We are now having another problem with our MacOS boards.  A student was running the dc-meter.sh last week successfully on Cataline MacOS.  He was playing with the ADALM1000 between the labs, and now the program hangs.  It does open up, but he cannot change any of the settings, etc.  We noticed there were fixes for hanging with macos in the latest release and so wanted to install it and see if this fixed his issue.  We went to https://github.com/analogdevicesinc/libsmu/blob/master/README.md  to follow the MacOS instructions.  However, one of the listed commands does not work.  brew install python3 python3-setuptools python3-pip

 I attach a screen grab with the error
Any help would be appreciated.

<Screen Shot 2020-09-21 at 10.29.57 AM.png>

prebys, 2020-09-22:
Thanks for the heads up about the smu utility.  It's definitely an easier way to go than PixulPulse to upgrade the firmware. PixelPulse has always been a nail biter involving a couple of restarts and hangs before it finally does the upgrade.  

I do have to run smu twice.  The first time puts it into Samba mode and then can't find it, and the second goes through, but it's still a lot faster than PixelPulse.  

I'm just going to update all the firmware myself before giving it to the students.

AlexandraTrifan, 2021-02-09:
The README page and the official libsmu documentation (https://wiki.analog.com/university/tools/m1k/libsmu) contain packages and build/install guides for both Linux and MacOS. Also, the Conda packages are available, which ease the libsmu install process. 
Assuming this issue is fixed I am closing this.
A new one can be opened if any problems appear.

closed 2021-02-09 by AlexandraTrifan