analogdevicesinc/libsmu #151: libsmu-1.0.2-g43b4bcd-setup-x86.exe

damercer, 2020-08-04:
You might already be aware from your testing, but just to let you know I tested this version of libsmu from the latest master build on AppVeyor:

libsmu-1.0.2-g43b4bcd-setup-x86.exe

It does not work trying to Run ALICE 1.3.7 from source code on either Python 2.7 or Python 3.7.

Just thought you might like confirmation that you still have something broken / not working.

Thanks

Doug

damercer, 2020-08-04:
HELP!
I need your help urgently. After doing the above on that test version of libsmu I uninstalled that version and double checked that everything was gone. The smu cli was gone and the pysmu stuff in site-packages for both versions of Python was also gone.

Now after re-installing the released version of libsmu 1.0.2 from a fresh down load of the ,exe it seems to install correctly but ALICE does not run properly any more (in either 2.7 or 3.7) Hangs when trying to access the m1k. It seems to be Python specific because the smu cli works i.e. smu -s streams data fine.

Help Help Help. How do I excise this evil test version from my system!

BTW. The Windows exe version made earlier with py2exe still runs correctly (made before installing the test version) which is self contained as to which libsmu / pysmu it uses.

Thanks

Doug

cristi-iacob, 2020-08-04:
Hi!

The second issue seems really strange in the context in which you are sure you've deleted everything from the last installation.
However, you can try the following procedure as well:
1. Uninstall pysmu. For this you can use `python -m pip uninstall pysmu`.
2. Install pysmu directly from the .whl files from the release (in order to avoid some potential issue from the installer). For this you should download the specific .whl file and install it with `python -m pip install pysmu`. 

Hope this solves the issue for you! 
And about the first issue, I have managed to reproduce it and will take a closer look on it. Thanks for reporting it!

damercer, 2020-08-04:
I have some more back ground on this problem. I went looking in the libsmu directory (C:Progam Files(x86)\Analog Devices\libsmu and noticed that the libsmu.dll file was dated 7/31 (the date from the bad test on AppVeyor) and should have been dated 1/20 from the current released build. I replaced that libsmu.dll from 7/31 with the one from 1/20 and things seem to be working again. It would seem that the un-install process DID NOT remove the file as I think it should have???!!!?? This looks like a bug to me.

I'll do some more testing and get back to you if I learn anything new.

damercer, 2020-08-04:
And / or the reinstall saw the file with a newer date and did not overwrite it with the one from the package being installed. Which I again think is not the proper action if the idea is to revert to an older (working) version.

Thanks

Doug

cristi-iacob, 2020-08-10:
Hi!

We have found the not-overwriting issue. We have fixed it and this will be available in the future release. Thanks for pointing this out.

About the issue with Alice: I have only managed to reproduce the issue when running Alice by the .exe file provided by the installer. When running it from the source files it works properly on my machine with both Python2 and Python3. It also seems that pysmu is installed properly by the installer (I can import it in the Python console). I tried to build Alice by myself, in order to obtain the .exe file locally, but haven't managed to do this. Can you please explain me the Alice's build process and how to obtain the .exe file? 

Thanks!

damercer, 2020-08-10:
The windows installer includes it's own local copy of libsmu which should be the one that it uses (and be the proper matching 32 bit one). I think that some times the search paths might get mixed up and it some how tries to use the wrong library?

Either way for some reason the libsmu.dll that came with the test version of master I downloaded was not working running from source code with 2.7 or 3.7. 

I'll revisit all this when you put out the next official release ( with all the accumulated bugs fixed).

Thanks

Doug

closed 2020-08-10 by damercer