analogdevicesinc/libsmu #152: Installing libsmu .deb package on Ubuntu 20.10

damercer, 2020-08-08:
Hi:
I'm trying to go through and test the steps to get libsmu (and ultimately ALICE) running on Linux.
Steps so far:
Installed VitrualBox VM software on Windows 10 computer
Made a Ubuntu amd64 20.10 VM
Run the VM
Installed various missing Python3 packages (numpy, tkinter, etc. )
Installed some of the packages listed in README.md to prepare to compile from source (but not all so gave up and moved on)
Downloaded copy of libsmu-1.0.2-ubuntu-18.04-amd64.deb from current release
in a terminal screen typed:
sudo dpkg -i libsmu-1.0.2-ubuntu-18.04-amd64.deb
It stopped with an error libboost stuff was missing.
I tried using various names to try to install boost:
sudo apt-get install libboost
sudo apt-get install boost
But none of the things I could think off worked to install boost?

Why is boost not included in the package? Or at least an explanation on how to set things up properly before installing package.

We can do a better job, need to do a better job of streamlining the process here.

Doug

damercer, 2020-08-08:
OK I tried to install books by typing:
sudo apt-get install libboost-all-dev (after a google search)
It went for minutes apparent;y installing libboost .....
When it finished I tried again to install the libsmu package:
sudo dpkg -i libsmu-1.0.2-ubuntu-18.04-amd64.deb
Returned the same error about various things from libboost not being installed.

THIS is INSANE...

Doug

damercer, 2020-08-09:
Further developments. It would seem that in spite of the errors when installing the libsmu-1.0.2-ubuntu-18.04-amd64.deb package at least part of libsmu was installed. The smu cli functions and connects with an M1k. But trying to import pysmu in Python 3.8 does not work.
Am I correct in assuming that the package install should have added the Python bindings as well? Or did this step not complete due the libboost errors?
I'm not super familiar with the Linux file system so I'm not sure I'm looking in the right places but I could not find a site-packages folder under any of the Python folders at least based on where things are in a Windows Python installation. So here is another question, where did the install package put the Python binding if it put them anywhere at all?
Also is there a way to do the Python bindings install manually after the fact?

All these steps need to be better explained in a How to install on Linux User's Guide!

Thanks

Doug

cristi-iacob, 2020-08-10:
Hi!

You are right, the build process described in the README.md file for Ubuntu is a little outdated. We will update it soon. 
Installing libboost-all-dev is the right thing to do. The latter error can be fixed by running the following command `pip install --upgrade cython`. 

About the .deb package installation on Ubuntu 20.: this process is not guaranteed to be successful. In fact there are differences between different versions of Ubuntu and therefore the installation process might not succeed. We will add an ubuntu20.deb file in the next release.

About the Python bindings: they have to be installed separately. In order to install them one should take the following steps (after building and installing libsmu):
`1. go to libsmu/bindings/python`
`2. python setup.py build`
`3. sudo python setup.py install`
We will cover these steps in the How to install on Linux User's Guide for the next release. We will also probably add a .deb file for the Python bindings.  

Thank you for the feedback!

damercer, 2020-08-10:
I'll give the manual steps a try later today. In the mean time I have a question about the rules file? Does the .deb package install the rules file (53-adi-m1k-usb.rules) because I didn't see it in the proper folder and manually put it there and then smu and (lsusb) found the board? If it doesn't that will be needed as well I think?

Would it be possible to have just one install package that installs both the libsmu library and the Python bindings (optionally maybe)? That would make things simpler for the average casual student users. We are getting a lot of non-EE/CS types i.e Physics students who will be doing labs with M1k this fall.

Thanks

Doug

cristi-iacob, 2020-08-11:
Hi!

Yes, the .deb package installs the rules file (at least it does on my machine and on two other virtual machines). Did you have any problem with the rules file installation? The .deb package should install the rules file in `/etc/udev/rules.d`. This happened on your Ubuntu 20 Virtual machine, right? If so, this might be one of the issues of using an Ubuntu 18 deb package on an Ubuntu 20 machine.

About the Python bindings installation: .deb packages do not support optional installations (like the Windows installers do). We will probably make two separate .deb packages (one for libsmu and one for the Python bindings) and explain in the How to install on Linux User's Guide what each of these is used for. This might also avoid some potential issues, like a person who only wants to use libsmu and does not have Python installed on the machine.

Thanks!

damercer, 2020-08-11:
When making a suggestion please be as pedantic as possible. I just wasted over a half hour and got no where. Booting up Linux in a virtual machine is a slow process and it does not run at full speed on top of that.
Where do I start from when you say:

1. go to libsmu/bindings/python

 I can't find any folder name starting with libsmu? I tried using search but it found nothing.

Please understand that this can be very frustrating for a non-linux user and try not to waste any more of my time by not being exact in your instructions. (assume nothing).

Thanks for your understanding.

Doug

cristi-iacob, 2020-08-11:
I'm sorry for that. I have just assumed that you were trying to build libsmu manually. 

In order to get to step one you should first clone the libsmu repository `git clone https://github.com/analogdevicesinc/libsmu.git`. This will create a folder named libsmu, containing all the packages files. If you do not have git installed on your virtual machine, you can do this with `sudo apt-get install git`.
After cloning the repository you should be able to do the steps from the previous comment. Note that you should install the libsmu .deb package before taking these steps. 

Just one small observation: on Ubuntu 20 Python 2 is not installed by default. You should either install it or use Python3 instead (replace python with python3 in the commands). Also you might need to install pip (or pip3). You can do this with `sudo apt-get install python-pip` (or `sudo apt-get install python3-pip`). 

Please let me know if there are any other problems!

damercer, 2020-08-11:
OK let me be pedantic.
All I'm trying to do for right now is test the ability to:
1: install a libsmu .deb package.
2: apparently "manually" add Python (3 for now ) bindings
3: so I can test ALICE.
So far I've only gotten the smu cli to work through these means.
I have installed / updated Python 3(.8) with tk interface and numpy successfully and the idle3 editor. Right now I think all I need is to get pysmu to import.

Thanks
Doug

cristi-iacob, 2020-08-12:
Great, you are almost done. From this point on you just have to run the following commands:
1. `git clone https://github.com/analogdevicesinc/libsmu.git`. This will create a folder named libsmu.
2. `cd libsmu/bindings/python`
3. `python3 setup.py build`
4. `sudo python3 setup.py install`

This should install pysmu and make it importable from Python.

damercer, 2020-08-14:
OK here is where things sit. I did step 1 above successfully. Obviously Step 2. Step 3 completed successfully (apparently). However for step 4 it ended with an error saying cython module was missing. To check on cython I did "pip install --upgrade cython" again and it said the installed package was up to date?

What next?

Doug

cristi-iacob, 2020-08-14:
Hi!

I can not reproduce the behavior from your machine. Can you, please, send me the error message you receive when using `pip install --upgrade cython`?

About the alternative solutions: the problem with this kind of package is that it is very restrictive. As also mentioned on that page: 

> For the moment, the software must be run form this directory!

This is because most of Alice's dependencies are built statically and are only imported at runtime because they are in the same folder (it's like creating a static Python image with pysmu already imported). This means that you can not import pysmu in Python in general, only if running the given Python from the tar archive. The downside of this process is that the library would become hardly maintainable (we should make a package like this for Pixelpulse, Alice, pysmu and so on for every new release, which may become a very large bug source). Yes, this solution works very good for Alice, but is not a good idea when talking about a library with many other usages. 
The generic script method is usually not a very good idea, because of the differences between each machine. This issue itself is a very good example: you and me are running the same scripts, which were successful for me, but unsuccessful for you. This is an isolated case, but considering the large number of computers on which libsmu is installed, it is quite difficult to write some script which would work on every single machine.

The good alternative is generating a .deb package. These packages solve the issue from the generic scripts (you can see a .deb package like a perfect generic script). This is because all these packages respect a very strict architecture and the apt-get system is based on this architecture. That is why is also nontrivial to build a .deb package. 
Anyway, we will build a .deb package for pysmu as well and afterwards it should be easy to install and use.

damercer, 2020-08-14:
The error is not with installing cython but with using it apparently.
The "sudo python3 setup.py install" step fails with an error that cython module is missing.
It take a long time to bring up Linux so I don't do it often. I can make copies of the terminal screen messages next time I'm in Linux.

Thanks

Doug

By the way Professors like this are desperate to get something that works right away and that is why they are resorting to things like that. We need to help them out.
We need to do better, We can do better!

cristi-iacob, 2020-08-14:
Ok, it seems that your python version does not have cython installed. You can install it with `sudo python3 -m pip install cython`. 
If this does not solve your issue, please send me some screenshots or the error messages, when you are getting on Ubuntu. Thanks!

Yes, we are aware of the increasing necessity of libsmu and Pixelpulse and are working right now on them.

damercer, 2020-08-14:
You are not listening to me(reading these messages).

I did that once (sudo python3 -m pip install cython ) already.

[External]

cristi-iacob, 2020-08-14:
Yes, forgot about that, sorry! I am waiting for the screenshots/error messages from you! 
Thank you!

AlexandraTrifan, 2020-09-16:
Hi,

We released the new version of libsmu. This includes more .deb packages (including one for ubuntu 20) and .deb packages for Python bindings.
Also we updated the build guide and provided instructions on how to either install .deb packages or build the library manually.
The instructions can be found here: https://github.com/analogdevicesinc/libsmu/blob/master/README.md

Please let us know whether things work this way.

-Alexandra

closed 2022-02-11 by AlexandraTrifan