analogdevicesinc/libsmu #153: MAC OS blues

saraheno, 2020-08-18:
Dear Eric and Doug,
I am reaching out for your help on using ALICE with MAC OS High Sierra.  I am working with a student who has a MAC in preparation for the start of classes.  I unfortunately only have windows10 machines.
We tried using the instructions at: https://canvas.ucdavis.edu/courses/475222/pages/setting-up-the-alice-software-suite-on-macos-and-ubuntu and the instructions at https://wiki.analog.com/university/tools/m1k/alice/desk-top-users-guide#linux_and_osx but both are giving different problems.

When using the instructions at ucdavis, unfortunately, it is not finding the pysmu libraries.  Looking at setup.sh in the alice-1.3.7-U directory, since his $OSTYPE is a “darwin”, the commands that should locate the libraries are:
export DYLD_LIBRARY_PATH=$PWD/pysmu-lib-MacOS
export PYTHONPATH=$PWD/pysmu-lib-MacOS/pythong3.7/site-packages/
From these names, I would guess that the pysmu library should be in $PYTHONPATH.  I do not see an “.so” with pysmu in its name (screenshot below) but maybe it is in one of the other so’s?  If it is in one of the other ones, is $PYTHONPATH not the right logical name?

[https://www.physics.umd.edu/~eno/misc/macosblues.png](url)
 
When using the instructions at wiki.analog, Since these instructions use ALICE 1.3.7, and there is a new 1.3.8 version, we then decided to try the instructions at: https://wiki.analog.com/university/tools/m1k/alice/desk-top-users-guide#linux_and_osx
However, these are challenging to follow. First note that “libsmu” is not defined anywhere on this page.  I assume these are the ALICE codes/libraries?  
 First, note that just after the green box in this area, it says “To manually install on Windows download” which is confusing to begin with, since this section is on linux and osx.  Then it says “for OS X and Linux” and then gives some libsmu versions for Darwin, which are 10.13.6 and 10.14.4.  However, his version of Darwin is darwin17.   Are any of these apprrpriate?  Also, when I go to the link https://github.com/analogdevicesinc/alice/releases I see “assets” for windows (the *-setup.exes) and I see the code.  I do not see any precompiled libraries for OSX

In conclusion:  right now we cannot get ALICE to work following either set of instructions.  Any help would be greatly appreciated.

Best
Sarah Eno

AlexandraTrifan, 2020-09-16:
Hi,

We created a new release for libsmu. This includes some fixes related to how libsmu works, but also some new packages. 
The new release packages can be found here: https://github.com/analogdevicesinc/libsmu/releases/tag/v1.0.3

Also, we updated the README and the build steps. You can find them in the README page ( https://github.com/analogdevicesinc/libsmu/blob/master/README.md ) or on the official documentation main page.
For MacOS, you can use the .tar package to install the base library, then jump to the Build Python Bindings step, to build and install Pysmu using your choice for the Python executable.

Please let us know if this help!

-Alexandra

saraheno, 2020-09-16:
Hi Alessandra,
Thank you for doing this.  My semester started two weeks ago, and my
students used the software patch provided by Eric.  We won't be doing
installs again until next semester.  But hopefully this will work well for
them then.
Best
Sarah Eno