analogdevicesinc/libsmu #154: Apple brew problem

damercer, 2020-08-19:
Hi:

I have received the following feedback from some one trying to bring up libsmu / pysmu Python on Mac OSX.

"—universal option has been deprecated in brew (since 2017 I believe), any suggestions? Thanks."

This might explain why people following the steps outlined in the libsmu README are not having success.

Doug

damercer, 2020-08-19:
Here is an update on progress this person is making:

"Leave off the --universal option:
brew install libusb
seems that I also need to do
brew link libusb 
I then ran into permission error with
brew install cmake boost pkg-config python
I followed some online suggestions 
sudo chown -R $(whoami) $(brew —prefix)
which returns an error but some how "brew install cmake boost pkg-config python” now works."

damercer, 2020-08-19:
OK, here is the final results from my friend who was trying to get pysmu to work on OSX:

"okay, I was finally able to build the pysmu package, but when I tried to use it in Python3.7, I get

>>> from pysmu import Session
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pysmu-1.0.2-py3.7-macosx-10.9-x86_64.egg/pysmu/__init__.py", line 6, in <module>
    from .libsmu import *
ImportError: dlopen(/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pysmu-1.0.2-py3.7-macosx-10.9-x86_64.egg/pysmu/libsmu.cpython-37m-darwin.so, 2): Library not loaded: @rpath/libsmu.1.dylib
  Referenced from: /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pysmu-1.0.2-py3.7-macosx-10.9-x86_64.egg/pysmu/libsmu.cpython-37m-darwin.so
  Reason: image not found"

Doug

cristi-iacob, 2020-08-19:
Hi!

We will try to reproduce this and will come back to you as soon as possible! Thanks for the feedback!

damercer, 2020-08-20:
Hi:
The latest news from my friend is that he has found the problem.

"it was a simple fix, **I just need to add libsmu/build/src to $DYLD_LIBRARY_PATH**.  now it works just like on PC! "

I think this may be the first person outside ADI to get libsmu working on a Mac in recent times.
Maybe this info can be incorporated in the instructions for installing on a Mac / included in any installers we might provide.
Thanks
Doug

AlexandraTrifan, 2020-08-28:
Hi,

We updated the install/build guide for both Linux and MacOS. We tested this on a clean Mac and it seems to work properly.
Could you let us know if this works for this case? If not, please let us know the issues and we will try to fix things.

The build guide can be found here ( https://github.com/analogdevicesinc/libsmu/blob/master/README.md ).

Thank you,
-Alexandra

damercer, 2020-08-28:
As you are aware I don't (yet?) have access to a Mac. My friend has it working now already and I don't know if he can spare the time to un-do what he has and try over with these instructions. I will pass along that we have put up an updated set of instructions.
Thanks
Doug

gwiederhecker, 2020-09-15:
@damercer , I have exactly the "image not found" error you mentioned. Do you know exactly what your friend means. by " add libsmu/build/src to $DYLD_LIBRARY_PATH." ?
Thanks!

cristi-iacob, 2020-09-15:
Hi!

To answer your question: you should run the following command
`DYLD_LIBRARY_PATH=/path/to/libsmu/build/src:$DYLD_LYBRARY_PATH`.

However, we have released a new test version of libsmu (and are currently preparing a new official release). You might want to install libsmu through the specific .pkg package for your MacOS version. You can find these packages [here](https://github.com/analogdevicesinc/libsmu/releases/tag/test_release). 

Please let us know if this solved the issue for you! Thanks!

gwiederhecker, 2020-09-15:
Dear @cristi-iacob, Thanks for your suggestion on the DYLIB path, it worked (to some extent)! Also the package link you kindly sent leads to a 404 web error.

Let me clarify a bit my attempt to get my ADALM1000 up and running as it my help other users struggling on this.

1.  I am running Catalina (10.15.6 (19G73)).
2.  I first installed PixelPulse with no issues, inlcuding the firmware upgrade. It works. But I wanted to run Alice 1.3 (Bode Analyzer, etc)
3. I have multiple Python installations running on my machine, including one based on HomeBrew (Python 3.7.7).  With a bit of struggle, I managed to install libmsu following the build instructions available on Github. 
4. This installation was working for the most part:

-  I can run `smu -l`
`ADALM1000: serial 2031205051485435313730333030323 : fw 2.17 : hw F`
- I can the python examples in the `/bindings/python/examples` folder, e.g.,
`/usr/local/Cellar/python3/3.7.7/bin/python3 read-continuous.py `
` 0.392877  0.000209  0.471952 -0.000211` 

5.  So you may ask, what is the issue then?
- When I tried to run the .pyw files shipped with alice-Version-1.3, most of them worked but the alice-desktop-1.3.pyw 
- I am getting the following message: `Python 3.x
DEPRECATION WARNING: The system version of Tk is deprecated and may be removed in a future release. Please don't rely on it. Set TK_SILENCE_DEPRECATION=1 to suppress this warning.
Windowing System is aqua` 
- and then `Traceback (most recent call last):
  File "alice-desktop-1.3.pyw", line 319, in <module>
    root.createcommand('::tk::mac::Quit', Bcloseexit)
NameError: name 'Bcloseexit' is not defined
`
- Then I started digging down this rabbit hole of fixing TK installation.
6. Following multiple suggestions on StackExchange,  I installed a fresh version of python 3.7.9 directly from Python.org which is supposed to be immune to bugged TCL/TK shipped with OSX. In this new python setup I couldn't get the libmsu running before your DYLIB trick.
7. For reference, what I did was: 
- Export the DYLIB path: `export DYLD_LIBRARY_PATH=/Users/gsw/Documents/GitHub/libsmu-master/build/src` followed by `cmake .. -DBUILD_PYTHON=ON -DUSE_PYTHON2=OFF`
- The following `make` command succeeded, followed by `sudo make install `
- After this, the `python3 read-continuous.py`command worked (at the examples folder): 
 `0.461496  0.000333  0.569000 -0.000102`
8. For my deception, there are still two issues:
- I need to export DYLIB path in every terminal session (although, there are guides out there to make this permanent in Catalina)
- **Most important,** I still get the `NameError: name 'Bcloseexit' is not defined` when I try to run the `alice-desktop-1.3.pyw`. This might be an issue to be investigate in a different issue though. Any suggestions are welcome.

**PS: I also installed the provided binaries on a Windows 10 bootup and theit worked out-of-the-box**

damercer, 2020-09-15:
Hi:
Your error on line 319:

Traceback (most recent call last): File "alice-desktop-1.3.pyw", line 319, in <module> root.createcommand('::tk::mac::Quit', Bcloseexit) NameError: name 'Bcloseexit' is not defined 

Is part of an untested (because I don't own a Mac) attempt to add a custom menu bar ( that only OSX needs ).
You can comment out / remove the  section starting at:
if (root.tk.call('tk', 'windowingsystem')=='aqua'):

Just don't try to use any of the stuff in the default OSX default menu (like exit) only use the Exit button in the main ALICE screen.

Doug

gwiederhecker, 2020-09-15:
@damercer , you solved it once again. Thanks! 
I am curious now as why to leave the exit command if it does not work (maybe in some version combination it does...).

cristi-iacob, 2020-09-16:
@gwiederhecker the reason you got an 404 error when accessing the test release is that we deleted it before you tried to access it, because we have made the official release (and deleted the test one). Sorry for this!

You can find the new official packages [here](https://github.com/analogdevicesinc/libsmu/releases/tag/v1.0.3). You might want to install libsmu from the .tar.gz archive, with the command described in the [README file](https://github.com/analogdevicesinc/libsmu/blob/master/README.md) in the MacOS section. Doing this you will most likely solve the DYLIB issue for you.

Thanks for your feedback and please let us know if you have any further issues!

damercer, 2020-09-16:
From the research I've been able to do on line and with absolutely no direct Mac experience, all I've been able to gather is that OSX by default includes a set of Menus along the top of an application's main screen. And only OSX does this. Those default menu commands need to be over-written by the application in order to work properly for that specific application. ALICE is developed and tested on non OSX computers and thus does not have a (tested) way of changing those OSX specific menus.

andrewnere, 2022-03-25:
So I've followed this thread, and am having issues on my mac still related to this thread.  Here's my setup:

Running: 10.15.7
Python: 3.7.9

Steps:
1) Installed PKG version Via: $sudo installer -pkg libsmu-1.0.4-darwin-10.15.7.pkg -target /    
2) Installed pysmu via: $python3 -m pip install --index-url https://test.pypi.org/simple/ pysmu
3) Downloaded the source from GITHUB here for the examples: https://github.com/analogdevicesinc/libsmu
4) Attempt to run example: $python3 libsmu-master/bindings/python/examples/get_samples.py 

Result:

Traceback (most recent call last):
  File "libsmu-master/bindings/python/examples/get_samples.py", line 10, in <module>
    from pysmu import Session, Mode
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pysmu/__init__.py", line 16, in <module>
    from .libsmu import *
ImportError: dlopen(/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pysmu/libsmu.cpython-37m-darwin.so, 2): Library not loaded: @rpath/libsmu.framework/Versions/1.0.4/libsmu
  Referenced from: /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pysmu/libsmu.cpython-37m-darwin.so
  Reason: image not found

Note that the .so file DOES exist in expected location

peterkinget, 2022-11-13:
I basically followed the same steps as andrewnere today (Nov. 13 2022) and get the following issue:

% python3
Python 3.10.8 (v3.10.8:aaaf517424, Oct 11 2022, 10:14:40) [Clang 13.0.0 (clang-1300.0.29.30)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pysmu
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pysmu/__init__.py", line 16, in <module>
    from .libsmu import *
ImportError: dlopen(/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pysmu/libsmu.cpython-310-darwin.so, 2): Library not loaded: @rpath/libsmu.framework/Versions/1.0.4/libsmu
  Referenced from: /Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pysmu/libsmu.cpython-310-darwin.so
  Reason: image not found
>>> 

* On a Mac with 10.15, I installed python 3.10 from python.org since it should have better tkinter integration
* I installed numpy ==> import tkinter and import numpy work fine
* I downloaded the libsmu package and installed it from the command line
* I installed the .whl for pysmu and that succeeded. I used 
python3 -m pip install --force-reinstall pysmu-1.0.4-cp310-cp310-macosx_10_15_x86_64.whl 
to make sure it was reinstalled; I was not sure which cp version to pick ... 
* then I tried python3 and still ran into the issue above; the libsmu so file is present in the location pysmu is looking for so the error must be something else ... 

Any suggestions?

Some of the documentation suggest to get a .tar.gz file for libsmu but I can only find one with the source files, not one with the library or the command line tools. 

Thanks, -- Peter Kinget