analogdevicesinc/libsmu #156: segmentation fault on Ubuntu 20.04

prebys, 2020-08-20:
I had Alice Desktop working on Linux Ubuntu 20.04 after building the pysmu library locally.

After was appeared to be a minor system upgrade yesterday, the program crashes with a segmentation violation as soon as it tries to communicate with the ADALM1000.  

It appears to crash at libusb_control_transfer().  I'm attaching a capture of the crash information (sorry, I couldn't seem to select the text).

I updated all packages and python modules and rebuilt the library, but it didn't help.   

I've tried it on multiple machines, and it fails except on the one where I didn't do the update yesterday.

I'd appreciate any advice.
[alice-ubuntu-crash.pdf](https://github.com/analogdevicesinc/libsmu/files/5105811/alice-ubuntu-crash.pdf)

damercer, 2020-08-21:
I just (8:00 PM EDT) did a clean install of Ubuntu 20.04 including all the latest updates. I then went through the install steps for getting to a running ALICE using the libsmu-1.0.2-ubuntu-18.04-amd64.deb from the current libsmu release. (it is for 18.04 and reports an error about libboost but ignore that) then built python bindings from a close of master.

Text file of install steps can now be found in the alice Version-1,3 branch

alice-desktop-1.3 from latest Version-1.3 branch ran perfectly.

Not sure what you updated that caused it to no longer function.

AlexandraTrifan, 2020-08-24:
Hi,

The libsmu-1.0.2-ubuntu-18.04-amd64.deb package does not install the python bindings. This .deb only installs the library and the smu utility. The Python bindings need to be manually build (for now). 
This can be done using the following steps:

`sudo apt-get install git python3 python3-pip python3-setuptools`
`sudo pip3 install cython`
`git clone https://github.com/analogdevicesinc/libsmu.git` This will create a folder named libsmu.
`cd libsmu/bindings/python`
`python3 setup.py build`
`sudo python3 setup.py install`

After this, the libsmu python bindings should be installed in the python site-packages directory and should be available.

We are working on the next release which will include a .deb package for Ubuntu 20 and will also include .deb packages for Python bindings (for all the Ubuntu versions we support). This way the Python bindings will no longer need to be installed manually.

Thank you.
-Alexandra

prebys, 2020-08-25:
This crashes with the following error:
prebys@eric-VirtualBox2:~/Downloads/libsmu/bindings/python$ python3 setup.py build
running build
running build_py
running build_ext
building 'pysmu.libsmu' extension
x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/prebys/Downloads/libsmu/include -I/usr/include/libusb-1.0 -I/usr/include/python3.8 -c /home/prebys/Downloads/libsmu/bindings/python/pysmu/libsmu.cpp -o build/temp.linux-x86_64-3.8/home/prebys/Downloads/libsmu/bindings/python/pysmu/libsmu.o -std=c++11
/home/prebys/Downloads/libsmu/bindings/python/pysmu/libsmu.cpp:643:10: fatal error: libsmu/version.hpp: No such file or directory
  643 | #include "libsmu/version.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

AlexandraTrifan, 2020-08-28:
Hi,

We prepared a test release ( https://github.com/analogdevicesinc/libsmu/releases/tag/test-release ). We are using the artifacts to run the last tests on packaging and other features before the official release. Could you try the .deb files for your Linux version and let us know if you still have issues?
A guide on installing this can be found here ( https://github.com/analogdevicesinc/libsmu/blob/test-release/README.md ).

Thank you,
-Alexandra

damercer, 2020-08-28:
Hi:
Here is my feedback on using the .deb packages from the test release on a fresh install of Ubuntu 20.04.
First the
sudo apt install -f ./<libsmu_package_name>.deb
and second
sudo apt install -f ./<python_package_name>.deb
steps seemed to run without errors. But you probably want to be more specific (pedantic) about running the
sudo ldconfig
step right after the above install steps. It is Required if you Don't DO a restart. This is not made clear enough in the readme.md steps.

Now for the bad news. While alice-destop-1.3.pyw (direct from the alice Version-1.3 branch) opens and runs it has problems. It works the first time the Run button is hit, but hangs if you hit Stop then try to hit Run again. It does not throw any Python level errors just hangs (must be at the C level) and you have to kill it by closing the Python Shell window (I was running it using IDLE). This also happens for the dc-meter-source-tool-1.3.pyw program as well but it hangs the first time you hit the Run button. Same behavior, no errors in the Python Shell and the only way to exit the program is by killing the shell window.

This is much the same problem I reported seeing in Windows for another test version of libsmu: see issue #151 which I closed pending another go at libsmu. Seems it might need to be re-opened!!! To get things to work again on Windows I reverted to the last known good libsmu v.1.0.2 release from Jan 20 which by the way was the one I got the other .deb package from and used on Upuntu 18.04 (and 20.04) that works perfectly. So SOMETHING has changed and BROKEN libsmu between then and now.

Thanks

Doug

prebys, 2020-09-02:
I verified that the Linux version works (a bit on 18.04 and 20.04), but I'm having the same trouble with hanging that damercer mentions.  Here is my weird magic bullet that works most of the time...

- delete the .cfg file before running the program for the first time
- start program.
- immediately exit to create .cfg file
- start again.
- Set up and use as desired.
- IMPORTANT!  Before exiting, set both sources back to DC from whatever other waveform they were set to (this is the weird part).  
- Exit
- When starting again, reselect all the modes, even if the check marks appear to be in the correct places.

If I remember the part about setting it back to DC before exiting, then it will run the next time. Otherwise, it hangs and I have to kill the process and delete the .cfg file to get it to work again.

Also, like damercer, the DC-meter-source app hangs as soon as I hit run and I have to kill the process to end it.  No magic bullet found for that one yet.

damercer, 2020-09-03:
@prebys can you give the stand alone executable for the ALICE desktop, Ubuntu-18.04 tar.gz that was recently added to the current release a try?
https://github.com/analogdevicesinc/alice/releases/tag/1.3.8
It is just the main program not the DC-meter-source tool. It was made with the older (working) version of libsmu and should work for you on Ubuntu 18.04 and 20.04.
Any feedback on how these stand alone executables work will be most helpful. If they work then the other apps can be built as well.
Thanks
Doug

cristi-iacob, 2020-09-11:
Hi!

You are right, there was a hidden bug in the new commits from libsmu. We solved it and updated the test release. We have tested the new modifications on alice-destop-1.3.pyw and dc-meter-source-tool-1.3.pyw from the latest alice release and the issue seems to be solved. Can you, please, check it out as well? The test release can be found [here](https://github.com/analogdevicesinc/libsmu/releases/tag/test_release) or in the Releases section from the libsmu github repository.

Thank you!

damercer, 2020-09-11:
Hi:
I'm going to test this right now. I'll get back to you when it's complete but in the mean time based on the comment attached to the change with regard to the calling of device->cancel() and session->end(), should these be added to the Python programs somewhere? The documentation is not always explicit about where, when and it what order things should be done to start, pause, restart and end communicating with the board.

Thanks
Doug.

damercer, 2020-09-11:
The new libsmu and pysmu .deb packages for Ubuntu 20.04 in the test release are now working on my test version of Ubuntu 20.04 (running in VirtualBox on Windows 10). So it looks like these packages are good to go for Linux users.
The "test" release does not include the Windows installers. I will download the .exe installers directly from AppVeyor and give them a try as well.
Thanks
Doug

cristi-iacob, 2020-09-14:
Hi!

Thanks for your feedback! 

About the cancel() and end() methods: their usage is the same with the usage their equivalents in a C++ program. The device->cancel() method cancels all current transfers from a device, even if the transfer is controlled by a session or not. The session->end() method finishes the pending transfers for non-continuous sessions before calling device->cancel() for each connected device. As there are multiple ways of communicating with the board, it is quite difficult to give a short answer; it depends on which method one uses (starts a transfer through session->start()/device->read()/device.get_samples() from the Python bindings etc.).

damercer, 2020-09-14:
Ok. I'll just leave things as there are in the program then (seems to work) unless you tell me to do otherwise.
I tested the Windows installer from AppVeyor for this test run and it works as expected as well now so I think we can probably close out this issue for now.
Doug

prebys, 2020-09-15:
I have confirmed the 1.0.3 release with Ubuntu 20.04 and the stability problems seem to be fixed.  Thanks.

AlexandraTrifan, 2020-09-16:
Since this issue seems to be solved, closing this for now. Can be reopened if anything appears.

closed 2020-09-16 by AlexandraTrifan