analogdevicesinc/libsmu #159: WriteTimeout error when changing waveform while reading

hehonglu123, 2020-09-04:
I've a script running `get_samples()` in a separate thread continuously and display data like an oscilloscope.  Sometimes upon start, when I first try changing the waveform while streaming, I'll get the following error
```
Traceback (most recent call last):
  File "m1k_service.py", line 84, in stream
    reading=self.device.get_samples(self.sample_size)
  File "pysmu\libsmu.pyx", line 681, in pysmu.libsmu.SessionDevice.get_samples
  File "pysmu\libsmu.pyx", line 379, in pysmu.libsmu.Session.get_samples
  File "pysmu\libsmu.pyx", line 253, in pysmu.libsmu.Session.run
pysmu.exceptions.WriteTimeout: data write timeout, no available fallback sample: The directory cannot be removed.
```
And this error forces my python script to exit.

cristi-iacob, 2020-09-08:
Hi!

I tried to reproduce this behavior, but didn't manage it. I have started a daemon thread which constantly calls the get_samples() method and afterwards was changing the waveform manually very often from the main thread (using the specific channel methods, e.g. sine, triangle, constant). For me all went well, the program never crashed and had the expected behavior.

Can you please provide some more details about your program and environment? On which OS are you running the program, which Python version do you use, which libsmu version do you use? It would be really helpful if you could send us some code snippets as well!

Thank you!

hehonglu123, 2020-09-09:
Thanks for the response. I created [a simple script](https://gist.github.com/hehonglu123/5ceb0ab31c8b4471f6f071a96199dfae) reproducing the behavior. 

Please let me know if you're able to reproduce. I'm using Win10 2004, libsmu v1.01 python3.7 binding.

cristi-iacob, 2020-09-11:
Hi!

Thanks for the script. The problem here seems not to be changing the waveform, but not having data to read at the beginning. In your script, you are starting the reading process before generating any samples. The script crashes because libsmu is trying to read samples, but they do not exist. 

In order to avoid this crash, you should first start the samples generation (by setting a waveform or by using the write() method) and only afterwards start the sample-reading thread. In this way, there will always exist samples to be read by the reading thread.

This main method rewrite should solve your issue:
```
def main():	
	m1k_obj=m1k()
	m1k_obj.device.channels['A'].mode =Mode.SVMI
	m1k_obj.device.channels['A'].square(0, 5, 10, 0, 0.75)
	m1k_obj.StartStreaming()

	time.sleep(2)
	m1k_obj.StopStreaming()
```

Hope this solves the problem for you. If you have any further questions, please let us know!

hehonglu123, 2020-09-11:
Thanks for the response! I see the point, but even at `HI_Z` mode it should still be possible to stream right? I tried following without any error:
```
def main():
	
	m1k_obj=m1k()
	m1k_obj.StartStreaming()
	time.sleep(2)
	m1k_obj.StopStreaming()
```

The interface I have starts streaming upon, and the user should be able to change waveform/mode while streaming.

cristi-iacob, 2020-09-11:
Yes, you are right. You can stream the data you receive in HI_Z mode. Also, the code snippet from your last comment looks like a proper way to do this thing.

cristi-iacob, 2020-09-11:
Yes, we managed to reproduce the error. The error was provoked by the time.sleep(0.1) from line 56. In this case, the channel is already switched to SVMI, but there are no samples generated at this time (because HI_Z is already turned off and the new generation process is not yet started). If you remove the time.sleep(0.1) statement, the new generation process will start practically instantly.

hehonglu123, 2020-09-11:
I see, so I added that `time.sleep()` to mimic possible lag for my interface. Seems like it does need generating right after switching to `SVMI`, even `time.sleep(0.001)` will cause the crash. Thanks!

closed 2020-09-11 by hehonglu123