analogdevicesinc/libsmu #165: Mac version not working.

prebys, 2020-09-16:
I'd appreciate it if someone could either make package installers that work or an archive that includes the python bindings for the latest version on MacOS

I was able to kludge together a working version of 1.0.2 from the tar file, but the tar file from version 1.0.3 lacks the python bindings.  

I tried to build the library with the python bindings, but it breaks (on 10.13) with
-- Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES) 
-- Could NOT find OpenMP_CXX (missing: OpenMP_CXX_FLAGS OpenMP_CXX_LIB_NAMES) 
-- Could NOT find OpenMP (missing: OpenMP_C_FOUND OpenMP_CXX_FOUND) 

The ADALM1000s have been flying off the shelf because a lot of people took the "supports Mac and Linux" at face value (we bought 40) and I don't think it's unreasonable to expect stuff to work as advertised.

AlexandraTrifan, 2020-09-16:
Hi,

The 1.0.3 tar file no longer includes the Python bindings binary files because we found that those link to exact versions. Since a system can have many different Python version, this would result in a mixup between the location of the Python version used to actually build the binaries, and the location of the Python version used to start a pysmu/libsmu application. 

Did you try to use the steps described in the README build guide that we updated?
`tar -xzvf <libsmu_package_name>.tar.gz --strip=3 -C /usr/local`
This step should install the base library (meaning only libsmu.dylib and include files) in "/usr/local". You can change that if you need to have it in another location.
Then, the build guide directs the user to installing the Python bindings:
`analog@analog:~$ git clone https://github.com/analogdevicesinc/libsmu.git`
`analog@analog:~$ cd libsmu/bindings/python`
`analog@analog:~$ python3 setup.py build`
`analog@analog:~$ sudo python3 setup.py install`
This will install the python binaries in the /lib/python3.x/site-packages corresponding to the Python3 executable that you chose to use. After this, you can delete the libsmu folder that you just cloned, and should be able to run "import pysmu". 

Could you please let us know if this works for you?

-Alexandra

prebys, 2020-09-16:
Yes, this appears to work, at least on 10.14.  Thanks.

-Eric

AlexandraTrifan, 2021-01-27:
Will close this issue. Can be reopened if the problem persists .

closed 2021-01-27 by AlexandraTrifan