analogdevicesinc/libsmu #167: macos installation instructions issue

saraheno, 2020-09-21:
Hi Alexandra, Doug,
We are now having another problem with our MacOS boards.  A student was running the dc-meter.sh last week successfully on Cataline MacOS.  He was playing with the ADALM1000 between the labs, and now the program hangs.  It does open up, but he cannot change any of the settings, etc.  We noticed there were fixes for hanging with macos in the latest release and so wanted to install it and see if this fixed his issue.  We went to https://github.com/analogdevicesinc/libsmu/blob/master/README.md  to follow the MacOS instructions.  However, one of the listed commands does not work.  brew install python3 python3-setuptools python3-pip

 I attach a screen grab with the error
Any help would be appreciated. 

[image]

AlexandraTrifan, 2020-09-21:
Please try to run:
`brew install python3`
`pip3 install --upgrade pip`
`pip3 install setuptools`
`pip3 install cython`

-Alexandra

saraheno, 2020-09-29:
Hi Alexandra.  I notice that the official instructions in the Readme still have the old command.  would it make sense to change that to these?

AlexandraTrifan, 2020-10-01:
Hi,

I updated the documentation to only install "python3" package.

-Alexandra

AlexandraTrifan, 2021-01-27:
Closing this. Can be reopened if there are any other issues.

closed 2021-01-27 by AlexandraTrifan