analogdevicesinc/libsmu #169: Mistake in Mac instructions puts libraries where python can't find them

prebys, 2020-10-07:
The Mac instructions at 
https://github.com/analogdevicesinc/libsmu/blob/master/README.md#build-instructions-for-libsmu-on-macos

say to unpack the Mac libraries with
tar -xzvf <libsmu_package_name>.tar.gz --strip=3 -C /usr/local

but this only works it python is installed in /usr/local/bin.  The python version from python.org for example, installs at 
/Library/Frameworks/Python.framework/Versions/3.8/bin/python3

and it won't find the libraries.  In this case, you would do:
tar -xzvf <libsmu_package_name>.tar.gz --strip=3 -C /Library/Frameworks/Python.framework/Versions/3.8

and it will work.

prebys, 2020-10-07:
Actually, I spoke too soon.  First of all, if it's not in /usr/local/lib, then the binding build can't find it, so I did both, but whereas this worked on my Catalina machine, on Mojave, when I try to import pysmu, I get:

>>> import pysmu
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pysmu-1.0.3-py3.8-macosx-10.9-x86_64.egg/pysmu/__init__.py", line 6, in <module>
    from .libsmu import *
ImportError: dlopen(/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pysmu-1.0.3-py3.8-macosx-10.9-x86_64.egg/pysmu/libsmu.cpython-38-darwin.so, 2): Library not loaded: @rpath/libsmu.1.dylib
  Referenced from: /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pysmu-1.0.3-py3.8-macosx-10.9-x86_64.egg/pysmu/libsmu.cpython-38-darwin.so
  Reason: image not found
>>> 

Can someone *please* just tell me where to put these libraries so python can find them?

AlexandraTrifan, 2021-01-27:
Hi,

Conda packages have been created: https://wiki.analog.com/university/tools/conda
They include the python bindings and library.
This type of package is more easier to use.
Could you let us know if this works for you?

-Alexandra