analogdevicesinc/libsmu #171: setup.py build fails on macos at #include "ios"

gsteele13, 2020-10-20:
Hi,

I'd love to get this board up and running and programming it in python. 

I've been following the instructions in the readme for MacOS: https://github.com/analogdevicesinc/libsmu

I've installed the 10.14.6 framework, but I'm hitting my head on the wall on the python bindings. My first hurdle was `port install libusb` 

(indeed, I needed the c++ libusb it seems not the python one of same name...)

But now I still don't make it through compiling the python bindings;

```
(myenv) TUD1001336 python$ python3 setup.py build
running build
running build_py
creating build
creating build/lib.macosx-10.7-x86_64-3.7
creating build/lib.macosx-10.7-x86_64-3.7/pysmu
copying pysmu/__init__.py -> build/lib.macosx-10.7-x86_64-3.7/pysmu
copying pysmu/utils.py -> build/lib.macosx-10.7-x86_64-3.7/pysmu
copying pysmu/exceptions.py -> build/lib.macosx-10.7-x86_64-3.7/pysmu
creating build/lib.macosx-10.7-x86_64-3.7/pysmu/_vendor
copying pysmu/_vendor/__init__.py -> build/lib.macosx-10.7-x86_64-3.7/pysmu/_vendor
copying pysmu/_vendor/enum.py -> build/lib.macosx-10.7-x86_64-3.7/pysmu/_vendor
running build_ext
Compiling /Users/gsteele/Documents/GitHub/libsmu/bindings/python/pysmu/libsmu.pyx because it changed.
[1/1] Cythonizing /Users/gsteele/Documents/GitHub/libsmu/bindings/python/pysmu/libsmu.pyx
/Users/gsteele/anaconda3/envs/myenv/lib/python3.7/site-packages/Cython/Compiler/Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /Users/gsteele/Documents/GitHub/libsmu/bindings/python/pysmu/libsmu.pyx
  tree = Parsing.p_module(s, pxd, full_module_name)
building 'pysmu.libsmu' extension
creating build/temp.macosx-10.7-x86_64-3.7
creating build/temp.macosx-10.7-x86_64-3.7/Users
creating build/temp.macosx-10.7-x86_64-3.7/Users/gsteele
creating build/temp.macosx-10.7-x86_64-3.7/Users/gsteele/Documents
creating build/temp.macosx-10.7-x86_64-3.7/Users/gsteele/Documents/GitHub
creating build/temp.macosx-10.7-x86_64-3.7/Users/gsteele/Documents/GitHub/libsmu
creating build/temp.macosx-10.7-x86_64-3.7/Users/gsteele/Documents/GitHub/libsmu/bindings
creating build/temp.macosx-10.7-x86_64-3.7/Users/gsteele/Documents/GitHub/libsmu/bindings/python
creating build/temp.macosx-10.7-x86_64-3.7/Users/gsteele/Documents/GitHub/libsmu/bindings/python/pysmu
gcc -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -I/Users/gsteele/anaconda3/envs/myenv/include -arch x86_64 -I/Users/gsteele/anaconda3/envs/myenv/include -arch x86_64 -I/Users/gsteele/Documents/GitHub/libsmu/include -I/opt/local/include/libusb-1.0 -I/Users/gsteele/anaconda3/envs/myenv/include/python3.7m -c /Users/gsteele/Documents/GitHub/libsmu/bindings/python/pysmu/libsmu.cpp -o build/temp.macosx-10.7-x86_64-3.7/Users/gsteele/Documents/GitHub/libsmu/bindings/python/pysmu/libsmu.o -std=c++11
warning: include path for stdlibc++ headers not found; pass '-std=libc++' on the
      command line to use the libc++ standard library instead
      [-Wstdlibcxx-not-found]
/Users/gsteele/Documents/GitHub/libsmu/bindings/python/pysmu/libsmu.cpp:636:10: fatal error: 
      'ios' file not found
#include "ios"
         ^~~~~
1 warning and 1 error generated.
error: command 'gcc' failed with exit status 1
(myenv) TUD1001336 python$ 
```

Any hints? 

Also, as a general comment: would it not be easier to somehow distribute the python binding pre-compiled? It is a lot to ask of a casual python programmer to have a full c++ build environment installed to install a package. 

In an ideal world, a `pip install pysmu` would be ideal...

Thanks,
Gary

tfcollins, 2020-10-21:
Conda packages have been created: https://wiki.analog.com/university/tools/conda
They include the python bindings and library.

gsteele13, 2020-10-21:
fantastic! i'll give them a try. (it might be good to update readme in github page...)

AlexandraTrifan, 2020-10-21:
We will soon update the README page to include Conda packages as well.

AlexandraTrifan, 2021-02-09:
Conda packages are mentioned in the README page and the official wiki documentation: https://wiki.analog.com/university/tools/m1k/libsmu 
Closing this issue.

closed 2021-02-09 by AlexandraTrifan